prominenceai/deepstream-services-library #2: Add build variant to MakeFile

rjhowell44, 2019-10-14:
Need build option to build obj's into a static lib.  Something like
```
$ make lib
``` 
or whatever approach works best

Sscsmatrix, 2019-10-16:
I updated the makefile with a make lib variant.  This creates a library called dsl-lib.a.  I verified that the various services are there but I am not sure exactly how you want to use this.  So I am going to commit what I have and then you can review it and I will update it.  This will also help me pipe-clean the issue comment system.

rjhowell44, 2019-10-16:
Looks great Mike... I'm closing the issue 
I've made a couple of additional changes.
- for development, `./test/catch.hpp` (single header test framework) is now our new `main()`
- `DslMain.cpp` has been archived (now `DslMain.cpp.arc`) and left for reference.
- the default build is now the test build and can be run by calling
```
$ ds-app
```
or, to see all results, passing and failing
```
$ ds-app -s
```

closed 2019-10-16 by rjhowell44