prominenceai/deepstream-services-library #8: Research simplest, cleanest option for including and calling lib from Python

rjhowell44, 2019-10-17:
I'm hoping something like CTYPES will work.  This issues is dependent on #7 

All C function defined in DslApi.h should be easily callable. All functions use and return simple intrinsic types only (char, unit). The Component 

Should be lots of info online.  Add finds to issues

rjhowell44, 2019-10-18:
@Sscsmatrix New pull request this morning... cleans up the single, client required header file
`DslApi.h`.  I removed few dependencies on other files, and all types used are now intrinsic.
Test cases are currently failing as some of my component API implementation is still "work in progress" (WIP). ... wow, test driven development. :)

Sscsmatrix, 2019-10-18:
I started playing around with ctypes.  I was able to create a shared library and import it.
The command below creates a shared library from a static library:
g++ -shared -o dsl-lib.so dsl-lib.a

then from python you can run:

>>>from ctypes import *

and then:

>>>cdll.LoadLibrary("dsl-lib.so")

which loads the library into python.

Sscsmatrix, 2019-10-18:
Hopefully the system captured my previous comment.  I tried this yesterday and the comment was nowhere to be found.

rjhowell44, 2019-10-20:
@Sscsmatrix, thanks for working on this.  I'm able to build the shared library (cool), but when I try the following:
```python
from ctypes import *

lib = cdll.LoadLibrary("dsl-lib.so")

print(lib.dsl_pipeline_new("pipeline1"))
print(lib.dsl_pipeline_delete("pipeline1"))
```
the script fails because it can't fine the symbols `dsl_pipeline_new` and `dsl_pipeline_delete`.

I'm not sure if we need something added to CFLAGS or something to tell the linker that `DslApi.h` has the `extern "C"` functions in it.  I know how to do in in Windows :-)

Would you have time to look into this further?  The little snippet of code above should run and print out 0 (success) for both function calls

Thanks Mike.

Sscsmatrix, 2019-10-20:
Robert this task is not done.  I was just using the comments to capture my progress.  I found that I need to modify the makefile in order to get the symbols for the dsl stuff included.  But when I did that python complained about the gserver functions not being found.  So I am working on that issue.  I also decided to check out swig to see if that is a better option.

rjhowell44, 2019-10-20:
thanks again @Sscsmatrix.

There are a couple of things I'm particularly interested in:
1. how does Python handle the dsl service return codes, as they are #define's and Python has no equivalent. Perhaps they will need to be redefined in a .py file as variables?
2. how simple is it to register callback functions from Python to C? given the below.

```C++
// callback function 
typedef void (*state_change_listener_cb)(uint prev_state, 
    uint curr_state, void* user_data);

// registration
DslReturnType dsl_pipeline_state_change_listener_add(const char* pipeline, 
    state_change_listener_cb listener, void* user_data);
```

Sscsmatrix, 2019-10-20:
Once I use -fPIC as a compile option, I can then use nm -D dsl-lib.so and see the symbols.  In a terminal, $nm -D dsl-lib.so | less

Unfortunately the following python sequence gives the following error:

import ctypes
lib = ctypes.CDLL("dsl-lib.so")

OSError: dsl-lib.so: undefined symbol: _gst_debug_min

So I am investigating if I need to either import other modules or modify the Makefile or shared library build command to reference gst symbols.

In addition, I am playing around with building my own simple shared library file and python.  Using a simple file might highlight something that I have missed and be easier to see than if I am trying to examine the dsl-lib.so file.
Robert, I will also look into your two items mentioned above.

Sscsmatrix, 2019-10-22:
So I have been able to fix the OSError.  I had to modify the Makefile to build the shared library and link in the gst files.  I added this line to the Makefile
so_lib:
        $(CC) -shared $(OBJS) -o dsl-lib.so $(LIBS) 

I then noticed that I had to place extern "C" for the calls in the DslApi.h file and the cpp file in order to prevent name mangling.  So now I am able to run the commands.  Unfortunately when I try and execute dsl_pipeline_new("pipeline1")  inside of python I get a segmentation fault.

sscsmatrix@sscsmatrix-desktop:~/Projects/CanAmMex/deepstream-services-library$ python3
Python 3.6.8 (default, Oct  7 2019, 12:59:55) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ctypes
>>> libc = ctypes.CDLL("dsl-lib.so")
>>> libc.dsl_pipeline_new("pipeline1")
Segmentation fault (core dumped)

So I am troubleshooting that issue.  Maybe I need to run some initialization code first??

Sscsmatrix, 2019-10-22:
Okay I was able to get rid of the Segmentation fault by running my code on the target.  My issue was that I was executing the code from an SSH session.  If I want to execute the software remotely then I need to use VNC.  So now I am going to start bringing in more of the API into the python library.

rjhowell44, 2019-10-23:
@Sscsmatrix sounds like great progress... just fyi... I made all of the DSL API's extern "C" some time ago... try pulling the latest when you get a change

Sscsmatrix, 2019-10-23:
I checked in changes to the Makefile and you should be able to build dal-lib.so and run it in Python as I described above.  So we should be able to close this issue.  After looking at other methods, it seems to me that using ctypes is the most universal approach and doesn't require installing additional third-party software like swig would require.

closed 2019-10-23 by rjhowell44