prominenceai/deepstream-services-library #36: Python example, creating URI sources from folder

rjhowell44, 2019-10-25:
@Sscsmatrix I thought of an interesting Python example that would accomplish a couple of things. Here's a list of items/steps if your good with this.
- [ ]  Setup examples source folder 
    * `./examples/videos`
    * `./examples/python`
- [ ] Populate the `videos` folder with a number mpeg and jpeg files ( > 10 would be great)
- [ ] Develop a Python example where... given a folder path, iterate through the list of files in the folder creating a URI source for each. Use the relative path for the URI, and filename as the unique source name.  There is no practical limit to how many source objects can be created, just how many can be used by Pipelines at one time. 

I'm working on a couple of dependencies first, and will let you know when done.

Sscsmatrix, 2019-10-30:
@rjhowell44 I added a couple of files that I have been working on.  This is definitely WIP code.  If you run this code it causes a segmentation fault after the add component function completes successfully.  Have you tested the add component function?

Sscsmatrix, 2019-10-30:
Here is a GDB backtrace of the error.  You can get it with the following:
gdb
(gdb) file python3
(gdb) run ds_test1.py
(gdb) backtrace

#0  0x0000007fb7540120 in std::__shared_ptr<DSL::Padtr, (__gnu_cxx::_Lock_policy)2>::operator bool() const () at dsl-lib.so
#1  0x0000007fb753c264 in DSL::Padtr::Unlink() () at dsl-lib.so
#2  0x0000007fb753ac00 in DSL::PipelineSourcesBintr::RemoveChild(std::shared_ptr<DSL::Bintr>) () at dsl-lib.so
#3  0x0000007fb7546b48 in DSL::PipelineBintr::RemoveSourceBintr(std::shared_ptr<DSL::Bintr>) () at dsl-lib.so
#4  0x0000007fb7551bb0 in DSL::SourceBintr::RemoveFromParent(std::shared_ptr<DSL::Bintr>) () at dsl-lib.so
#5  0x0000007fb752db78 in DSL::Services::PipelineComponentRemove(char const*, char const*) () at dsl-lib.so
#6  0x0000007fb7525aa4 in dsl_pipeline_component_remove () at dsl-lib.so
#7  0x0000007fb76f8d28 in ffi_call_SYSV () at /usr/lib/aarch64-linux-gnu/libffi.so.6
#8  0x0000007fb76f9698 in ffi_call () at /usr/lib/aarch64-linux-gnu/libffi.so.6
#9  0x0000007fb7743f34 in _ctypes_callproc () at /usr/lib/python3.6/lib-dynload/_ctypes.cpython-36m-aarch64-linux-gnu.so
#10 0x0000007fb773b298 in  () at /usr/lib/python3.6/lib-dynload/_ctypes.cpython-36m-aarch64-linux-gnu.so
#11 0x000000000044e738 in _PyObject_FastCallDict ()
#12 0x000000000052842c in  ()
#13 0x000000000052dba4 in _PyEval_EvalFrameDefault ()
#14 0x0000000000527208 in  ()
#15 0x000000000052883c in  ()
#16 0x000000000052dba4 in _PyEval_EvalFrameDefault ()

rjhowell44, 2019-10-31:
Hey @Sscsmatrix ... not surprising that the ComponentRemove is failing... still a work in progress. I just checked in a bunch of refactoring that will allow me to complete this.

Sscsmatrix, 2019-10-31:
@rjhowell44 

I am not sure where to put this information but I wanted to capture it somewhere.  I wanted to see if I could set a breakpoint in the dsl-lib.so and get symbolic debugging working.

I edited the Makefile by adding the line: -g3 -gdwarf-2 after -fPIC and then rebuilding the library.  I then copied the new dsl-lib.so to the same directory where ds_test1.py is located.

$ gdb python3
(gdb) break main
(gdb) run ds_test1.py
(gdb) set stop-on-solib 1
(gdb) cont
(gdb) info shared
(gdb) break dsl_pipeline_component_remove
(gdb) cont

Running cont a few times you eventually hit the breakpoint in the shared memory function for dsl_pipeline_component_remove.

rjhowell44, 2019-11-01:
that is awesome Mike... that will be a big help moving forward

Sscsmatrix, 2019-11-13:
@rjhowell44
I updated ds_test1.py so that it works with the latest dsl-lib.so updates.  This program takes as input a video file and displays it.  You need to run it with python 2.7.  So the calling sequence is python ds_test1.py sample_720p.h264

I also added a file ds_csi.py that uses a CSI device and displays realtime video.

rjhowell44, 2019-12-13:
@Sscsmatrix This issue has become a little stale at this point, so I'm going to close and will come up with more of a general strategy  and a list of things to cover with examples.

I've copied your notes on Debugging to the repo's [Wiki page](https://github.com/canammex-tech/deepstream-services-library/wiki) for now, as a place to gather general notes and info.

closed 2019-12-13 by rjhowell44