prominenceai/deepstream-services-library #52: Research RTSP

rjhowell44, 2019-10-31:
@Sscsmatrix while you're waiting on functionality, I thought you might want to take on another little research project.  It would be helpful if we had a better understanding of the Real-Time Streaming Protocol (RTSP).... info on 
* what it is and what it's used for, 
* why it's important for what we're doing
* what types of applications uses it
* who's using it on GitHub

Sscsmatrix, 2019-11-05:
@rjhowell44 I downloaded a couple of RFCs, RFC2326 and RFC1889.  It appears that this protocol specifies how to use UDP and TCP to reliably stream media.  One of the use cases would be to use this protocol so that a client could start watching a video before it was completely downloaded from a server.  So it makes sense to me that this would be part of Gstreamer.  Any way I am going to review the RFC's and also read up on what Gstreamer documentation has to say about it.  I will also spend some time looking at GitHub to see who else might use it.

rjhowell44, 2019-11-06:
@Sscsmatrix I'll write up a few notes here on how the deepstream reference app is using it

closed 2019-12-21 by rjhowell44