prominenceai/deepstream-services-library #62: Latest update (11/10/2019) crashes gstreamer

Sscsmatrix, 2019-11-10:
@rjhowell44 
After the latest commits, executing the script below:
python3 ds_test1.py sample_720p.h264

causes this error message:

(python3:9865): GStreamer-CRITICAL **: 12:02:58.754: gst_element_add_pad: assertion 'GST_IS_ELEMENT (element)' failed
terminate called without an active exception
Aborted (core dumped)

If I revert back to commit 039f3960a2b3cf6f5676130ee33176d315b62fa4, then the error does not occur.

Sscsmatrix, 2019-11-10:
@rjhowell44 

It appears that the problem occurs when I call:
src = DSL.dsl_source_uri_new(src_name, uri, 0, 0)

Sscsmatrix, 2019-11-10:
@rjhowell44 
So it looks like the AddGhostPadToParent from DslElementr.h is where the issue is occurring.

Sscsmatrix, 2019-11-10:
@rjhowell44 
Here is some more debug information.
(python3:14727): GStreamer-CRITICAL **: 14:04:36.948: gst_element_add_pad: assertion 'GST_IS_ELEMENT (element)' failed
0:00:02.014312550 14727     0x3b92b800 ERROR                    DSL src/DslElementr.h:129:AddGhostPadToParent:  : Failed to add Pad 'src' for element'src-queue'
terminate called without an active exception
Aborted (core dumped)

rjhowell44, 2019-11-10:
@Sscsmatrix I'm just getting up to that level now... just FYI, I still have a few sections of URI source commented out. I've been focusing on CSI first as it's much simpler.. I hope to have my first Pipeline working with a CSI source tonight, and when done will go back to the URI.  I let you know when I have something at the API level working.

Sscsmatrix, 2019-11-14:
I can now execute my script, provided that I use python instead of python 3.  Since that should probably be a different issue, I am going to close this issue.

closed 2019-11-14 by Sscsmatrix