prominenceai/deepstream-services-library #98: Fix primary GIE deserialization errors on load model engine

rjhowell44, 2019-11-25:
@Sscsmatrix the Primary GIE issues I was having were related to config and model file issues.  Strangely enough, I'm unable to run the combination of 
* `test/config/config_infer_primary_nano.txt` 
* `test/models/Prmary_Detector_Nano/resnet10.caffemodel`
... results in deserialization errors when loading the engine on Play

However, the combination of 
* `test/config/config_infer_primary.txt` 
* `test/models/Prmary_Detector/resnet10.caffemodel_b1_fp16.engine`
...seems to work for me...  pretty sure I bought a Nano 

I'll try and search the dev.talk when I get a chance, but glad to be finally moving past this issue.

I'm working on the OSD boxing next so we can visually verify.

closed 2019-11-25 by rjhowell44

Sscsmatrix, 2019-11-25:
@rjhowell44 Glad that you found this.  I was trying to get it working on Saturday but I was running into the same problem.  So I started looking at the DeepStream-test1-app to try and figure out what the difference was.  Any way now I can play with your solution!

rjhowell44, 2019-11-25:
It turns out that it was a path issue, and I have the Nano configs working now.  The problem is, that if the plugin can't find your model at the path specified, it looks for a default and loads an incompatible engine.  I just did a new pull request that has the basic on-screen-display working.  The simple sample scripts are doing more now... getting to be fun