prominenceai/deepstream-services-library #133: Implement Window Sink (EGL/GLES)

rjhowell44, 2019-12-27:


rjhowell44, 2019-12-30:
@Sscsmatrix I managed to jack up the repo a bit yesterday... It's problem with having the mdel engine binaries in the repo.  Please make sure to keep a copy of your folder, before doing a pull as I may have clobbered something in your most recent pull request.  I added your nano config files back in, but I think I missed your script.

That said, there is a new Window Sink component to play with, Implements the functionality  that I thought the Overlay Sink was going to give us.  There is a new example script

closed 2019-12-30 by rjhowell44

Sscsmatrix, 2019-12-31:
@rjhowell44 I noticed that you made a bunch of commits and some of my work “disappeared”.  I just figured that I screwed up and you didn’t care for my changes.  Not to worry, I have my stuff backed up and I can roll back your changes and find anything that is missing.  I am anxious to play with the Windows sink component.  I noticed that when I run our stuff remotely with VNC, the video does not show up on the remote system but it does appear on the attached monitor.  If I run a similar Nvidia example, the window is displayed on both.  Maybe this fixes that issue.