prominenceai/deepstream-services-library #227: Errors when i try start

trickititit, 2020-03-04:
Hi,
First error get in make (all):
/usr/bin/ld: cannot find -laprutil-1
/usr/bin/ld: cannot find -lapr-1

This error i try resolve editing makefile:

LIBS+= -L$(LIB_INSTALL_DIR) \
	-lX11 \
	-L/usr/lib/aarch64-linux-gnu \
	-laprutil-1 \
	-lapr-1 \
	-lnvdsgst_meta \
	-lnvds_meta \
	-lnvdsgst_helper \
	-lnvds_utils \
	-lglib-$(GLIB_VERSION) \
	-lgstreamer-$(GSTREAMER_VERSION) \
	-Lgstreamer-video-$(GSTREAMER_VERSION) \
	-Lgstreamer-rtsp-server-$(GSTREAMER_VERSION) \
	-Wl,-rpath,$(LIB_INSTALL_DIR)

After compiling end successed

Second error get when i try exec python examples:

ImportError: /home/jetson/Downloads/deepstream-services-library/examples/python/pyds.so: undefined symbol: PyInstanceMethod_Type

rjhowell44, 2020-03-04:
Hi, trickititit, I will look into your makefile issue. Second, as you're aware, the pyds.so bindings are from Nvidia, so I'm not sure what's going on with the ImportError.  Are you using python3?  

I was wondering if you tried the dsl.so binary provided in the release before building.?

trickititit, 2020-03-04:
Yes, i use python3.
Ok, you can share laprutil-1 and lapr-1??

They are not in this folder:
opt/nvidia/deepstream/deepstream-4.0/lib/

rjhowell44, 2020-03-04:
Please see https://stackoverflow.com/questions/22075531/how-to-install-apr-in-ubuntu

apr = Apache Portable Runtime.  Separate from DeepStream and GStreamer, but required to build X11 Windows for video rendering.

rjhowell44, 2020-03-04:
I will add a note to the documentation

rjhowell44, 2020-03-06:
Hi, trickititit, do you have what you need? can I close this issue?

trickititit, 2020-03-10:
===============================================================================
test cases:  309 |  264 passed | 45 failed
assertions: 1613 | 1567 passed | 46 failed

trickititit, 2020-03-11:
Ok, please help?
how to access the frame for clipping a detected object?
I found where get rects.

rjhowell44, 2020-03-11:
Hi trickititit, 
First, the Nano GIE model files, required to run the test cases, are not checked into the repository. They exceed GitHub's size restriction, event when compressed.... thus the test failures. I will do some experimenting today to see if the compressed versions can be added to a release vs. the repo.  

Second, RE:  _how to access the frame for clipping._  We will need look through the NVIDIA examples to see what they have, and post a comment on their dev board if we can't figure it out.  It is not something I have done myself yet. I'll look into it today.

trickititit, 2020-03-18:
Hi, not results? thanks

rjhowell44, 2020-03-18:
Hi trickititit,

I've moved this request to a new issue  #228  please refer to comments for current progress.

Closing this issue.

closed 2020-03-18 by rjhowell44