prominenceai/deepstream-services-library #240: Can't build on Jetson Nano (aprutil-1 and apr-1 not found)

jlerasmus, 2020-04-22:
_Disclaimer: I am new to Linux, C++, Make, etc._

I am unable to successfully build the dsl-test-app on my Jetson Nano Devkit following [these](https://github.com/canammex-tech/deepstream-services-library/blob/master/docs/building-dsl.md) instructions.

At the end of the process the following errors are display:
```
/usr/bin/ld: cannot find -laprutil-1
/usr/bin/ld: cannot find -lapr-1
collect2: error: ld returned 1 exit status
Makefile:124: recipe for target 'dsl-test-app' failed
make: *** [dsl-test-app] Error 1
```

I tried installing libapr1 and libaprutil1, but that did not work:
```
sudo apt-get install libapr1
sudo apt-get install libaprutil1
sudo ldconfig
```

I would appreciate your help with this as I am eager to try out your lib.

Herewith the full log ouput:
```
sudo make
g++ -c -o src/Dsl.h.gch -DPLATFORM_TEGRA -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/opt/nvidia/deepstream/deepstream-4.0/sources/apps/apps-common/includes -I/opt/include -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/usr/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/include/glib-2.0/glib -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/local/cuda-10.0/targets/aarch64-linux/include -I./src -I./test -I./test/api -DDSL_VERSION='L"v0.04.alpha"' -DDS_VERSION_MINOR=0 -DDS_VERSION_MAJOR=4 -DDSL_LOGGER_IMP='"DslLogGst.h"' -DNVDS_KLT_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_klt.so"' -DNVDS_IOU_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_iou.so"' -fPIC  `pkg-config --cflags gstreamer-1.0 gstreamer-video-1.0 gstreamer-rtsp-server-1.0 x11 opencv4` src/Dsl.h
g++ -c -o src/DslServices.o -DPLATFORM_TEGRA -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/opt/nvidia/deepstream/deepstream-4.0/sources/apps/apps-common/includes -I/opt/include -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/usr/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/include/glib-2.0/glib -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/local/cuda-10.0/targets/aarch64-linux/include -I./src -I./test -I./test/api -DDSL_VERSION='L"v0.04.alpha"' -DDS_VERSION_MINOR=0 -DDS_VERSION_MAJOR=4 -DDSL_LOGGER_IMP='"DslLogGst.h"' -DNVDS_KLT_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_klt.so"' -DNVDS_IOU_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_iou.so"' -fPIC  `pkg-config --cflags gstreamer-1.0 gstreamer-video-1.0 gstreamer-rtsp-server-1.0 x11 opencv4` src/DslServices.cpp
g++ -c -o src/DslBranchBintr.o -DPLATFORM_TEGRA -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/opt/nvidia/deepstream/deepstream-4.0/sources/apps/apps-common/includes -I/opt/include -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/usr/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/include/glib-2.0/glib -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/local/cuda-10.0/targets/aarch64-linux/include -I./src -I./test -I./test/api -DDSL_VERSION='L"v0.04.alpha"' -DDS_VERSION_MINOR=0 -DDS_VERSION_MAJOR=4 -DDSL_LOGGER_IMP='"DslLogGst.h"' -DNVDS_KLT_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_klt.so"' -DNVDS_IOU_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_iou.so"' -fPIC  `pkg-config --cflags gstreamer-1.0 gstreamer-video-1.0 gstreamer-rtsp-server-1.0 x11 opencv4` src/DslBranchBintr.cpp
g++ -c -o src/DslOsdBintr.o -DPLATFORM_TEGRA -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/opt/nvidia/deepstream/deepstream-4.0/sources/apps/apps-common/includes -I/opt/include -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/usr/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/include/glib-2.0/glib -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/local/cuda-10.0/targets/aarch64-linux/include -I./src -I./test -I./test/api -DDSL_VERSION='L"v0.04.alpha"' -DDS_VERSION_MINOR=0 -DDS_VERSION_MAJOR=4 -DDSL_LOGGER_IMP='"DslLogGst.h"' -DNVDS_KLT_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_klt.so"' -DNVDS_IOU_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_iou.so"' -fPIC  `pkg-config --cflags gstreamer-1.0 gstreamer-video-1.0 gstreamer-rtsp-server-1.0 x11 opencv4` src/DslOsdBintr.cpp
g++ -c -o src/DslPipelineSourcesBintr.o -DPLATFORM_TEGRA -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/opt/nvidia/deepstream/deepstream-4.0/sources/apps/apps-common/includes -I/opt/include -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/usr/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/include/glib-2.0/glib -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/local/cuda-10.0/targets/aarch64-linux/include -I./src -I./test -I./test/api -DDSL_VERSION='L"v0.04.alpha"' -DDS_VERSION_MINOR=0 -DDS_VERSION_MAJOR=4 -DDSL_LOGGER_IMP='"DslLogGst.h"' -DNVDS_KLT_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_klt.so"' -DNVDS_IOU_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_iou.so"' -fPIC  `pkg-config --cflags gstreamer-1.0 gstreamer-video-1.0 gstreamer-rtsp-server-1.0 x11 opencv4` src/DslPipelineSourcesBintr.cpp
g++ -c -o src/DslPipelineBintr.o -DPLATFORM_TEGRA -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/opt/nvidia/deepstream/deepstream-4.0/sources/apps/apps-common/includes -I/opt/include -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/usr/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/include/glib-2.0/glib -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/local/cuda-10.0/targets/aarch64-linux/include -I./src -I./test -I./test/api -DDSL_VERSION='L"v0.04.alpha"' -DDS_VERSION_MINOR=0 -DDS_VERSION_MAJOR=4 -DDSL_LOGGER_IMP='"DslLogGst.h"' -DNVDS_KLT_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_klt.so"' -DNVDS_IOU_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_iou.so"' -fPIC  `pkg-config --cflags gstreamer-1.0 gstreamer-video-1.0 gstreamer-rtsp-server-1.0 x11 opencv4` src/DslPipelineBintr.cpp
g++ -c -o src/DslPipelineSGiesBintr.o -DPLATFORM_TEGRA -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/opt/nvidia/deepstream/deepstream-4.0/sources/apps/apps-common/includes -I/opt/include -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/usr/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/include/glib-2.0/glib -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/local/cuda-10.0/targets/aarch64-linux/include -I./src -I./test -I./test/api -DDSL_VERSION='L"v0.04.alpha"' -DDS_VERSION_MINOR=0 -DDS_VERSION_MAJOR=4 -DDSL_LOGGER_IMP='"DslLogGst.h"' -DNVDS_KLT_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_klt.so"' -DNVDS_IOU_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_iou.so"' -fPIC  `pkg-config --cflags gstreamer-1.0 gstreamer-video-1.0 gstreamer-rtsp-server-1.0 x11 opencv4` src/DslPipelineSGiesBintr.cpp
g++ -c -o src/DslTrackerBintr.o -DPLATFORM_TEGRA -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/opt/nvidia/deepstream/deepstream-4.0/sources/apps/apps-common/includes -I/opt/include -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/usr/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/include/glib-2.0/glib -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/local/cuda-10.0/targets/aarch64-linux/include -I./src -I./test -I./test/api -DDSL_VERSION='L"v0.04.alpha"' -DDS_VERSION_MINOR=0 -DDS_VERSION_MAJOR=4 -DDSL_LOGGER_IMP='"DslLogGst.h"' -DNVDS_KLT_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_klt.so"' -DNVDS_IOU_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_iou.so"' -fPIC  `pkg-config --cflags gstreamer-1.0 gstreamer-video-1.0 gstreamer-rtsp-server-1.0 x11 opencv4` src/DslTrackerBintr.cpp
g++ -c -o src/DslGieBintr.o -DPLATFORM_TEGRA -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/opt/nvidia/deepstream/deepstream-4.0/sources/apps/apps-common/includes -I/opt/include -I/opt/nvidia/deepstream/deepstream-4.0/sources/includes -I/usr/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/include/glib-2.0/glib -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/local/cuda-10.0/targets/aarch64-linux/include -I./src -I./test -I./test/api -DDSL_VERSION='L"v0.04.alpha"' -DDS_VERSION_MINOR=0 -DDS_VERSION_MAJOR=4 -DDSL_LOGGER_IMP='"DslLogGst.h"' -DNVDS_KLT_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_klt.so"' -DNVDS_IOU_LIB='"/opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_mot_iou.so"' -fPIC  `pkg-config --cfla [... cut ...] -lnvbufsurface -lnvbufsurftransform -lglib-2.0 -lgstreamer-1.0 -Lgstreamer-video-1.0 -Lgstreamer-rtsp-server-1.0 -L/usr/local/cuda-10.0/lib64/ -lcudart -Wl,-rpath,/opt/nvidia/deepstream/deepstream-4.0/lib `pkg-config --libs gstreamer-1.0 gstreamer-video-1.0 gstreamer-rtsp-server-1.0 x11 opencv4`
/usr/bin/ld: cannot find -laprutil-1
/usr/bin/ld: cannot find -lapr-1
collect2: error: ld returned 1 exit status
Makefile:124: recipe for target 'dsl-test-app' failed
make: *** [dsl-test-app] Error 1
```

rjhowell44, 2020-04-22:
Hi jlerasmus,

sudo apt-get install libapr1 libapr1-dev should typically work.

Please see https://stackoverflow.com/questions/22075531/how-to-install-apr-in-ubuntu for some additional suggestions. 

Just FYI, you can always **download** the latest version of the pre-built lib from the [Release Tab](https://github.com/canammex-tech/deepstream-services-library/releases) and skip the build.

And when first developing your scripts, it's always advisable to turn on logging with
```
$ export GST_DEBUG=1,DSL:4
```
And don't hesitate to ask questions or provide suggestions if anything comes up

Regards,

jlerasmus, 2020-04-23:
Hi Robert, thanks for the feedback.

1. The following worked `sudo apt-get install libapr1 libapr1-dev libaprutil1 libaprutil1-dev`

2. Silly me, got so used to cloning repos and building stuff that I never thought of checking for a release. Good job of a having releases.

3. I will keep the debugging in mind.

Regards.

closed 2020-04-23 by jlerasmus

rjhowell44, 2020-08-27:
@Sscsmatrix see the above