prominenceai/deepstream-services-library #250: Deepstream 5 Support and Smart Recrod

jlerasmus, 2020-06-11:
Hi Robert,

Do you intend to upgrade/support Deepstream 5 (currently Developer Preview)?

I also saw your work on ODE services which looks promising.  I think integrating Smart Record with such events is desirable.  A Python sample or 2 for this integration would be a nice touch.

Regards.

rjhowell44, 2020-06-11:
Hi jterasmum,

Yes, on both accounts... I hope to wrap up a first version of ODE services and alpha v0.5 by the end of the week,  Once released I'm going to start on the Deepstream 5 upgrade, and the Smart recording will be the first new ODE action to implement.   I'll be adding new github Issues for both items, but will leave this issue open until then. 

I'm glad to here you think the ODE services look promising.  Comments and requests are always very much appreciated.  

Regards,

jlerasmus, 2020-06-12:
Great!  Looking forward to the release and the new features.

For now Python samples on the intended use of the ODE services would be great for completing the feature.  I am sure there will be more comments and requests.

rjhowell44, 2020-06-16:
@jlerasmus There's a sandbox script that I've been playing around with a little [here](https://github.com/canammex-tech/deepstream-services-library/blob/ode_services/examples/python/ode_sandbox.py)... it's a bit complicated for a first example, but I wanted to try out a number of the different combinations to see how things look. It's kind of fun to run if you're able to build the `ode_services` branch and build the lib. I feel good about the the ODE Handler, Actions, and Areas, but I'm not convinced I have the Trigger types (occurrence, absence, intersection, etc.) right just yet.  More tomorrow.

jlerasmus, 2020-06-16:
Looks interesting.  Wanted to try it out, but the ODE methods are missing from [dsl.py](https://github.com/canammex-tech/deepstream-services-library/blob/ode_services/dsl.py).  Maybe you missed the commit?

rjhowell44, 2020-06-16:
you are correct.. so sorry.

rjhowell44, 2020-06-17:
@jlerasmus more updates pushed to the ode_services branch... hopefully you're able to try the script out now.  I've made some progress on the ODE API documentation under the branch as well.

jlerasmus, 2020-06-19:
@rjhowell44 I just pulled the latest ode_services branch and was able to compile and run the ode_sandbox.py sript, but it seems the ode triggers are not triggering.

**Please note:** this is from my Nano running Jetpack 4.4 DP with Deepstream 5.0 DP which I know is not supported (I have fixed the build errors and compiled the lib, although there are a few warning regarding float > uint type conversions)

This is what I am experiencing:
- The video starts playing in the window
- The 3 classes from the display actions are drawn on the top left
- The 2 areas are drawn
- All bounding boxes for people, bicycle and vehicles are drawn

But
- No person or vehicle are shaded red
- No bicycle image is captured
- The counts always stays zero

**PS** I will try and run this from a Jetpack 4.3 with Deepstream 4 setup.

Something else... when changing the PGIE interval for example to 4, the areas start flashing.  I have not checked but it seems they are only added on inferencing.  Does this also mean that the triggers will only be performed from actual inferencing, and not updated by the tracker's results?

rjhowell44, 2020-06-19:
@jlerasmus I'm hopeful this is just a version compatibility issue. The fact that you are getting type conversion warnings... I'm not surprised the behavior is different.  
 
RE: *Something else... when changing the PGIE interval for example to 4, the areas start flashing.* This is my error. I'm filtering on a `bInferDone` flag instead of just reporting it. I will fix today. 

Hopefully, it's not a big deal for you to backdate and try this on the previous Jetpack/Deepstream. 

Thanks for the detailed feedback... I very much appreciate it.

rjhowell44, 2020-07-02:
Hi @jlerasmus ,  sorry, but I had to support some other work for a bit there.. but the upgrade to DS 5.0 is now done.  Starting on the Smart Record next... 

regards,

rjhowell44, 2020-07-05:
New, SR specific issue

Implement new DSL component for DS 5.0 Smart Record #263

closed 2020-07-05 by rjhowell44