prominenceai/deepstream-services-library #251: Development Setup

jlerasmus, 2020-06-12:
Hi Robert,

What is your Development setup currently, or recommended setup?  I am a experienced C# developer, but have limited knowledge on C++ and Linux development.

I currently have a dual boot Ubuntu host setup with VSCode and the C++ extension, with which I am able to navigate the code, have code completion available and to compile.

It would be great know your setup with regards to Host OS, IDE, Debugger, etc?  Maybe at some point I can assist with development :-)

Regards.

rjhowell44, 2020-06-12:
I do all my development on the Jetson Nano target. One of the many things I love about this platform is that the entire workflow can be done from the device. 

Once you have everything installed to run the Deepstream demo, including the Apache runtime dependencies for the XWindow support, you should be able to clone the repo from the device and follow the build directions [here](https://github.com/canammex-tech/deepstream-services-library/blob/master/docs/building-dsl.md). 

All test cases run on the device, but you will need your own versions of the Inference models/engines as they are two large to add to the Repo.  I believe this issue will be solved by upgrading to DS 5.0, where I  everything needed is available for download? Not sure.

I use [CodeLite](https://codelite.org/) as my editor/debugger. I tried to get an Ubuntu version of VSCode to work on the device, but I ran into a number of issues. 

I would suggest upgrading to a 128GB card if you haven't already. 

Any help on development would be greatly appreciated... really hoping to get others involved on all aspects of design, development, and test... and I would love to know more about your plans/needs for your project if you're willing to share.

Regards,

jlerasmus, 2020-06-15:
Hi Robert,

For VSCode on the Jetson Nano, you could use the instructions [here](https://github.com/JetsonHacksNano/installVSCode).  This works great for Python, C# (.NET Core) and other editing, but the C++ extension does not currently support ARM64 platform - thus no intellisense.  The support seems to be coming soon, see the issue [here](https://github.com/microsoft/vscode-cpptools/issues/2506)

CodeLite seems as a good alternative.  I wanted to give a try, but dit not manage the get a full install running.  I did a install following [these instructions](https://wiki.codelite.org/pmwiki.php/Main/Repositories), but that only installed v10 with no C++ extensions listed.  How did you go about installing it?  Or did you build from source?

I would be glad to assist with development, once I have more experience on GStreamer, Deepstream, etc. For now I will probably be more of a user/tester as I try out different options.

For now I want to continue the discussion on one of my use cases on the [forum](https://forums.developer.nvidia.com/t/saving-meta-data-matching-smartrecord-files/127478) to gather other input as well.  But I believe the common requirement for many other use cases would be Infer - Track - Analytics - Rules (Triggers) - Actions, with video recordings of the actions.

Regards.

rjhowell44, 2020-06-16:
I will definitely try the VSCode instructions... CodeLite is ok, but not great... not much to offer for python development.

jlerasmus, 2020-06-16:
I also forgot to mention that you can use the [SSH Remote Development Extensions](https://code.visualstudio.com/docs/remote/ssh), which allows you to run VSCode on your host and develop on the Jetson Nano (freeing up some resources on the Nano).

I do hope they get the ARM64 support released for the C++ Extension soon.

rjhowell44, 2020-06-17:
When you use the SSH Remote, do you have any issues with the XWindows or RTSP Sink?

jlerasmus, 2020-06-19:
They work for me either running via ssh (which then opens up the display on the Nano) or from a terminal on the Nano itself.  RTSP streaming seems slow, will need to investigate.

Make sure you are logged in on the Nano itself if login is required, otherwise I get EGL sink errors (have to double check the RTSP before login)

Also, I always need to `export DISPLAY=:1` even on the Nano itself (with a screen and keyboard attached).  The display is confirmed after running `w` from the terminal.  This has always been the case, even with the Deepstream samples.  Not sure why this is though>

jlerasmus, 2020-07-22:
I think we can close this issue.

Regarding the display-id, I have posted in the Nvidia Developer Forums: https://forums.developer.nvidia.com/t/deepstream-display-id/143053

closed 2020-07-22 by jlerasmus