prominenceai/deepstream-services-library #259: ODE Services - Open Items

rjhowell44, 2020-06-30:


jlerasmus, 2020-07-24:
**[Feature] Polygon Area**
The rectangular are to trigger actions is a good start, but polygon areas would be a great addition.

**[Feature]Line Crossing Trigger**
Many scenarios require line crossing detection, with optional direction detection.

Maybe these should use the new deepstream-analytics.

rjhowell44, 2020-07-24:
I would like to avoid using the deepstream-analytics if possible as I really think this concept of triggers/areas/actions could grow overtime...  just so we're not waiting on NVIDIA for new analytics. 

I'm thinking of going with [GEOS](https://trac.osgeo.org/geos)... or something similar, Should provide lots of options.

jlerasmus, 2020-07-24:
Makes sense to not wait on NVIDIA for something... and I do like the triggers/areas/actions

rjhowell44, 2020-07-24:
That said, the plan has always been to support as much, if not everything, that NVIDIA adds to DeepStream... so I'll definitely be looking at how the deepstream-analytics best fits in. 

I opened a new topic on the NVIDIA board [DeepStream, request for new NvOSD_PolygonParams type](https://forums.developer.nvidia.com/t/deepstream-request-for-new-nvosd-polygonparams-type/143959/2)

maybe if enough people like the request it will get added.. seems missing at this point.

On another topic, the current Trigger Area criteria is fixed at -- "at least one pixel of intersection" -- to trigger, which is too restrictive. I was thinking of extending this to allow for a percentage of intersection... and change the area to be inclusive/exclusive.

re:**[Feature] Line Crossing Trigger** 

For sure we can do something here... I saw an interesting demo video for a new hikvision fish eye camera that was mounted looking down over an office space..  where you can specify floor areas and then trigger on someone crossing from one area into another.  It should be easy to support this with a single Trigger using multiple areas.  What was really cool about the demo however, was that they tracked the number of people and their locations overtime, and could then overlay a heat map onto the video... all very easy to do with the triggers/areas/action

jlerasmus, 2020-07-27:
I have liked your forum post (maybe I should post a reply).  I definitely need a polygon area as the rectangular area does not feel right with the perspective of the cameras, and is a bit limiting.

I would agree that adding inclusive/exclusive criteria would improve the area trigger.  And as you mentioned, having better control of what results an inclusion/exclusion would be great.  Maybe a start is to have fixed calculations like: 
- Percentage-Overlap (IOU) = Specify a percentage of pixels that should overlap
- Inside = Whole bounding box to be included
- Outside = Whole bounding box to be excluded
- Center = The center of the bounding box to be included/excluded
- Bottom-Center = The bottom-center of the bounding box to be included (on top off)

I think the idea of having multiple areas with trigger criteria specifying when the object crosses from one area to another should work.  This can surely be used to simulate a line-crossing trigger, but I do feel that it might be more natural to have a line drawn with a direction arrow would be natural for users.  Maybe draw the 2 areas with the line to be explicit.

Please share the Hikvision demo video, maybe I would get a better understanding?

Having a heatmap overlay would be cool

rjhowell44, 2020-07-29:
I think we may need to iterate on this a bit, as there are a number of interesting things to think through.

First, I think lines and line crossing are required... and whether it's crossing a side of a shape or a line, the method of detection should be similar. 

Also, we can do polygons now, it is just a matter of how to display them. One option would be to just use lines to draw the border  and not worry about filling the background. 

Back to Areas, I like the idea of being able to specify a percentage of overlap but there are a few questions. For example, what does 100% overlap mean if the object is larger than the area, vs. smaller.  Must the object be inside the Area, or is covering 100% of the Area work as well? 

Being able to specify required Points of overlay is very interesting as well... any type of corner, center, or edge-center. would work.   Trying to think through how percentage and points could work together. or are they mutually exclusive?  Setting the center as a required point and a percentage?  Corners maybe? 

Have you played around with the polygons and Line crossing in the DeepStream release?  I'm assuming they must have some of the same issues, what if it's a short line and only one corner of an object crosses it?  I need to spend some time on it myself.

Here's the hikvision demo https://www.youtube.com/watch?v=xGYcYtCvT2Y&t=320s it's actual a video on RTSP setup, but he talks about the new camera he's using and some of the new features.  including the heat maps.  The areas are actually draw using lines extending to the parameter.

jlerasmus, 2020-07-30:
> Also, we can do polygons now...

Great! Is this already implemented as I didn't see anything?  It would be great if it could have RGBA background, but I don't think it is an immediate requirement.  The deepstream analytics sample also only draws the border.

Regarding the areas, I would think that the options I listed are mutually exclusive.

I can't think of a real world example where one would require the center-point to cross more than x-percentage into an area, or over a line.  Maybe these videos help to clarify the center-point, or bottom-point options.  For visual clarity, it would be good to draw the detection point used as in the samples links below
- Center
https://www.pyimagesearch.com/2018/08/13/opencv-people-counter/
https://www.youtube.com/watch?v=2kPFhc2gb5E
- Bottom-Center (on-top-off)
https://www.youtube.com/watch?v=CVN8UL3TFvM
- Custom-Point (user defined function to calculate detection point used)
https://www.youtube.com/watch?v=loxLfeIOtr8

Regarding the percantage-overlap, maybe something like the typical IOU calculation might work as a start.  I need to think this through and draw some diagrams.  As reference: https://www.pyimagesearch.com/2016/11/07/intersection-over-union-iou-for-object-detection/

> Have you played around with the polygons and Line crossing in the DeepStream release...

I did put up a simple test, and does seem to work reasonably well but I need to spend some more time here.

> Here's the hikvision demo...

Thanks for the link.  The intersection analysis is really interesting.

rjhowell44, 2020-07-30:
@jlerasmus  sorry, what I meant is we (I) can implement it now without waiting for nvidia to provide support for the display.  

but it now looks like they are going to add in the next release.  https://forums.developer.nvidia.com/t/deepstream-request-for-new-nvosd-polygonparams-type/143959

Thanks for the links I'll check them out

rjhowell44, 2020-08-06:
@jlerasmus I've now realized that developing the ODE Handler as a pipeline/branch component with a fixed location between GIEs/Tracker and Tiler was a mistake. The Handler, as a true batch-meta handler, should be added to a Tiler, Muxer or OSD directly. 

* adding to Tiler/Demuxer sink-pad will work as it does now, with all triggers and actions acting on the pre-tiled (pre-demuxed) batched streams, 
* adding to the OSD sink-pad, downstream of the Tiler/Demuxer, to work on the single 2D tiled or demuxed stream.

After making this change, we focus on the model for Triggers and Areas again,

jlerasmus, 2020-08-07:
@rjhowell44 I trust you decision and priorities on this.

rjhowell44, 2021-03-03:
@jlerasmus  with the addition of line and polygon area types... closing this issue... and we can open new issues for specific additions/improvements.

closed 2021-03-03 by rjhowell44