prominenceai/deepstream-services-library #269: Implement new Display Types API

rjhowell44, 2020-07-13:
New display types, color, font, text, line, rectangle, and circle.

New Overlay actions will be implemented to overlay Display objects created from the text, line, rectangle, and circle display types.

**Constructors**
* `dsl_display_type_rgba_color_new`
* `dsl_display_type_rgba_font_new`
* `dsl_display_type_rgba_text_new`
* `dsl_display_type_rgba_line_new`
* `dsl_display_type_rgba_rectangle_new`
* `dsl_display_type_rgba_circle_new`

**Destructors**
* `dsl_display_type_delete`
* `dsl_display_type_delete_many`
* `dsl_display_type_delete_all`

**Methods**
* `dsl_display_type_list_size`

rjhowell44, 2020-07-13:
Types API has been implemented... needs documentation

rjhowell44, 2020-07-26:
will cover documentation in a separate v0.07.alpha docs issue

closed 2020-07-26 by rjhowell44

bharathraja, 2022-07-13:
its working, when i added extra library flags "-L/usr/local/lib -lopencv_core -lopencv_imgproc -lopencv_highgui" to makefile. i will update in git and close the issue. i think it is due to some linking error

> $(APP): $(OBJS) Makefile	
> 	$(CXX) -L/usr/local/lib -lopencv_core -lopencv_imgproc -lopencv_highgui -o $(APP) $(OBJS) $(CFLAGS) $(LIBS)

.