prominenceai/deepstream-services-library #281: v0.07.alpha RTSP source not working

jlerasmus, 2020-07-22:
Hi Robert,

I have pulled your v0.07.alpha branch to test the new smart record integration and ODE improvements, but I am no longer able to use my RTSP sources.  I have tested with your sample `1rtsp_1csi_live_pgie_tiler_osd_window.py` without the csi-source and changed to one of my IP cam urls.

I know this is work in progress, just wanted to give a heads-up.  ODE changes looks good, I am trying to setup a complete live sample.

Herewith the log.

```
0:00:00.048500464 13608     0x32288300 INFO                     DSL src/DslServices.cpp:318:GetServices:  : Services Initialization
0:00:00.069694211 13608     0x32288300 INFO                     DSL src/DslSourceBintr.cpp:389:DecodeSourceBintr:  : URI Path = rtsp://<removed>
0:00:00.080933934 13608     0x32288300 INFO                     DSL src/DslServices.cpp:2682:SourceRtspNew:  : New RTSP Source 'rtsp-source' created successfully
0:00:00.081124615 13608     0x32288300 INFO                     DSL src/DslServices.cpp:3036:PrimaryGieNew:  : Infer config file: ../../test/configs/config_infer_primary_nano.txt
0:00:00.081204564 13608     0x32288300 INFO                     DSL src/DslServices.cpp:3048:PrimaryGieNew:  : Model engine file: ../../test/models/Primary_Detector_Nano/resnet10.caffemodel_b8_gpu0_fp16.engine
0:00:00.081408317 13608     0x32288300 INFO                     DSL src/DslGieBintr.cpp:53:GieBintr:  : Creating GIE  'gie-primary-gie' with unique Id = 1557520627
0:00:00.143929333 13608     0x32288300 INFO                     DSL src/DslServices.cpp:3059:PrimaryGieNew:  : New Primary GIE 'primary-gie' created successfully
0:00:00.146579066 13608     0x32288300 INFO                     DSL src/DslServices.cpp:3877:TilerNew:  : New Tiler 'tiler' created successfully
0:00:00.155026449 13608     0x32288300 INFO                     DSL src/DslServices.cpp:4291:OsdNew:  : New OSD 'on-screen-display' created successfully
0:00:00.158960162 13608     0x32288300 INFO                     DSL src/DslServices.cpp:4726:SinkWindowNew:  : New Window Sink 'window-sink' created successfully
0:00:00.159576892 13608     0x32288300 INFO                     DSL src/DslServices.cpp:5784:PipelineNew:  : New PIPELINE 'pipeline' created successfully
0:00:00.161956255 13608     0x32288300 INFO                     DSL src/DslPipelineSourcesBintr.cpp:268:SetStreamMuxDimensions:  : Setting StreamMux dimensions: width = 1920, height = 1920
0:00:00.162588870 13608     0x32288300 INFO                     DSL src/DslServices.cpp:5873:PipelineComponentAdd:  : Component 'rtsp-source' was added to Pipeline 'pipeline' successfully
0:00:00.163644044 13608     0x32288300 INFO                     DSL src/DslServices.cpp:5873:PipelineComponentAdd:  : Component 'window-sink' was added to Pipeline 'pipeline' successfully
0:00:00.164089677 13608     0x32288300 INFO                     DSL src/DslPipelineSourcesBintr.cpp:246:SetStreamMuxBatchProperties:  : Setting StreamMux batch properties: batch-size = 1, batch-timeout = 4000000
0:00:00.164366296 13608     0x32288300 INFO                     DSL src/DslSourceBintr.cpp:109:LinkToSink:  : Linking Source 'rtsp-source' to Pad 'sink_0' for StreamMux 'stream_muxer'
0:00:00.164718802 13608     0x32288300 INFO                     DSL src/DslPipelineBintr.cpp:273:LinkAll:  : Pipeline 'pipeline' Linked up all Source 'sources-bin' successfully
0:00:00.164781668 13608     0x32288300 INFO                     DSL src/DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'window-sink'
0:00:00.164818804 13608     0x32288300 INFO                     DSL src/DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'sinks-bin'
0:00:00.165418189 13608     0x32288300 INFO                     DSL src/DslSinkBintr.cpp:92:LinkToSource:  : Linking Sink 'window-sink' to Pad 'src_0' for Tee 'sink_bin_tee'
0:00:00.165473972 13608     0x32288300 INFO                     DSL src/DslElementr.h:168:IsFactoryName:  : commparing expected factory'nvstreamdemux' with actual 'tee' for element 'sink_bin_tee'
0:00:00.166208307 13608     0x32288300 INFO                     DSL src/DslBranchBintr.cpp:414:LinkAll:  : Branch 'pipeline' Linked up all Sinks 'sinks-bin' successfully
0:00:00.166264923 13608     0x32288300 INFO                     DSL src/DslBintr.h:227:SetState:  : Changing state to 'PLAYING' for Bintr 'pipeline'

Using winsys: x11 
0:00:00.217757625 13608     0x32288300 INFO                     DSL src/DslPipelineBintr.cpp:560:HandleBusSyncMessage:  : Prepare window handle received from source sink-bin-eglgles
0:00:00.217942940 13608     0x32288300 INFO                     DSL src/DslPipelineBintr.cpp:683:CreateXWindow:  : Creating new XWindow with width = 1280: height = 720
0:00:00.221177683 13608     0x32288300 INFO                     DSL src/DslBintr.h:241:SetState:  : State change will complete asynchronously for Bintr 'pipeline'
0:00:00.396496145 13608   0x7f4c004b70 INFO                     DSL src/DslSourceBintr.cpp:983:HandleSourceElementOnPadAdded:  : Caps structs name application/x-rtp
Segmentation fault (core dumped)
```

rjhowell44, 2020-07-22:
Hi @jlerasmus ... nice to hear from you.

Yes, sorry... work in progress.  I'm adding in the support for the record-tap now. But pleas check back. I should have it back together shortly.  

I would love to get your feedback on everything new.

Also, be aware... there are a number of issues with the current version of the smart-record lib. Hopefully the DS 5.0 GA release will have them resolved.

jlerasmus, 2020-07-22:
Yeah, glad to get some time to work on this again.  I have been swamped with other projects.

I will check back for updates, for now I am trying to setup a use case using DSL.  Should be able to give more feedback later, but something I am definitely missing for now is polygon areas to trigger actions, and I believe line-crossing will be next.  Maybe these should use the new deepstream-analytics.

Should I post findings/requirments as seperate issues, or on #259 "ODE Services - Open Items"

rjhowell44, 2020-07-22:
Adding to the current  #259 "ODE Services - Open Items" works to start with

rjhowell44, 2020-07-26:
@jlerasmus the RTSP source is working again, at least for my limited camera collection. Please let me know if you find otherwise.

jlerasmus, 2020-07-27:
@rjhowell44, I can confirm that the RTSP source is working now.  I am testing with 8x ProVision ISR and 1x Hikvision cameras.

I do believe that there is some latency/frame dropping issue, especially with the ProVision ISR cameras.  I would need to confirm this with FPS counters, but I can definitely see this visually on the embedded clocks from each camera.  Maybe a feature request is to add FPS stats to DSL like with deepstream-app.

This latency does not seem to occur with the deepstream-app.

rjhowell44, 2020-07-29:
Resolving any latency/frame dropping issue will be my new top priority.. especially while I have someone with 8 cameras to help verify (yes, I have camera envy).   I know of one issue I've been meaning to get to that could be responsible for this.  I'll work on getting a build with the FPS counters and some optimization next. Thanks for the help. Very much appreciated.

rjhowell44, 2020-07-30:
@jlerasmus I've checked in a couple of fixes into the v0.07.alpha tip that should help with the throughput issues.  I haven't had a chance to implement the FPS stats, but I will.

rjhowell44, 2020-07-30:
@jlerasmus sorry, the KTL tracker is still not performing will.. IOU seems much better.

jlerasmus, 2020-07-30:
@rjhowell44, thanks for looking into this.  I am more than willing to help testing any performance issues (and others).

Looking forward to have the FPS stats available, as this would be the only concrete way to compare results.  I do think that one should also be able to log these stats to the console, as with the sample deepstream-app, and not only have them as overlays.

I will try out the latest changes and report back.  I am currently using the KTL tracker but will try with the IOU tracker as well.

rjhowell44, 2020-08-04:
@jlerasmus I found that I was enabling the `qos` property for the Render Sinks by default, and then not handling QoS events when they occurred.  I'm defaulting (hard-coding) the property to `False` for now, and will add Set/Get routines to enable [QoS management](https://gstreamer.freedesktop.org/documentation/plugin-development/advanced/qos.html?gi-language=c) in the future. 

After making this change, frame-rates seem comparable to the deepstream app. 

The issue: Implement new Meter Sink for measuring/reporting Pipeline throughput #298  has been implemented and checked into the v0.07.alpha branch.

See [8uri_file_meter_sink_performace_reporting.py](https://github.com/canammex-tech/deepstream-services-library/blob/v0.07.alpha/examples/python/8uri_file_meter_sink_performace_reporting.py) as an example of how to use the Meter until I can get the API reference updated.

Let me know how everything looks... when you get chance... 

Thanks,

jlerasmus, 2020-08-05:
@rjhowell44 Thank you for adding this feature.

The meter-sink is reporting the expected FPS counters and is comparable to the deepstream-app.

Visually is looks way better and I can't see any issue by looking at the embedded clocks or general stream from each camera.  For fast moving objects (e.g. cars) though, there are some jumpiness - but this is the same with the deepstream-app and is therefore not related with DSL.

I will investigate further, maybe it is something with OSD, Tiler or even StreamMuxer.  I will try and confirm this with gst-launch as well.  I would appreciate if you have some pointers on this...

Here is an updated `meter_sink_handler` that prints the output in columns like the deepstream-app does, just easier for comparison.  Note that I could not get the values passed by using the `client_data`, it is always None even when specifying it when the sink is created. Maybe I am doing something wrong.

```
# Global metering values
meter_report_count = 0
meter_header_interval = 12

def meter_sink_handler(session_avgs, interval_avgs, source_count, client_data):
    """Meter Sink client callback funtion"""

    # Reference global variables (could not get values passed via client_data)
    global meter_report_count
    global meter_header_interval

    # Print header on interval
    if (meter_report_count % meter_header_interval == 0):
        header = ""
        for source in range(source_count):
            subheader = f"FPS {source} (AVG)"
            header += "{:<15}".format(subheader)
        print()
        print(header)

    # Print FPS counters
    counters = ""
    for source in range(source_count):
        counter = "{:.2f} ({:.2f})".format(interval_avgs[source], session_avgs[source])
        counters += "{:<15}".format(counter)
    print(counters)

    # Increment reporting count
    meter_report_count += 1

    return True
```

rjhowell44, 2020-08-06:
@jlerasmus See Resolve `client_data` not passed back correctly on callback #305

Once I have this resolved, if I could ask you to send me a pull request with the above (using the client data) added to the example script?  I'm more than happy to include any improvements. 

Thanks for all the help

rjhowell44, 2020-08-06:
@jlerasmus RE: I will investigate further, maybe it is something with OSD, Tiler or even StreamMuxer. I will try and confirm this with gst-launch as well. I would appreciate if you have some pointers on this...

According to the [Low-Level Tracker Library Comparisons and Tradeoffs](https://docs.nvidia.com/metropolis/deepstream/plugin-manual/index.html#page/DeepStream_Plugin_Manual/deepstream_plugin_details.02.02.html) that the KTL Tracker is very CPU intensive. The image looks much better with the IOU Tracker, not that the tracking is better and probable less suitable in most cases. 

I find the OSD adds minimal overhead with just boxes and labels... but adding large blended rectangles sure does. 

Setting the `qos` reporting property for the Sink allows you to see which combinations of components cause the frames to show up late, sooner than other combinations... keeping in mind that the qos measurements add a fair bit of overhead as well.

rjhowell44, 2020-08-07:
@jlerasmus I've checked some changes into the v.0.07.alpha branch for *Resolve client_data not passed back correctly on callback* #305. 

The example performance script has been updated with your callback above, plus... please see the new use of the `client_data` and and how to cast. 

This will need to need to be covered in the overview, and I should add to other python examples as well.

jlerasmus, 2020-08-07:
@rjhowell44, thanks for resolving the `client_data` bug and updating the sample.

Regarding the jumpiness (seems that frames are skipped), this even occurs without adding pgie, tracker or osd.  Pipeline is thus x8 rtsp sources, tiler and either overlay or window sink.  I also see this with the deepstream-app, also without pgie and tracker, but seems less pronounced.  Overlay-sink vs Window-sink does not seem to make a difference.

Increasing the latency on the rtsp sources for example to 500ms does seem to help, but increasing it to 2000ms seems to make it worse. I need to confirm what the deepstream-app does if latencies are set.  Preferably I would want to keep the latencies as low as possible.

With the above rtsp-tiler-sink pipeline, CPU sits arround 30%-35% and no GPU usage - so definitely not a bottleneck.

Also, it seems that setting "attach-sys-ts" property on the stream-muxer does seem to help both DSL test and deepstream-app.  But this is all visual confirmation and no concrete timings.  Not even sure how to do this.

Regarding the `qos`, any pointers on how to use it or where to read-up on that.

rjhowell44, 2020-08-08:
@jlerasmus re: `qos` the only think that I've been able to find is the GST overview  [QoS management](https://gstreamer.freedesktop.org/documentation/plugin-development/advanced/qos.html?gi-language=c). 

re "Regarding the jumpiness" I spend most of my time staring at the nvidia sample video while working on infrastructure, so I don't have a lot of experience with live streams and video quality.  I'm glad you're looking into this :+1:

rjhowell44, 2020-08-12:
@jlerasmus let me know if we can close this issue... although, I hate having the discussing lost in a closed issues..

may need to move to Discord for discussions in the future so as to keep some of the information around.

jlerasmus, 2020-08-12:
@rjhowell44 we can close the issue as the RTSP source is working.

Regarding the jumpiness... I am still trying different options and it seems to be related to nvstreammuxer. It seems to interfere with the latency specified on the rtsp source, or the ntp timings. I have switch to testing different settings using gst-launch with a pipeline of rtsp-sources=>nvstreammuxer

closed 2020-08-12 by jlerasmus