prominenceai/deepstream-services-library #283: v0.05 rtsp not working

xmyyzy123, 2020-07-23:
Hi,
 This work is really awesome!
 I am trying to play rtsp source with DSL, but got this error.
 Here with the log:

> 0:00:00.015210331 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:281:GetServices:  : Services Initialization
0:00:00.015519922 24856 0x560f6776ea00 INFO                     DSL DslSourceBintr.cpp:387:DecodeSourceBintr:  : URI Path = rtsp://
0:00:00.020504886 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:2191:SourceRtspNew:  : New RTSP Source 'rtsp-source' created successfully
0:00:00.020537865 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:2534:PrimaryGieNew:  : Infer config file: config_infer_primary_yoloV3.txt
0:00:00.020572790 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:2543:PrimaryGieNew:  : Model engine file: model_b1_int8.engine
0:00:00.020618009 24856 0x560f6776ea00 INFO                     DSL DslGieBintr.cpp:53:GieBintr:  : Creating GIE  'gie-primary-gie' with unique Id = 1557520627
0:00:00.052669228 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:2561:PrimaryGieNew:  : New Primary GIE 'primary-gie' created successfully
0:00:00.053579795 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:3365:TilerNew:  : New Tiler 'tiler' created successfully
0:00:00.054451059 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:3772:OsdNew:  : New OSD 'on-screen-display' created successfully
0:00:00.054706612 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:4146:SinkFakeNew:  : New Fake Sink 'rtsp-sink' created successfully
0:00:00.054829557 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:4984:PipelineNew:  : New PIPELINE 'pipeline' created successfully
0:00:00.055413589 24856 0x560f6776ea00 INFO                     DSL DslPipelineSourcesBintr.cpp:268:SetStreamMuxDimensions:  : Setting StreamMux dimensions: width = 1920, height = 1920
0:00:00.055572633 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:5073:PipelineComponentAdd:  : Component 'rtsp-source' was added to Pipeline 'pipeline' successfully
0:00:00.055640005 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:5073:PipelineComponentAdd:  : Component 'primary-gie' was added to Pipeline 'pipeline' successfully
0:00:00.055696301 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:5073:PipelineComponentAdd:  : Component 'tiler' was added to Pipeline 'pipeline' successfully
0:00:00.055757148 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:5073:PipelineComponentAdd:  : Component 'on-screen-display' was added to Pipeline 'pipeline' successfully
0:00:00.056011528 24856 0x560f6776ea00 INFO                     DSL DslServices.cpp:5073:PipelineComponentAdd:  : Component 'rtsp-sink' was added to Pipeline 'pipeline' successfully
0:00:00.056044608 24856 0x560f6776ea00 INFO                     DSL DslPipelineSourcesBintr.cpp:246:SetStreamMuxBatchProperties:  : Setting StreamMux batch properties: batch-size = 1, batch-timeout = 4000000
0:00:00.056136624 24856 0x560f6776ea00 INFO                     DSL DslSourceBintr.cpp:109:LinkToSink:  : Linking Source 'rtsp-source' to Pad 'sink_0' for StreamMux 'stream_muxer'
0:00:00.056249128 24856 0x560f6776ea00 INFO                     DSL DslPipelineBintr.cpp:273:LinkAll:  : Pipeline 'pipeline' Linked up all Source 'sources-bin' successfully
0:00:00.056274487 24856 0x560f6776ea00 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'primary-gie'
0:00:00.056575254 24856 0x560f6776ea00 INFO                     DSL DslBranchBintr.cpp:280:LinkAll:  : Branch 'pipeline' Linked up Primary GIE 'primary-gie' successfully
0:00:00.056588585 24856 0x560f6776ea00 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'tiler'
0:00:00.056775312 24856 0x560f6776ea00 INFO                     DSL DslBranchBintr.cpp:354:LinkAll:  : Branch 'pipeline' Linked up Tiler 'tiler' successfully
0:00:00.056787560 24856 0x560f6776ea00 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'on-screen-display'
0:00:00.057145639 24856 0x560f6776ea00 INFO                     DSL DslBranchBintr.cpp:369:LinkAll:  : Branch 'pipeline' Linked up OSD 'on-screen-display' successfully
0:00:00.057162092 24856 0x560f6776ea00 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'rtsp-sink'
0:00:00.057172355 24856 0x560f6776ea00 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'sinks-bin'
0:00:00.057212986 24856 0x560f6776ea00 INFO                     DSL DslSinkBintr.cpp:101:LinkToSource:  : Linking Sink 'rtsp-sink' to Pad 'src_0' for Tee 'sink_bin_tee'
0:00:00.057226316 24856 0x560f6776ea00 INFO                     DSL DslElementr.h:168:IsFactoryName:  : commparing expected factory'nvstreamdemux' with actual 'tee' for element 'sink_bin_tee'
0:00:00.057515446 24856 0x560f6776ea00 INFO                     DSL DslBranchBintr.cpp:414:LinkAll:  : Branch 'pipeline' Linked up all Sinks 'sinks-bin' successfully
0:00:00.057530432 24856 0x560f6776ea00 INFO                     DSL DslBintr.h:227:SetState:  : Changing state to 'PLAYING' for Bintr 'pipeline'
Creating LL OSD context new
Deserialize yoloLayerV3 plugin: yolo_83
Deserialize yoloLayerV3 plugin: yolo_95
Deserialize yoloLayerV3 plugin: yolo_107
0:00:04.334478643 24856 0x560f6776ea00 INFO                     DSL DslBintr.h:241:SetState:  : State change will complete asynchronously for Bintr 'pipeline'
0:00:04.454866032 24856 0x7fd78c003f70 INFO                     DSL DslSourceBintr.cpp:914:HandleSourceElementOnPadAdded:  : Caps structs name application/x-rtp
0:00:04.454946449 24856 0x7fd78c003f70 INFO                     DSL DslSourceBintr.cpp:929:HandleSourceElementOnPadAdded:  : Video decode linked for URI source 'rtsp-source'
(DSL_test:24856): GLib-GObject-WARNING **: 21:49:04.776: g_object_set_is_valid_property: object class 'nvv4l2decoder' has no property named 'enable-max-performance'
(DSL_test:24856): GLib-GObject-WARNING **: 21:49:04.776: g_object_set_is_valid_property: object class 'nvv4l2decoder' has no property named 'bufapi-version'
0:00:04.526482221 24856 0x7fd778003770 INFO                     DSL DslSourceBintr.cpp:914:HandleSourceElementOnPadAdded:  : Caps structs name application/x-rtp
0:00:04.526536568 24856 0x7fd778003770 ERROR                    DSL DslSourceBintr.cpp:925:HandleSourceElementOnPadAdded:  : Failed to link de-payload to pipeline
terminate called without an active exception
Aborted (core dumped)<

 Any help would be greatly appreciated！

rjhowell44, 2020-07-23:
Hi @xmyyzy123,

The RTSP source in v0.05 has some limitations, a fixed `rtph264depay` for one, I'm in the process of adding the stream-selection support in v0.07. Can you tell me a little about your camera settings and which protocol value you are using when you create the Source? 

Also, do you play to upgrade to DS 5.0 ?  Unfortunately, I'm not able to support any fixes/updates to pre-5.0 releases

xmyyzy123, 2020-07-24:
Hi @rjhowell44,
I tried DS 5.0, still failed. 
Also, when I use dsl_source_uri_new, it does not work.
I use like this:
`` const wchar_t* uri = L"/home/streams/sample_1080p_h264.mp4"; ``
`` dsl_source_uri_new(L"uri-source", uri, false, DSL_CUDADEC_MEMTYPE_DEVICE, false, 0); ``
Is this right?
this is the log:

> 0:00:00.097502807  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:280:GetServices:  : Services Initialization
0:00:00.098911446  9272   0x557562d8a0 INFO                     DSL DslSourceBintr.cpp:389:DecodeSourceBintr:  : URI Path = file:/home/streams/sample_1080p_h264.mp4
0:00:00.120658381  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:2163:SourceUriNew:  : New URI Source 'uri-source' created successfully
0:00:00.120793581  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:2533:PrimaryGieNew:  : Infer config file: config_infer_primary_yoloV3.txt
0:00:00.120895149  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:2545:PrimaryGieNew:  : Model engine file: model_b1_gpu0_int8.engine
0:00:00.121087661  9272   0x557562d8a0 INFO                     DSL DslGieBintr.cpp:53:GieBintr:  : Creating GIE  'gie-primary-gie' with unique Id = 1557520627
Warn: 'threshold' parameter has been deprecated. Use 'pre-cluster-threshold' instead.
0:00:01.074293224  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:2564:PrimaryGieNew:  : New Primary GIE 'primary-gie' created successfully
0:00:01.084782787  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:3368:TilerNew:  : New Tiler 'tiler' created successfully
0:00:01.103720348  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:3775:OsdNew:  : New OSD 'on-screen-display' created successfully
0:00:01.104525020  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:4149:SinkFakeNew:  : New Fake Sink 'rtsp-sink' created successfully
0:00:01.105023099  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:4987:PipelineNew:  : New PIPELINE 'pipeline' created successfully
0:00:01.116323895  9272   0x557562d8a0 INFO                     DSL DslPipelineSourcesBintr.cpp:268:SetStreamMuxDimensions:  : Setting StreamMux dimensions: width = 1920, height = 1080
0:00:01.117019351  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:5076:PipelineComponentAdd:  : Component 'uri-source' was added to Pipeline 'pipeline' successfully
0:00:01.117446263  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:5076:PipelineComponentAdd:  : Component 'primary-gie' was added to Pipeline 'pipeline' successfully
0:00:01.117668374  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:5076:PipelineComponentAdd:  : Component 'tiler' was added to Pipeline 'pipeline' successfully
0:00:01.117876374  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:5076:PipelineComponentAdd:  : Component 'on-screen-display' was added to Pipeline 'pipeline' successfully
0:00:01.118841942  9272   0x557562d8a0 INFO                     DSL DslServices.cpp:5076:PipelineComponentAdd:  : Component 'rtsp-sink' was added to Pipeline 'pipeline' successfully
0:00:01.118987958  9272   0x557562d8a0 INFO                     DSL DslPipelineSourcesBintr.cpp:246:SetStreamMuxBatchProperties:  : Setting StreamMux batch properties: batch-size = 1, batch-timeout = 4000000
0:00:01.119652438  9272   0x557562d8a0 INFO                     DSL DslSourceBintr.cpp:109:LinkToSink:  : Linking Source 'uri-source' to Pad 'sink_0' for StreamMux 'stream_muxer'
0:00:01.120182293  9272   0x557562d8a0 INFO                     DSL DslPipelineBintr.cpp:273:LinkAll:  : Pipeline 'pipeline' Linked up all Source 'sources-bin' successfully
0:00:01.120275829  9272   0x557562d8a0 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'primary-gie'
0:00:01.121810709  9272   0x557562d8a0 INFO                     DSL DslBranchBintr.cpp:280:LinkAll:  : Branch 'pipeline' Linked up Primary GIE 'primary-gie' successfully
0:00:01.121859573  9272   0x557562d8a0 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'tiler'
0:00:01.122882836  9272   0x557562d8a0 INFO                     DSL DslBranchBintr.cpp:354:LinkAll:  : Branch 'pipeline' Linked up Tiler 'tiler' successfully
0:00:01.122927316  9272   0x557562d8a0 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'on-screen-display'
0:00:01.124736916  9272   0x557562d8a0 INFO                     DSL DslBranchBintr.cpp:369:LinkAll:  : Branch 'pipeline' Linked up OSD 'on-screen-display' successfully
0:00:01.124789172  9272   0x557562d8a0 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'rtsp-sink'
0:00:01.124828916  9272   0x557562d8a0 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'sinks-bin'
0:00:01.125035828  9272   0x557562d8a0 INFO                     DSL DslSinkBintr.cpp:101:LinkToSource:  : Linking Sink 'rtsp-sink' to Pad 'src_0' for Tee 'sink_bin_tee'
0:00:01.125090484  9272   0x557562d8a0 INFO                     DSL DslElementr.h:168:IsFactoryName:  : commparing expected factory'nvstreamdemux' with actual 'tee' for element 'sink_bin_tee'
0:00:01.126487059  9272   0x557562d8a0 INFO                     DSL DslBranchBintr.cpp:414:LinkAll:  : Branch 'pipeline' Linked up all Sinks 'sinks-bin' successfully
[... cut ...]
0:00:25.554153721  9272   0x557562d8a0 INFO                     DSL DslPipelineBintr.cpp:526:HandleBusWatchMessage:  : Message type:: GST_MESSAGE_STREAM_STATUS
0:00:25.554627129  9272   0x557562d8a0 ERROR                    DSL DslPipelineBintr.cpp:751:HandleErrorMessage:  : Error message 'Internal data stream error.' received from 'src-queue'
0:00:25.554889241  9272   0x557562d8a0 INFO                     DSL DslPipelineBintr.cpp:526:HandleBusWatchMessage:  : Message type:: GST_MESSAGE_ELEMENT

>

rjhowell44, 2020-07-24:
After you moved to DS 5.0, which version of DSL are you now using ... still v0.05.alpha? or are you using a later version?

DS.5.0 will only work with v0.06.alpha or latter (0.07 is still in progress)

Sorry, for the basic questions, but it looks like you are using C/C++ , correct?

The source startup looks correct in the above... but, the primary-gie is not happy with the stream format. 

I see you are using a Yolo detector, something I have not had a change to verify yet, but I'll make it  a priority now.

xmyyzy123, 2020-07-24:
Hi, @rjhowell44,
I am using v0.06 after moving to DS 5.0.
Yes, I am using C++.
The stream format is OK, I tested in Deepstream.
I also used other detector in the models folder, failed.
I have a question, do you notice this line:

> 0:00:00.098911446 9272 0x557562d8a0 INFO DSL DslSourceBintr.cpp:389:DecodeSourceBintr: : URI Path = file:/home/streams/sample_1080p_h264.mp4

In Deepstream,  uri which as input is used like this:

> file:///home/streams/sample_1080p_h264.mp4

Is that the problem?

rjhowell44, 2020-07-24:
Just FYI, as I'm sure you've noticed, all the current examples are in Python.... one of the best files to see working C++ code is 

```/test/api/DslPipelinePlayComponentsTest.cpp```

The path looks good... the following shows that the Source has come up 
````
0:00:17.522395616 9272 0x7f00013050 INFO DSL DslSourceBintr.cpp:730:HandleSourceElementOnPadAdded: : Caps structs name video/x-raw
0:00:17.523840831 9272 0x7f00013050 INFO DSL DslSourceBintr.cpp:750:HandleSourceElementOnPadAdded: : Video decode linked for URI source 'uri-source'
```

Everything looks good right up to 
```
GST_MESSAGE_TAG
0:00:18.671663301 9272 0x5575a972d0 WARN nvinfer gstnvinfer.cpp:1946:gst_nvinfer_output_loop: error: Internal data stream error.
0:00:18.671905221 9272 0x5575a972d0 WARN nvinfer gstnvinfer.cpp:1946:gst_nvinfer_output_loop: error: streaming stopped, reason not-linked (-1)
0:00:18.672463973 9272 0x557562d8a0 ERROR DSL DslPipelineBintr.cpp:751:HandleErrorMessage: : Error message 'Internal data stream error.' received from 'gie-primary-gie'
```

Can you setup the following and we can take a look the Pipeline graph?

https://github.com/canammex-tech/deepstream-services-library/blob/master/docs/debugging-dsl.md#creating-pipeline-graphs

rjhowell44, 2020-07-24:
Also, you can try the Pipeline without the GIE   Just Source, Tiler, OSD and Sinks

xmyyzy123, 2020-07-24:
@rjhowell44,
Yes, without GIE, no error.

rjhowell44, 2020-07-24:
I'll try and look at the  Yolo examples tomorrow..  getting late here for me

xmyyzy123, 2020-07-24:
@rjhowell44, 
ok, thank you.
But this is another detector, same error:
```

0:00:00.065954040 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:280:GetServices:  : Services Initialization
0:00:00.067225409 18682   0x55a0b13aa0 INFO                     DSL DslSourceBintr.cpp:389:DecodeSourceBintr:  : URI Path = file:/home/streams/sample_1080p_h264.mp4
0:00:00.076384711 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:2163:SourceUriNew:  : New URI Source 'uri-source' created successfully
0:00:00.076520744 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:2533:PrimaryGieNew:  : Infer config file: config_infer_primary_nano.txt
0:00:00.076607721 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:2545:PrimaryGieNew:  : Model engine file: resnet10.caffemodel_b1_gpu0_fp16.engine
0:00:00.076874187 18682   0x55a0b13aa0 INFO                     DSL DslGieBintr.cpp:53:GieBintr:  : Creating GIE  'gie-primary-gie' with unique Id = 1557520627
0:00:00.200893086 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:2564:PrimaryGieNew:  : New Primary GIE 'primary-gie' created successfully
0:00:00.204802236 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:3368:TilerNew:  : New Tiler 'tiler' created successfully
0:00:00.210326983 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:3775:OsdNew:  : New OSD 'on-screen-display' created successfully
0:00:00.211249902 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:4149:SinkFakeNew:  : New Fake Sink 'rtsp-sink' created successfully
0:00:00.211695153 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:4987:PipelineNew:  : New PIPELINE 'pipeline' created successfully
0:00:00.215195660 18682   0x55a0b13aa0 INFO                     DSL DslPipelineSourcesBintr.cpp:268:SetStreamMuxDimensions:  : Setting StreamMux dimensions: width = 1920, height = 1080
0:00:00.215852689 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:5076:PipelineComponentAdd:  : Component 'uri-source' was added to Pipeline 'pipeline' successfully
0:00:00.216183059 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:5076:PipelineComponentAdd:  : Component 'primary-gie' was added to Pipeline 'pipeline' successfully
0:00:00.216424053 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:5076:PipelineComponentAdd:  : Component 'tiler' was added to Pipeline 'pipeline' successfully
0:00:00.216652535 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:5076:PipelineComponentAdd:  : Component 'on-screen-display' was added to Pipeline 'pipeline' successfully
0:00:00.217712735 18682   0x55a0b13aa0 INFO                     DSL DslServices.cpp:5076:PipelineComponentAdd:  : Component 'rtsp-sink' was added to Pipeline 'pipeline' successfully
0:00:00.217829376 18682   0x55a0b13aa0 INFO                     DSL DslPipelineSourcesBintr.cpp:246:SetStreamMuxBatchProperties:  : Setting StreamMux batch properties: batch-size = 1, batch-timeout = 4000000
0:00:00.219021705 18682   0x55a0b13aa0 INFO                     DSL DslSourceBintr.cpp:109:LinkToSink:  : Linking Source 'uri-source' to Pad 'sink_0' for StreamMux 'stream_muxer'
0:00:00.219869135 18682   0x55a0b13aa0 INFO                     DSL DslPipelineBintr.cpp:273:LinkAll:  : Pipeline 'pipeline' Linked up all Source 'sources-bin' successfully
0:00:00.220014801 18682   0x55a0b13aa0 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'primary-gie'
0:00:00.221501820 18682   0x55a0b13aa0 INFO                     DSL DslBranchBintr.cpp:280:LinkAll:  : Branch 'pipeline' Linked up Primary GIE 'primary-gie' successfully
0:00:00.221570652 18682   0x55a0b13aa0 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'tiler'
0:00:00.222847302 18682   0x55a0b13aa0 INFO                     DSL DslBranchBintr.cpp:354:LinkAll:  : Branch 'pipeline' Linked up Tiler 'tiler' successfully
0:00:00.222970823 18682   0x55a0b13aa0 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'on-screen-display'
0:00:00.225138840 18682   0x55a0b13aa0 INFO                     DSL DslBranchBintr.cpp:369:LinkAll:  : Branch 'pipeline' Linked up OSD 'on-screen-display' successfully
0:00:00.225231384 18682   0x55a0b13aa0 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'rtsp-sink'
0:00:00.225319801 18682   0x55a0b13aa0 INFO                     DSL DslBintr.h:208:SetBatchSize:  : Setting batch size to '1' for Bintr 'sinks-bin'
0:00:00.225538811 18682   0x55a0b13aa0 INFO                     DSL DslSinkBintr.cpp:101:LinkToSource:  : Linking Sink 'rtsp-sink' to Pad 'src_0' for Tee 'sink_bin_tee'
0:00:00.225591675 18682   0x55a0b13aa0 INFO                     DSL DslElementr.h:168:IsFactoryName:  : commparing expected factory'nvstreamdemux' with actual 'tee' for element 'sink_bin_tee'
0:00:00.227398153 18682   0x55a0b13aa0 INFO                     DSL DslBranchBintr.cpp:414:LinkAll:  : Branch 'pipeline' Linked up all Sinks 'sinks-bin' successfully
0:00:00.227484010 18682   0x55a0b13aa0 INFO                     DSL DslBintr.h:227:SetState:  : Changing state to 'PAUSED' for Bintr 'pipeline'
[... cut ...]
0:00:10.285590314 18682   0x55a0b13aa0 INFO                     DSL DslPipelineBintr.cpp:526:HandleBusWatchMessage:  : Message type:: GST_MESSAGE_STREAM_STATUS
0:00:10.286341200 18682   0x55a0b13aa0 ERROR                    DSL DslPipelineBintr.cpp:751:HandleErrorMessage:  : Error message 'Internal data stream error.' received from 'src-queue'
0:00:10.286527026 18682   0x55a0b13aa0 INFO                     DSL DslPipelineBintr.cpp:526:HandleBusWatchMessage:  : Message type:: GST_MESSAGE_ELEMENT
```

xmyyzy123, 2020-07-24:
[image]

xmyyzy123, 2020-07-24:
@rjhowell44 
I seem to find the problem.
When I do not use the fake sink, no error occurred.
I tried RTSP sink and window sink, it worked well.
Does it seem something wrong with fake sink?

xmyyzy123, 2020-07-24:
fake sink does not linked. 
[image]

rjhowell44, 2020-07-24:
@xmyyzy123 

I've opened...  Fake Sink element not linking to Sink Queue in Bintr #285 ... for this issue

Thanks for the debugging.

rjhowell44, 2020-07-25:
@xmyyzy123 the sink issue has been resolved in the tip of the [v0.07.alpha](https://github.com/canammex-tech/deepstream-services-library/tree/v0.07.alpha) branch

xmyyzy123, 2020-07-26:
@rjhowell44,
thank you.
I will try.

rjhowell44, 2020-07-26:
@xmyyzy123 I should have mentioned that the RTSP Source in the  v0.07.alpha has been updated, if you could let me know if it works for you now?  that would be great.

xmyyzy123, 2020-07-27:
@rjhowell44,
thank you, it works now.

rjhowell44, 2020-07-27:
@xmyyzy123 can we close this issue now?

closed 2020-07-27 by xmyyzy123