prominenceai/deepstream-services-library #301: Implement new Tiler services for `show-source` management

rjhowell44, 2020-08-04:
Implement new Tiler services

1. get service to get a Source's name from an x and y Tiler offset 
2. get/set services to get/set the Tiler's `show-source` property.

```C
#define DSL_TILER_ALL_SOURCES NULL

DslReturnType dsl_tiler_source_name_get(const wchar_t* tiler, 
    uint x_offset, uint y_offset, const wchar_t** source);

DslReturnType dsl_tiler_source_show_get(const wchar_t* tiler, const wchar_t** source);
DslReturnType dsl_tiler_source_show_set(const wchar_t* tiler, const wchar_t* source);
```

rjhowell44, 2020-08-05:
done, needs documentation...

jlerasmus, 2020-08-05:
@rjhowell44 thanks for this, works like a charm for key events.  An improvement would be to get the source-id resolved from coordinates and passed through to the button handler attached with `dsl_pipeline_xwindow_button_event_handler_add`.

This would allow the user to then show the specific source on mouse-click like with the deepstream-app.  It would also then be required to pass through the button that was pressed. #290 would allow for displaying the source information.

Is it possible to have keyboard and/or mouse events with an overlay sink?

rjhowell44, 2020-08-06:
Hey @jlerasmus 
RE:  ***An improvement would be to get the source-id resolved from coordinates***
Yes, I agree... I've just been trying to think through the use cases. My original though was to have a second Tiler service to show a source based on x/y offsets,  

But for the case where the window/button dimensions are scaled differently from Tiler, the client would need to do the scaling in the button-pressed callback. 

Another option would be to provide a Pipeline service that returns the Source Id based on Window coordinates, and then the client could call the show-source with the Id. There may be other reasons for selecting a source, beyond just showing it. 

Let me know your thoughts. 

RE: ***Is it possible to have keyboard and/or mouse events with an overlay sink?***
There is no way of adding a key event handler to the sink directly, that I'm aware of... but the overlay sink needs to overlay a window - the examples, as I'm sure you're aware of, all use the main window with Id "0" - so I'm assuming you can install your own main window key handler, or create a new xWindow and overlay the sink, or something like that.

jlerasmus, 2020-08-07:
@rjhowell44 my current use case is to show the source by clicking on the specific source's tile, and then show that source (as with the deepstream-app sample).  Basically the same as by pressing the source id on the keyboard (e.g. 0 through 7).

Another use case might be to show the source that triggered an action, e.g. when a person is detected show the source for 5 seconds (or until another source triggers an event), and then switch back to tiled mode.

I currently have something working in the python button-click-handler, but it only works if the window is not scaled/resized as that information is not available (only the xpos and ypos is passed).  Also, there is no indication of what button was pressed, therefore one can't have something like left-click to show the source, and right-click to return to tiled mode.

A pipeline service that resolves the source-id from the window coordinates should do the trick.

rjhowell44, 2020-08-08:
@jlerasmus I like the action Idea... will definitely include something.

rjhowell44, 2020-08-29:
This may be enhanced in the future. Going with the following for the v0.07.alpha release

```C
/** 
 * @brief Gets the current Show Source setting for the named Tiler
 * @param[in] name unique name of the Tiler to query
 * @param[out] source name of the current source shown by the Tiler. 
 * A value of DSL_TILER_ALL_SOURCES (equal to NULL) indicates all sources are shown
 * @param[out] current remaining timeout value, 0 if showing all sources and the timer is not running.
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_TILER_RESULT
 */
DslReturnType dsl_tiler_source_show_get(const wchar_t* name, 
    const wchar_t** source, uint* timeout);

/** 
 * @brief Shows a single source instead of all tiled sources - the default tile mode.
 * @param[in] name unique name of the Tiler to update
 * @param[in] source unique name of the source to show,
 * @param[in] timeout time to show the source in units of seconds, before showing all-sources again
 * A value of 0 indicates no timeout. 
 * @param[in] has_precedence if true will take precedence over a currently show single source. 
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_TILER_RESULT
 */
DslReturnType dsl_tiler_source_show_set(const wchar_t* name, 
    const wchar_t* source, uint timeout, boolean has_precedence);

/** 
 * @brief Shows all sources and stops the show-source timer if running.
 * @param[in] name unique name of the Tiler to update
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_TILER_RESULT
 */
DslReturnType dsl_tiler_source_show_all(const wchar_t* name);

```

closed 2020-08-29 by rjhowell44