prominenceai/deepstream-services-library #303: Deepstream 5.0 GA Upgrade

jlerasmus, 2020-08-05:
Deepstream 5.0 GA was released yesterday: https://forums.developer.nvidia.com/t/announcing-general-availability-for-deepstream-5-0/145366

Will probably be more of an effort to update JetPack and Deepstream than DSL.

rjhowell44, 2020-08-11:
@jlerasmus just fyi, my latest branch [refactor_ode_meter](https://github.com/canammex-tech/deepstream-services-library/tree/refactor_ode_meter) is building/passing with the 5.0 GA release.  There were just a few additional changes required from the dev pre-release.

jlerasmus, 2020-08-11:
@jlerasmus I will upgrade and have a test run, will provide feedback

rjhowell44, 2020-08-11:
@jlerasmus be aware, there are a couple of issues I'm still working through.  
1. the Meter PPH client callback is not getting passed into DSL correctly... the address is always 1.
2. I've discovered a problem with adding Display Meta to multiple frames within a single batched buffer... adding source names to each frame for example. Adding one name to one frame works, but as soon as I add a second, Deepstream seems to get confused as to which frame/source to add it to.  Perhaps I need to use a blocking pad probe, or I doing something else wrong. See [ode_always_trigger_display_source_info_1.py](https://github.com/canammex-tech/deepstream-services-library/blob/refactor_ode_meter/examples/python/ode_always_trigger_display_source_info_1.py) and [ode_always_trigger_display_source_info_2](https://github.com/canammex-tech/deepstream-services-library/blob/refactor_ode_meter/examples/python/ode_always_trigger_display_source_info_2.py).

rjhowell44, 2020-08-12:
@jlerasmus the changes required for the DeepStream 5.0 GA release have been added to the v0.07.alpha branch

Let me know if we can close this. Thanks

jlerasmus, 2020-08-12:
@rjhowell44 for some reason I could notnupgrade to JP4.4 in-place and therefore needed to reflash and setip everything again... this is done.

Next I will test and provide feedback.

jlerasmus, 2020-08-12:
@rjhowell44 I have not tested everything, but most seem good.

1. The example script 8uri_file_pph_meter_performace_reporting.py seems to be working fine.
2. `ode_always_trigger_display_source_info_1.py` and `ode_always_trigger_display_source_info_2.py` is still alternating between source names

I believe we can close this issue, as these are related to the source info actions.

closed 2020-08-12 by jlerasmus