prominenceai/deepstream-services-library #307: Refactor ODE Handler from Bintr to Handler

rjhowell44, 2020-08-07:
The ODE Handler should not be a bintr component in a fixed Pipeline/Branch location, but more of a pure batch-meta handler that can be added to the sink pads of Tiler's, Demuxer's and OSD components. 

Need to refactor the ODE Handler and other components to support this.

jlerasmus, 2020-08-07:
@rjhowell44 not related to the ODE handler, but should this not also be the case for the MeterSinkBintr? This would allow one to attach it as a batch meta handler without tee-ing of and creating a seperate sink just to measure

rjhowell44, 2020-08-08:
@jlerasmus yes, no doubt... I'll rework this issue to include both. Thanks

rjhowell44, 2020-08-08:
@jlerasmus here's what I'm proposing,.. 

In hindsight, going with the term **batch-meta-handler** was a mistake as there are other things to do with the buffer than read/modify metadata...  object and frame capture are examples. Using **pad-probe-handler** is more accurate

So while I'm refactoring...

Implement new DSL PPH API with three derived classes... for now
```C++
// callback typedef for the custom pad-probe-handler
typedef boolean (*dsl_pph_custom_client_handler_cb)(void* buffer, void* client_data);

// callback typedef for the meter pad-probe-handler
typedef boolean (*dsl_pph_meter_client_handler_cb)
    (double* session_fps_averages, double* interval_fps_averages, uint source_count, void* client_data);

// new Custom PPH
DslReturnType dsl_pph_custom_new(const wchar_t name,
    dsl_pph_custom_client_handler_cb client_handler, void* client_data);

// new Meter PPH
DslReturnType dsl_pph_meter_new(const wchar_t* name, uint interval,
    dsl_pph_meter_client_handler_cb client_handler, void* client_data);

// new ODE PPH
DslReturnType dsl_pph_ode_new(const wchar_t* name);
```
All existing  `*_batch_meta_handler_*` services will be updated, for example

```C++
DslReturnType dsl_osd_batch_meta_handler_add(const wchar_t* name, uint pad, 
    dsl_batch_meta_handler_cb handler, void* user_data);

DslReturnType dsl_osd_batch_meta_handler_remove(const wchar_t* name, 
    uint pad, dsl_batch_meta_handler_cb handler);
```
will become

```C++
DslReturnType dsl_osd_pph_add(const wchar_t* name, const wchar_t* handler, uint pad);

DslReturnType dsl_osd_pph_remove(const wchar_t* name, const wchar_t* handler, uint pad);
```
All exisitng `ode_handler` and `sink_meter` APIs will be mapped over.

jlerasmus, 2020-08-10:
@rjhowell44 this makes sense.

My only remark is that `client_data` should probably be passed as parameter when adding the pad-probe-handler

```c++
DslReturnType dsl_osd_pph_add(const wchar_t* name, const wchar_t* handler, uint pad, void* client_data);
```

rjhowell44, 2020-08-11:
@jlerasmus I see where you're going with the above, however not all PPH's will require client callbacks and client data... the ODE PPH is one example.  I'm hoping most won't

For the case of the Custom, Meter, and others that do... if you wanted to use a single client_callback function with different client data for uniqueness, you would need to do something like the following.

```Python
retval = dsl_pph_custom_new('my-custom-pph-1', my_single_callback, my_custom_data_1)
retval = dsl_pph_custom_new('my-custom-pph-2', my_single_callback, my_custom_data_2)
```
 
hopefully that's not to convoluted?

jlerasmus, 2020-08-11:
@rjhowell44 you are right, this should/will only apply to specific PPH's.

For the custom-handler cases, and only on multiple usage might it feel convulated, so I think it is ok.

rjhowell44, 2020-08-12:
@jlerasmus this work is done, the three PPH types can be moved around from component to component and sink pad to source pad... and in doing so, a few issues/questions have come up which I will cover over on the ODE Open Items thread. 

Closing this issue

closed 2020-08-12 by rjhowell44