prominenceai/deepstream-services-library #310: Implement ODE Action for Tiler to Show the Source on ODE Occurrence

rjhowell44, 2020-08-12:
Going to start out simple and just show the source id as identified by the frame meta that contains the object meta that triggered the ODE occurrence,  it will be up to the user to reset the view back to all sources in a key or button handler. 

```C++
DslReturnType dsl_ode_action_tiler_source_show_new(const wchar_t* name, const wchar_t* tiler);
```

but will most likely need to extend the behavior for more automated control.  One thought would be to specify a frame_count with the constructor... the number of frames before returning to all sources.  Or a mode parameter to specify if showing single or all sources.   

Considering the current python example, that shows how a pair of occurrence and absence triggers can be used to toggle behavior...  the same could be used to show a single source on first occurrence and then go back to all sources on first absence.

jlerasmus, 2020-08-12:
Sounds good.

Regarding the frame-count, I would rather consider having a time based window (e.g. 5 seconds) to support different frame rates. This timer should probably reset when the action is called, basically a sliding window to return to all sources after the last call.

The occurence and absence trigger pair should work in some cases, but might be a problem if source-1 triggers show-source on occurence, then source-2 triggers on occurence, but source-1 then resets to show-all on absence therefore showing all while the source-2 should still be shown. This should not be the case with a sliding window reset to show-all.  I hope this makes sense.

rjhowell44, 2020-08-12:
@jlerasmus good point, the occurrence-absence only works for actions that don't affect other sources in the batch.  You've sold me on the timer. Is units of seconds sufficient? 

RE: sliding window. 

Considering that a trigger can filter on a single source or all sources, I'm thinking this would work best as 
* using one trigger for all sources, once an occurrence is detected on one source. 
  * show the source, and start the timer. 
    * for each subsequent frame in the batch, and each batch of frames following
      * as long as the timer is running, no other source can update the show source setting. 
      * if a new occurrence is detected for the same source, 
        * restart the timer.
  * once the timer expires, any source can trigger a new occurrence.

jlerasmus, 2020-08-12:
@rjhowell44 glad to hear my sales hat did the job ;-)

Timeout unit in seconds is perfect. I don't expect a user to specify anything more than 10 to 15 seconds, and also not less than 1 second.

What you proposed sounds rigt. This should prevent  unnecessary switches while a source is shown.

Another thing to keep in mind is that such action's timer should also be reset when explicitly showing all sources (e.g. pressing escape or right clicking)

rjhowell44, 2020-08-12:
@jlerasmus actually, I'm a little off with my purposeful above. 

Each source will need it's own trigger with different criteria (Areas and such),  The Tiler will need to manage the timer and control over the show_source setting

So now that you sold me, you need to help answer some of the following :)

So  both the Tiler and Action APIs will change...
```C++
// a timeout of 0 will show the source indefinitely, or until show-source is called again.
DslReturnType dsl_tiler_source_show_get(const wchar_t* name, 
    const wchar_t** source, uint* timeout);

// if dsl_tiler_source_show_set() is called with source == the current show-source then the timeout will 
// be restarted, If called with a new source while the timer is runner, should the call be
//    - ignored? 
//    - or is the new source id queued and set on timeout?
// if called with show-all-sources (source = NULL)  while the timeout is running, does show-all-sources 
// take precedence? When calling to show-all-sources with a timeout, should the Tiller switch back to the
//  previous source value on timeout ? or is the timeout ignored?
DslReturnType dsl_tiler_source_show_set(const wchar_t* name, 
    const wchar_t* source, uint timeout);

DslReturnType dsl_ode_action_tiler_source_show_new(const wchar_t* name, 
    const wchar_t* tiler, uint timeout);
```

thanks

jlerasmus, 2020-08-13:
@rjhowell44 you are right.  We should always keep in mind that the same rules might apply to all sources, but also different rules will apply for each source (especially with areas)

Agreed that the best single place for timeout management would be the tiler instance being called on.

Here is what I think for an initial implementation:
- show-all-sources
  - should take precedence, cancel the timer and show all sources
  - I don't believe one would want to switch back from show-all after a timeout.  The general use case would be to place focus on the source with the event, and switch back to showing all.
  - to make the intention clear, maybe have a `dsl_tiler_source_show_all(name)`
- show-single-source should
  - if timeout == 0
    - show the single source
  - if timeout > 0
    - if timer is NOT running
      - show the source and start the timer
    - if timer is running
      - same source => ignore call and let timer runout to show all (depending on the triggers setup, this next trigger might show it again in any case
      - different source => switch to new source and (re)start timer
        - this might cause quick switching between sources, but should show the last source that triggered an action

rjhowell44, 2020-08-13:
@jlerasmus I read your mind.. that's exactly how I implemented it with `dsl_tiler_source_show_all(name)` 

I've merged everything into the v0.07.alpha branch. I've updated the existing show source example script... but I have not tried the new action yet... there is no script for this.  I will get to that tomorrow..

rjhowell44, 2020-08-13:
@jlerasmus I just re-read your notes and there is one thing that I've done differently

re: 
* if timer is running
  * same source => ignore call and let timer runout to show all (depending on the triggers setup, this next trigger might show it again in any case
    * different source => switch to new source and (re)start timer
      * this might cause quick switching between sources, but should show the last source that triggered an action

I have

* if timer is running
  * same source => extend the timer with new timeout and  runout to show all
  * different source ignore the call

If each source is using an occurrence trigger, the shared action will be calling `dsl_tiler_show_source` for each object which could lead to numerous switches per batch... could be a flickering image. 

May still need to iterate on this... lots to think about.

jlerasmus, 2020-08-13:
@rjhowell44 I think let's keep it as you have it now for an initial implementation. I will try and setup a complete example with my 9 cameras to see how it behaves in the real world (although this is a low activity scenario, might behave different in a high activity scenario)

As a future improvement, one solution might be to have an option passed in to allow user customization, e.g.
- SAME_SOURCE_COMPLETE
- SAME_SOURCE_EXTEND
- DIFFERENT_SOURCE_SWITCH
- DIFFERENT_SOURCE_IGNORE

closed 2020-09-01 by rjhowell44