prominenceai/deepstream-services-library #331: Capture Full Frame Action not saving correct source

jlerasmus, 2020-08-30:
@rjhowell44, I have modified your sample `ode_occurrence_rtsp_start_record_tap_action.py` to use my rtsp sources, and to also capture the frame by adding a capture-frame-action to the occurence trigger.

The saved frames (jpegs) is not from the correct source that triggered the action, although the filename generated is correct.  They mostly seem to be saved from the "same-other" source (e.g. src-4), but there are a few frames from other sources in between (might be that it is the first frame in the batch being saved).

The smart-recordings (videos) is correct though.  The order of the actions also does not make a difference.

```python
    # Create a new Capture Action to capture the full-frame to jpeg image, and save to file. 
    # The action will be triggered on firt occurrence of a bicycle and will be saved to the current dir.
    imageDir = './rtsp/images/'
    if not os.path.exists(imageDir):
        print(f'Creating directory {imageDir}')
        os.makedirs(imageDir)
    retval = dsl_ode_action_capture_frame_new(components.capture_frame, outdir=imageDir)
    if retval != DSL_RETURN_SUCCESS:
        return retval

    retval = dsl_ode_trigger_action_add_many(components.occurrence_trigger, 
        actions=[
            components.ode_notify,
            components.start_record,
            components.capture_frame,
            None
        ])
    if (retval != DSL_RETURN_SUCCESS):
        return retval`
```

rjhowell44, 2020-08-30:
@jlerasmus thanks, I'll look into this today

rjhowell44, 2020-08-30:
@jlerasmus please try with the v0.07.alpha updates as this should be fixed.

jlerasmus, 2020-08-31:
@rjhowell44, I gave this a try but unfortunately there seems to be an issue with the capture-frame-action as it causes an exception which then results in the PPH to be removed.  This error occurs with the latest changes with annotation true or false.

I have added a quick try-catch arround the action's HandleOccurence within the `OccurrenceOdeTrigger`, but I have no knowledge on how to print out a proper C++ stacktrace (in .NET this is quite easy :-)).

```c++
        for (const auto &imap: m_pOdeActions)
        {
            DSL_ODE_ACTION_PTR pOdeAction = std::dynamic_pointer_cast<OdeAction>(imap.second);
            try
            {
                LOG_INFO("Trigger: '" << GetName() << "' => Action: '" << pOdeAction->GetName() << "'");
                pOdeAction->HandleOccurrence(shared_from_this(), pBuffer, pDisplayMeta, pFrameMeta, pObjectMeta);
            }
            catch(...)
            {
                LOG_ERROR("Trigger: '" << GetName() << "' => Action: '" << pOdeAction->GetName() << "' threw exception");
            }
        }
        return true;
```

I do believe the error handling around the Triggers=>Actions could be improved to rather catch action exceptions and log errors rather than to remove the trigger "silently". This would enable continuation of the trigger, and execution of other actions in the list.

Let me know if there is anything I can do to help debug this.

rjhowell44, 2020-08-31:
Hey @jlerasmus clearly I'm still figuring this opencv stuff out... looking into the issue now.  

RE: *I do believe the error handling around the Triggers=>Actions could be improved*

Please feel free to open an issue for anything like this.  There will come a point either late alpha or early beta where the exceptions and log messages will be the focus of review/improvement.

jlerasmus, 2020-08-31:
@rjhowell44, I think I have got this fixed.  I need to double check and will do PR in a few minutes for you to review.

jlerasmus, 2020-08-31:
Created PR #336.  Please double check and let me know is something is not right

rjhowell44, 2020-08-31:
@jlerasmus thanks again for your help here... I'm thinking of releasing this today, unless there is anything else you're aware that needs to be addressed first?

jlerasmus, 2020-08-31:
@rjhowell44, this took me while to figure out using trial and error. I definitely need some pointers on how to log stack traces and/or debug the DSL lib code.

Either way, it seems to be working fine now, as well as the frame annotation. I will keep testing and see if something else comes up.

I think you can continue with the release.

closed 2020-09-01 by rjhowell44