prominenceai/deepstream-services-library #332: [Feature Request] Capture Frame Action to annotate image

jlerasmus, 2020-08-30:
An improvement to the capture-frame ode action (`dsl_ode_action_capture_frame_new`) would be to have an option to annotate the image with the bounding boxes.  As example, this [Deepstream Python Sample](https://github.com/NVIDIA-AI-IOT/deepstream_python_apps/blob/master/apps/deepstream-imagedata-multistream/deepstream_imagedata-multistream.py) is doing it.

E.g.
```python
retval = dsl_ode_action_capture_frame_new(components.capture_frame, outdir='./images', annotate=True)
    if retval != DSL_RETURN_SUCCESS:
        return retval
```

My understanding is that these handlers are executed synchronous, therefore it might be an option to add to an internal queue which is processed asynchronous.

rjhowell44, 2020-08-30:
@jlerasmus with the new Pad Probe Handler approach,  you can now create multiple ODE PPHs and add one down stream of the OSD  (OSD src pad for example) .  They can even be added to the sink pad of A Sink component which could be connected to a dumexer. 

That said, what I've discovered in the 5.0 GA release is that the Tiler is now zeroing out the frame number and source frame dimensions which is causing the frame-capture to fail.

See https://forums.developer.nvidia.com/t/deepstream-5-why-are-frame-num-width-and-height-values-always-0-after-tiler/146419

My current plan is to look into adding some user_meta to provide the values in the next release... but please feel free to drop nvidia a comment letting them know that their implementation is less that ideal :)

... and yes the image is currently captured synchronously. We can look into handling asynchronously if needed.

rjhowell44, 2020-08-30:
@jlerasmus I should have thought about this a little more before answering so quickly.  Having a second ODE PPH for your case doesn't make sense. And I'm liking you proposal above.

A couple of questions...

It's possible to  annotate just the object that triggered the event - for the case of object occurrence triggers, and all objects for summation, range and other frame-level triggers.  Would it be ok if I made this automatic?

And are default text and box colors sufficient?

rjhowell44, 2020-08-31:
@jlerasmus  the above has been implemented and is in the v0.07.alpha branch.  Feel free to play around with the fonts, sizes, etc. and send me a pull request if you find something that looks better. 

I'd like to release the v0.07.alpha if the frame_capture updates are working for you?  and then address anything new in the next release 

Let me know, thanks,
Robert.

jlerasmus, 2020-08-31:
@rjhowell44, thanks for this. I think this works as an initial implementation.

closed 2020-08-31 by jlerasmus