prominenceai/deepstream-services-library #339: Capture Frame Action - Segmentation Fault - 8 Cameras

jlerasmus, 2020-09-02:
@rjhowell44 I am still experiencing some issues with the capture-frame-action causing segmentation faults, but this seems to be only when I am running on 8 rtsp sources.  Anything less seems to be fine. 

There seem to be differences with the Deepstream dsexample plugin (sources/gst-plugins/gst-dsexample/gstdsexample.cpp)

I would appreciate any help on this.

rjhowell44, 2020-09-03:
@jlerasmus a couple of questions to help me in my investigation. 
* do your cameras run at the same frame rate? Other than the case of timeout, Is the batch size fixed at 8?
* do you set the Pipeline's batch_size and batch_timeout (streammux) attributes explicitly? 
* and have you tried setting a very large timeout? or to a value equal to or just greater than the frame rate?

jlerasmus, 2020-09-03:
@rjhowell44, to answer your questions

- My cameras are all running 1920x1080 @25FPS
- I do not set batch_size and batch_timeout explicitly
- I was double checking the default DSL_DEFAULT_STREAMMUX_BATCH_TIMEOUT values, which is currently set at 4000000.  My understanding is that the value should be specified in microseconds, therefore the defaults equates to a timeout of 4 seconds (https://www.unitconverters.net/time-converter.html)
- I have changed these defaults to 40000 (1/FPS = 1/25 * 1000 * 1000), but still get the segmentation fault. 

The Deepstream c++ samples uses the 40000 value in the config files, while the Deepstream Python samples uses 4000000... so I am confused.

Running the debugger from VSCode show the segmentation fault occuring at line 250 in DslOdeAction.cpp
```c++
err = NvBufSurfTransform (surface, dstSurface, &bufSurfTransform);
```
This happens on the first capture-frame execution, and happens every time when I configure more than 8 sources.

One thing that bothered me while trying to fix the issues was that the transform was being applied to all frames in the batch (if my understanding is correct), while I think it is only necessary to have a single frame for the destination surface.  Maybe the code from `get_converted_mat` from `sources/gst-plugins/gst-dsexample/gstdsexample.cpp` is of good reference, without the scaling.  

I could try implementing it, but I am sure that I will be messing up something with the points :-(

rjhowell44, 2020-09-03:
@jlerasmus re: *The Deepstream c++ samples uses the 40000 value in the config files, while the Deepstream Python samples uses 4000000... so I am confused*  

that makes two of us... but the default value in dslapi.h should be changed... in fact all defaults need to be reviewed, scrutinized before beta... 

re: *One thing that bothered me while trying to fix the issues was that the transform was being applied to all frames in the batch*

first, we should only be transforming the one frame based on the batch_id... 

```C
int surfaceIndex = pFrameMeta->batch_id;
```
I incorrectly had this as `source_id` which is wrong (and you fixed) as the sources in the batch can come in any order. I looked carefully through the dsexample and `get_converted_mat` and dsl's capture action code, and I see two issues that I'm looking into now... both related to 

```C
int retval = NvBufSurfaceCreate(&dstSurface, surface->batchSize,
     &bufSurfaceCreateParams);	
```

First, the return value is not checked for error before moving on and calling `NvBufSurfTransform` which is where the crash is happening.  Second, why is `surface->batchSize` used if only transforming the one frame? 

the fact that a batchSize of 7 works and 8 fails... I'm thinking that we've hit a threshold that is failing the `NvBufSurfaceCreate` and then an invalid `dstSurface` is passed to `NvBufSurfTransform` causing the fault.

rjhowell44, 2020-09-03:
@jlerasmus could you try fixing the batch size to 1 and try on your setup?

rjhowell44, 2020-09-04:
@jlerasmus I believe I have this fixed... will submit shortly.

jlerasmus, 2020-09-04:
@rjhowell44, sorry for the late reply.

> could you try fixing the batch size to 1 and try on your setup?

Setting `NvBufSurfaceCreate()` batchSize to 1 stops the segmentation fault, but results in a `-3` (NvBufSurfTransformError_Invalid_Params) error from `NvBufSurfTransform()`

But I believe I have also found and fixed the issue moments ago as I can now run 9 rtsp sources (by setting `dsl_source_num_in_use_max_set(9)`) successfully with frames being captured.  My fix lies here, but would like to compare with what you have found.  This is taken from the `gstdsexample.cpp`
```c++
        NvBufSurface ip_surf;
        ip_surf = *surface;
        ip_surf.numFilled = ip_surf.batchSize = 1;
        ip_surf.surfaceList = &(surface->surfaceList[surfaceIndex]);

        err = NvBufSurfTransform(&ip_surf, dstSurface, &bufSurfTransform);
```

In short, this is the working code.  In addition to the above, I have changed back the NvBufSurface calls to only initialize index 0, and changed the `in_mat ` creation to use index 0 (`dstSurface->surfaceList[0]`)
```c++
        int retval = NvBufSurfaceCreate(&dstSurface, 1, &bufSurfaceCreateParams);
        if (retval != 0)
        {
            g_print("NvBufSurfaceCreate failed with error %d\n", retval);
        }

        NvBufSurfTransformConfigParams bufSurfTransformConfigParams;
        NvBufSurfTransform_Error err;

        bufSurfTransformConfigParams.compute_mode = NvBufSurfTransformCompute_Default;
        bufSurfTransformConfigParams.gpu_id = surface->gpuId;
        bufSurfTransformConfigParams.cuda_stream = cudaStream;
        err = NvBufSurfTransformSetSessionParams(&bufSurfTransformConfigParams);

        NvBufSurfaceMemSet(dstSurface, 0, 0, 0);

        NvBufSurface ip_surf;
        ip_surf = *surface;
        ip_surf.numFilled = ip_surf.batchSize = 1;
        ip_surf.surfaceList = &(surface->surfaceList[surfaceIndex]);

        err = NvBufSurfTransform(&ip_surf, dstSurface, &bufSurfTransform);
        //err = NvBufSurfTransform(surface, dstSurface, &bufSurfTransform);
        if (err != NvBufSurfTransformError_Success)
        {
            g_print("NvBufSurfTransform failed with error %d while converting buffer\n", err);
        }

        NvBufSurfaceMap(dstSurface, 0, 0, NVBUF_MAP_READ);
        NvBufSurfaceSyncForCpu(dstSurface, 0, 0);

        cv::Mat bgr_frame = cv::Mat(cv::Size(bufSurfaceCreateParams.width,
                                             bufSurfaceCreateParams.height),
                                    CV_8UC3);

        cv::Mat in_mat = cv::Mat(bufSurfaceCreateParams.height,
                                 bufSurfaceCreateParams.width, CV_8UC4,
                                 dstSurface->surfaceList[0].mappedAddr.addr[0],
                                 dstSurface->surfaceList[0].pitch);

        cv::cvtColor(in_mat, bgr_frame, CV_RGBA2BGR);

        // Annotation code removed

        cv::imwrite(filespec.c_str(), bgr_frame);

        NvBufSurfaceUnMap(dstSurface, 0, 0);
        NvBufSurfaceDestroy(dstSurface);
        cudaStreamDestroy(cudaStream);
        gst_buffer_unmap(pBuffer, &inMapInfo);
```

Looking at the `gstdsexample.cpp`, I believe the error handling and cleanup of memory and cude streams should be improved though.

rjhowell44, 2020-09-04:
my fixes are essentially the same, but I've reworked everything for better error handling a cleanup..  trying to wrap it up now.. but getting late here...

jlerasmus, 2020-09-04:
No rush, I will let my fixes run and report back.

rjhowell44, 2020-09-04:
@jlerasmus are you setting a specific source name in your trigger create?

jlerasmus, 2020-09-04:
@rjhowell44 Yes, I am.  My test example basically the `ode_occurrence_rtsp_start_record_tap_action.py` sample modified with additional sources, and additional capture-frame and print actions

rjhowell44, 2020-09-05:
@jlerasmus I've pushed my changes for the capture issues into the v0.08.alpha... I've tested both frame and object capture with 4 streams... if you could confirm on your setup and let me know... 

Thanks
Robert.

jlerasmus, 2020-09-07:
@rjhowell44 this seems to be working fine with the latest v.0.08.alpha branch with both object and frame captures on 9 rtsp sources.  I will setup a long-running test and log a new issue if something pops up.

Thanks a lot for looking into this.

By the way, I like how you have refactored the code and with the cleanup in the destructors.

closed 2020-09-07 by jlerasmus