prominenceai/deepstream-services-library #350: Trigger actions not executed in the same order as added

jlerasmus, 2020-09-07:
ODE Actions added to a trigger are not executed in the same order as it was added, but seems to be executed alphabetically.

This is not the expected behavior according to the [ODE Trigger API documentation](https://github.com/canammex-tech/deepstream-services-library/blob/master/docs/api-ode-trigger.md)...

> ODE Triggers use settable criteria to parse the frame and detected-object metadata looking for occurrences of specific object-detection-events: Absence, Occurrence, Intersection, etc. Triggers, on ODE occurrence, **iterate through their collection of ordered ODE Actions invoking each**.

I have confirmed this by adding log statement to the `OccurrenceOdeTrigger::CheckForOccurrence` action loop, and got the following output:
```
DSL src/DslOdeTrigger.cpp:494:CheckForOccurrence:[00m  : Trigger 'src-2-occurrence-trigger' => Action 'src-2-capture-frame' => HandleOccurrence
DSL src/DslOdeTrigger.cpp:494:CheckForOccurrence:[00m  : Trigger 'src-2-occurrence-trigger' => Action 'src-2-capture-object' => HandleOccurrence
DSL src/DslOdeTrigger.cpp:494:CheckForOccurrence:[00m  : Trigger 'src-2-occurrence-trigger' => Action 'src-2-ode-notify' => HandleOccurrence
DSL src/DslOdeTrigger.cpp:494:CheckForOccurrence:[00m  : Trigger 'src-2-occurrence-trigger' => Action 'src-2-print' => HandleOccurrence
DSL src/DslOdeTrigger.cpp:494:CheckForOccurrence:[00m  : Trigger 'src-2-occurrence-trigger' => Action 'src-2-start-record' => HandleOccurrence
```

The actions was added in the following order:
```python
    retval = dsl_ode_trigger_action_add_many(components.occurrence_trigger,
                                             actions=[
                                                 components.print,
                                                 components.ode_notify,
                                                 components.start_record,
                                                 components.capture_frame,
                                                 components.capture_object,
                                                 None
                                             ])
```

I do believe they should be executed in the same order as added.

rjhowell44, 2020-09-07:
@jlerasmus all collections (implemented using C++ maps) are, currently, ordered by index which is their unique name.  To execute in the order of addition adds a little extra complexity, but can be done.

At a minimum... this need to be documented so that some could order them using a numeric prefix to each `'1-my-print'`, `'2-my-ode'`,.. which I really don't like.  

Let me take a look at this and see what I can do.

jlerasmus, 2020-09-07:
@rjhowell44, as long as this does not impact performance (One does not want sorting operations happening all the time)

Currently I have prefixed my actions to execute them in order.  The only need for this currently is for informational purposes and by printing info before and after actions, but this might be useful in the future if one action depends on the results of another.

jlerasmus, 2020-10-02:
@rjhowell44 as you mentioned all collections are ordered by key, resulting in this affecting the execution order of triggers as well with negative results when using `AbsenceOdeTrigger`.

I have run the `ode_occurrence_absence_interchange.py` with slight modifications to check for the bicycle instead of persons.  The modified sample has trigger names `new-bicycle-occurrence` and `new-bicycle-absence`.  In this case, with the absence trigger executing first, the sample clearly shows that the absence trigger is firing and resetting after each occurrence trigger, with continuous visual flashes (causing slowdown).

If I rename the triggers to `1-new-bicycle-occurence` and `2-new-bicycle-absence`, the sample works as expected.

This does bring up a question if certain triggers should be added in specific sequence, which for ease of use might be a problem.

rjhowell44, 2020-10-05:
@jlerasmus agreed. I will address this in the next release.

rjhowell44, 2021-03-17:
dependent on #402

rjhowell44, 2021-08-04:
Merged into v0.19.alpha. No docs updates are required.

closed 2021-08-04 by rjhowell44