prominenceai/deepstream-services-library #352: Error and segmentation fault on quitting application when rtsp-tap is added

jlerasmus, 2020-09-07:
Running sample `ode_occurrence_rtsp_start_record_tap_action.py` logs error and throws segmentation fault when pressing 'q' to exit application.

Error:
```
0:00:08.434763531  3639     0x12566460 ERROR                    DSL src/DslTapBintr.cpp:114:UnlinkFromSource:  : TapBintr 'src-0-record-tap' failed to unlink from Decode Source Tee
```

Debugger breaks at line 209 of `DslNodetr.h` when the segmentation fault occurs:
```
return GST_ELEMENT(m_pGstObj);
```

If the record-tap is not added with `dsl_source_rtsp_tap_add`, the error log and segementation fault does not occur.  The following is logged to the console though:
```
nvbuf_utils: dmabuf_fd 1306 mapped entry NOT found
```

rjhowell44, 2020-09-07:
Thanks @jlerasmus I will look into this. 

RE: *nvbuf_utils: dmabuf_fd 1306 mapped entry NOT found* 

This I'm aware of and have reported such to nvidia as this showed up with the 5.0 release.  They of course want me to reproduce using their app which I have not had time to follow up on as of yet.

jlerasmus, 2020-09-21:
@rjhowell44 I am still getting the segfault on the latest v0.08.alpha branch

```
0:00:07.297015085 11595      0xe3ee8f0 INFO                     DSL src/DslPipelineBintr.cpp:747:HandleXWindowEvents:  : Key released = 'q'
key released =  q
0:00:07.297340143 11595      0xe3ee8f0 INFO                     DSL src/DslServices.h:664:GetMainLoopHandle:  : Returning Handle to MainLoop
0:00:07.298293964 11595      0xe06ff60 INFO                     DSL src/DslBintr.h:223:SetState:  : Changing state to 'NULL' for Bintr 'pipeline'
0:00:07.298432665 11595      0xe06ff60 INFO                     DSL src/DslBintr.h:229:SetState:  : State change completed synchronously for Bintr'pipeline'
0:00:07.298548709 11595      0xe06ff60 INFO                     DSL src/DslPipelineSourcesBintr.cpp:230:UnlinkAll:  : Unlinking stream_muxer from src-0
0:00:07.298599960 11595      0xe06ff60 INFO                     DSL src/DslSourceBintr.cpp:144:UnlinkFromSink:  : Unlinking and releasing request Sink Pad for StreamMux stream_muxer
0:00:07.299076063 11595      0xe06ff60 WARN                     DSL src/DslSourceBintr.cpp:946:UnlinkAll:  : *********************************source
0:00:07.299201846 11595      0xe06ff60 INFO                     DSL src/DslTapBintr.cpp:109:UnlinkFromSource:  : Unlinking and releasing requested Source Pad for Decode Source Tee src-0-record-tap
0:00:07.299283671 11595      0xe06ff60 ERROR                    DSL src/DslTapBintr.cpp:114:UnlinkFromSource:  : TapBintr 'src-0-record-tap' failed to unlink from Decode Source Tee
Segmentation fault (core dumped)
```

jlerasmus, 2020-09-21:
@rjhowell44, I also got  a different error when quitting from terminal using Ctrl+C, but this occurs randomly (sorry, only had INFO logging enabled)
```
^C0:00:29.381972166  1106     0x2edb7210 WARN                     DSL src/DslSourceBintr.cpp:946:UnlinkAll:  : *********************************source
0:00:29.382190347  1106     0x2edb7210 ERROR                    DSL src/DslTapBintr.cpp:114:UnlinkFromSource:  : TapBintr 'src-1-record-tap' failed to unlink from Decode Source Tee
g_mutex_clear() called on uninitialised or locked mutex
Aborted (core dumped)
```

Here is the full Debug log from when 'q' is pressed:
```
0:00:06.554164941  2037      0x5475cf0 INFO                     DSL src/DslPipelineBintr.cpp:747:HandleXWindowEvents:  : Key released = 'q'
key released =  q
0:00:06.554435415  2037      0x5475cf0 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Services::GetMainLoopHandle()
0:00:06.554482655  2037      0x5475cf0 INFO                     DSL src/DslServices.h:664:GetMainLoopHandle:  : Returning Handle to MainLoop
0:00:06.554503801  2037      0x5475cf0 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Services::GetMainLoopHandle()
0:00:06.554596459  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Services::PipelineDeleteAll()
0:00:06.554667085  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::GstNodetr::RemoveAllChildren()
0:00:06.554755368  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Base::GetName()
0:00:06.554791879  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Base::GetName()
0:00:06.554831880  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Base::GetName()
0:00:06.554877975  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Base::GetName()
0:00:06.554911100  2037      0x50f5f60 DEBUG                    DSL src/DslNodetr.h:379:RemoveAllChildren:  : Removing Child GstNodetr'iou-tracker' from Parent GST BIn'pipeline'
0:00:06.554955997  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555050009  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555083551  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555101416  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555230533  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Base::GetName()
0:00:06.555253763  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Base::GetName()
0:00:06.555278034  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Base::GetName()
0:00:06.555295899  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Base::GetName()
0:00:06.555314649  2037      0x50f5f60 DEBUG                    DSL src/DslNodetr.h:379:RemoveAllChildren:  : Removing Child GstNodetr'on-screen-display' from Parent GST BIn'pipeline'
0:00:06.555340639  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555358973  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555383244  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555400692  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555507725  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Base::GetName()
0:00:06.555530382  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Base::GetName()
0:00:06.555553455  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Base::GetName()
0:00:06.555571112  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Base::GetName()
0:00:06.555590123  2037      0x50f5f60 DEBUG                    DSL src/DslNodetr.h:379:RemoveAllChildren:  : Removing Child GstNodetr'primary-gie' from Parent GST BIn'pipeline'
0:00:06.555614134  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555631842  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555654186  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555670749  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Nodetr::GetGstElement()
0:00:06.555751063  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Base::GetName()
0:00:06.555769761  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Base::GetName()
0:00:06.555792574  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Base::GetName()
0:00:06.555810127  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Base::GetName()
[... cut ...]
0:00:06.564437623  2037      0x50f5f60 DEBUG                    DSL src/DslNodetr.h:111:UnlinkFromSink:  : Unlinking Source 'src-depay' from Sink 'src-parse'
0:00:06.564471530  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::Nodetr::UnlinkFromSink()
0:00:06.564502676  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:75:~LogFunc: DSL::GstNodetr::UnlinkFromSink()
0:00:06.564544344  2037      0x50f5f60 DEBUG                    DSL src/DslLogGst.h:69:LogFunc: DSL::Nodetr::GetGstElement()
Segmentation fault (core dumped)
```

rjhowell44, 2020-09-23:
@jlerasmus  just fyi, I was finally able to get my live setup running again... should be able to fix the remaining issues tomorrow

rjhowell44, 2020-09-24:
@jlerasmus the segmentation fault issue has been resolved..

jlerasmus, 2020-09-25:
@rjhowell44 thanks, I will test and confirm

jlerasmus, 2020-09-28:
@rjhowell44, I can confirm that the segmentation fault is fixed.  I believe you can close the issue.

As a final note regarding the `nvbuf_utils` message, herewith my debug log:
```
  0:00:09.260938622 15019     0x2ddd4e30 INFO                     DSL src/DslPipelineBintr.cpp:747:HandleXWindowEvents:  : Key released = 'q'
  key released =  q
  0:00:09.261218835 15019     0x2ddd4e30 INFO                     DSL src/DslServices.h:664:GetMainLoopHandle:  : Returning Handle to MainLoop
  0:00:09.261435298 15019     0x2da56760 INFO                     DSL src/DslServices.cpp:7717:ReturnValueToString:  : Result = 0 = DSL_RESULT_SUCCESS
  DSL_RESULT_SUCCESS
  0:00:09.262414691 15019     0x2da56760 INFO                     DSL src/DslBintr.h:223:SetState:  : Changing state to 'NULL' for Bintr 'pipeline'
  0:00:09.262549798 15019     0x2da56760 INFO                     DSL src/DslBintr.h:229:SetState:  : State change completed synchronously for Bintr'pipeline'
  0:00:09.262663602 15019     0x2da56760 INFO                     DSL src/DslPipelineSourcesBintr.cpp:238:UnlinkAll:  : Unlinking stream_muxer from src-0
  0:00:09.262707666 15019     0x2da56760 INFO                     DSL src/DslNodetr.h:607:UnlinkFromSinkMuxer:  : Unlinking and releasing requested Sink Pad '0x2ddb3160' for Bintr 'src-0'
  0:00:09.272987496 15019     0x2ddd4d90 WARN                 rtspsrc gstrtspsrc.c:5999:gst_rtspsrc_try_send:<src-0> send interrupted
  0:00:09.273032340 15019     0x2ddd4d90 WARN                 rtspsrc gstrtspsrc.c:7673:gst_rtspsrc_close:<src-0> TEARDOWN interrupted
  0:00:09.275547492 15019     0x2da56760 INFO                     DSL src/DslNodetr.h:626:UnlinkFromSinkMuxer:  : Bintr 'src-0' changed state to NULL successfully
  0:00:09.276242974 15019     0x2da56760 INFO                     DSL src/DslSourceBintr.cpp:881:UnlinkAll:  : Stream management disabled for RTSP Source 'src-0'
  0:00:09.276436832 15019     0x2da56760 INFO                     DSL src/DslNodetr.h:547:UnlinkFromSourceTee:  : Unlinking and releasing requested Src Pad '0x2ddb8d00' for Bintr 'pre-decode-queue'
  0:00:09.276853506 15019     0x2da56760 INFO                     DSL src/DslTapBintr.cpp:71:UnlinkFromSourceTee:  : Unlinking and releasing requested Source Pad for TapBintr src-0-record-tap
  0:00:09.276958560 15019     0x2da56760 INFO                     DSL src/DslNodetr.h:547:UnlinkFromSourceTee:  : Unlinking and releasing requested Src Pad '0x2ddb8aa0' for Bintr 'src-0-record-tap'
  0:00:09.278355931 15019     0x2da56760 INFO                     DSL src/DslMultiComponentsBintr.cpp:193:UnlinkAll:  : Unlinking sink_bin_tee from window-sink
  0:00:09.278428692 15019     0x2da56760 INFO                     DSL src/DslNodetr.h:547:UnlinkFromSourceTee:  : Unlinking and releasing requested Src Pad '0x2ddb8f60' for Bintr 'window-sink'
  nvbuf_utils: dmabuf_fd 1306 mapped entry NOT found
  0:00:09.340796749 15019     0x2da56760 INFO                     DSL src/DslServices.cpp:6652:ComponentDeleteAll:  : All Components deleted successfully
  0:00:09.340959564 15019     0x2da56760 INFO                     DSL src/DslServices.cpp:3275:PphDeleteAll:  : All Pad Probe Handlers deleted successfully
  0:00:09.341097067 15019     0x2da56760 INFO                     DSL src/DslServices.cpp:2920:OdeTriggerDeleteAll:  : All ODE Triggers deleted successfully
  0:00:09.341151755 15019     0x2da56760 INFO                     DSL src/DslServices.cpp:1994:OdeAreaDeleteAll:  : All ODE Areas deleted successfully
  0:00:09.341264257 15019     0x2da56760 INFO                     DSL src/DslServices.cpp:1864:OdeActionDeleteAll:  : All ODE Actions deleted successfully
  0:00:09.341403739 15019     0x2da56760 INFO                     DSL src/DslServices.cpp:921:DisplayTypeDeleteAll:  : All Display Types deleted successfully
```

rjhowell44, 2020-09-28:
@jlerasmus Thanks for the log. I will try and chase this down with Nvidia over the coming weeks

closed 2020-09-28 by rjhowell44