prominenceai/deepstream-services-library #353: [Feature Request] RTSP Reconnect

jlerasmus, 2020-09-07:
The Deepstream samples implement RTSP reconnection functionality which from initial testing seems to work as expected.  It would be great if such functionality could be wrapped by DSL.

rjhowell44, 2020-09-07:
@jlerasmus yes, will definitely add.

rjhowell44, 2020-09-10:
@jlerasmus I'm working on this now... here are my current thoughts/plans for your review.

New parameter `reconnect_interval` added the the RTSP Source constructor
```C
/**
 * @brief creates a new, uniquely named RTSP Source component
 * @param[in] name Unique Resource Identifier (file or live)
 * @param[in] protocol one of the constant protocol values [ DSL_RTP_TCP | DSL_RTP_ALL ]
 * @param[in] cudadec_mem_type, use DSL_CUDADEC_MEMORY_TYPE_<type>
 * @param[in] intra_decode
 * @param[in] drop_frame_interval, set to 0 to decode every frame.
 * @param[in] latency in milliseconds
 * @param[in] reconnect_interval interval at which to a attempt a stream reconnect on the 
 * event a connection is lost. Set to 0 to disable.
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_SOURCE_RESULT otherwise.
 */
DslReturnType dsl_source_rtsp_new(const wchar_t* name, const wchar_t* uri, uint protocol,
    uint cudadec_mem_type, uint intra_decode, uint drop_frame_interval, uint latency, uint reconnect_interval);
```

And then four new Services to query and update a Source's reconnect attributes.
```C
/**
 * @brief Gets the current Reconnect Interval for the named RTSP Source
 * @param[in] name name of the source object to query
 * @param[in] reconnect_interval time between successive reconnect attempts in seconds.
 * A value of 0 indicates that reconnect management is disabled for the named Source
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_SOURCE_RESULT otherwise.
 */
DslReturnType dsl_source_rtsp_reconnect_interval_get(const wchar_t* name, uint* reconnect_interval);

/**
 * @brief Sets the current Reset Interval for the named RTSP Source
 * @param[in] name name of the source object to updated
 * @param[in] reconnect_interval time between successive reconnect attempts in seconds.
 * A value of 0 indicates that reconnect management is disabled for the named Source
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_SOURCE_RESULT otherwise.
 */
DslReturnType dsl_source_rtsp_reconnect_interval_set(const wchar_t* name, uint reconnect_interval);

/**
 * @brief Gets the current reconnect stats for the named RTSP Source
 * @param[in] name name of the source object to query
 * @param[out] last_time time of the last reconnect from system time in seconds (see timeval <sys/time.h>)
 * @param[out] last_count the count of reconnections for the named source, since first played or cleared.
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_SOURCE_RESULT otherwise.
 */
DslReturnType dsl_source_rtsp_reconnect_stats_get(const wchar_t* name, uint* last_time, uint* last_count); 

/**
 * @brief Clears the reconnect stats for the named RTSP Source
 * @param[in] name name of the source object to update
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_SOURCE_RESULT otherwise.
 */
DslReturnType dsl_source_rtsp_reconnect_stats_clear(const wchar_t* name); 

```

jlerasmus, 2020-09-10:
@rjhowell44 this is great news. Appreciate your commitment on issues/requests.

Your proposal seems fine. The only request I have is that there should be an option (maybe a callback/handler) to allow for the application be notified when
- the source goes down
- when reconnecting
- when reconnected

Maybe a struct for the stats can be defined and passed along.

This would allow the application to perform other actions e.g. notify the user, logging for analytics, showing of logo instead of blank screen, etc.

Maybe a general state/health handler (like ODE) is an option.

rjhowell44, 2020-09-10:
@jlerasmus I'm already on it... I'm in the process of including listener add/remove services now.  Also, now that I'm further through development, there are really two intervals 
1. the reconnect interval - interval between the last-buffer-time and current-time before deciding that the connecting is lost, and trying to reconnect  - nvidia defines this value in the config files.
2. the reset interval - interval between successive reset attempts while the connection remains down (nvidia has this hard-coded to 3s).

First, what's coming to mind as I write this, 1 and 2 might be better labeled as
1. buffer_timeout
2. reset_timeout

With this in mind... your above suggested notifications work... and would essentially be
* source goes down --  on-buffer-timeout
* when reconnecting -- on each reset-timeout
* when reconnected -- source returns to a state of playing

I'm now thinking of adding the reset_timeout default in dslapi.h and changing the input parameter to the RTSP source constructor to just 
```C
DslReturnType dsl_source_rtsp_new(const wchar_t* name, const wchar_t* uri, uint protocol,
    uint cudadec_mem_type, uint intra_decode, uint drop_frame_interval, uint latency, uint timeout);
```
let me know what you think

jlerasmus, 2020-09-10:
@rjhowell44 the 2 timeout values and notifications you have described makes total sense.

From an API usage perspective, I like recommended defaults but also prefer if they can be overridden. The `dsl_source_num_in_use_max_set` comes to mind. In this specific case it might make more sense to allow override per source (e.g. off-site cam might have more latency and connectivity issues), but global override should work.

rjhowell44, 2020-09-10:
@jlerasmus for the Get/Set services, I now have
```C
/**
 * @brief Gets the current Reconnect Interval for the named RTSP Source
 * @param[in] name name of the source object to query
 * @param[out] timeout current time to wait between successive frames before determining the 
 * connection is lost. If set to 0 then timeout is disabled.
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_SOURCE_RESULT otherwise.
 */
DslReturnType dsl_source_rtsp_timeout_get(const wchar_t* name, uint* timeout);

/**
 * @brief Sets the current Reset Interval for the named RTSP Source
 * @param[in] name name of the source object to updated
 * @param[in] timeout time to wait between successive frames before determining the 
 * connection is lost. Set to 0 to disable timeout.
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_SOURCE_RESULT otherwise.
 */
DslReturnType dsl_source_rtsp_timeout_set(const wchar_t* name, uint timeout);

/**
 * @brief Gets the current reconnect stats for the named RTSP Source
 * @param[in] name name of the source object to query
 * @param[out] last_time time of the last reconnect from system time in seconds (see timeval <sys/time.h>)
 * @param[out] last_count the count of reconnections for the named source, since first played or cleared.
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_SOURCE_RESULT otherwise.
 */
DslReturnType dsl_source_rtsp_reconnect_stats_get(const wchar_t* name, uint* last_time, uint* last_count); 

/**
 * @brief Clears the reconnect stats for the named RTSP Source
 * @param[in] name name of the source object to update
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_SOURCE_RESULT otherwise.
 */
DslReturnType dsl_source_rtsp_reconnect_stats_clear(const wchar_t* name); 

/**
 * @brief Gets the current reset stats for the named RTSP Source
 * @param[in] name name of the source object to query
 * @param[out] is_in_reset true if the source is currently in a reset cycle
 * @param[out] reset_count the count of reset attempts for the current reset cycle
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_SOURCE_RESULT otherwise.
 */
DslReturnType dsl_source_rtsp_reset_stats_get(const wchar_t* name, boolean* is_in_reset, uint* reset_count); 
```
For the client calback, I'm know thinking of matching the Pipeline state change listener
```C
DslReturnType dsl_pipeline_state_change_listener_add(const wchar_t* pipeline, 
    dsl_state_change_listener_cb listener, void* userdata);

DslREturnType dsl_source_rtsp_state_change_listener_add(const wchar_t* source,
    dsl_state_change_listener_cb listener, void* userdata);
```
The Callback would be the same for both, the one used for the Pipeline today
```C
typedef void (*dsl_state_change_listener_cb)(uint prev_state, uint curr_state, void* user_data);
```
and both share the same state values
```C
#define DSL_STATE_NULL                                              1
#define DSL_STATE_READY                                             2
#define DSL_STATE_PAUSED                                            3
#define DSL_STATE_PLAYING                                           4
#define DSL_STATE_IN_RESET                                          5
#define DSL_STATE_INVALID_STATE_VALUE                               UINT32_MAX
```
The client is free to call any of the APIs defined above in the call back to query or update the settings.  Setting the buffer timeout to 0 for example, would stop the source from trying to reconnect.  In theory, the user could even change the URI for the source.

jlerasmus, 2020-09-11:
@rjhowell44 sounds good.

To be clear on my understanding:
- `dsl_source_rtsp_reconnect_stats_get` will give you the amount of successfull reconnects from the begiing or reset
- `dsl_source_rtsp_reset_stats_get` will give you the amount of connection retry attempts in that reset cycle until reconnected. Calling this after successful reconnection should return (false, 0) right?

rjhowell44, 2020-09-11:
@jlerasmus yes, except for...

1. `dsl_source_rtsp_reconnect_stats_get` will give you the amount of successful reconnects from the beginning or from **clear reconnect stats**

rjhowell44, 2020-09-14:
@jlerasmus I'm now proposing [this](https://github.com/canammex-tech/deepstream-services-library/blob/rtsp_stream_mgt/docs/api-source.md).  

Please review and provide feedback when you can... thanks!

jlerasmus, 2020-09-14:
@rjhowell44, this looks good.

- `dsl_source_rtsp_reconnection_params_set` now provides the defaults and allow for individual source overrides.  Thanks for taking this into consideration.

- I agree with having a combined stats call `dsl_source_rtsp_reconnection_stats_get`.  Maybe the `last` parameter could be renamed to `last_time` or `last_reconnect_time` to be a bit more descriptive.

rjhowell44, 2020-09-16:
@jlerasmus I have this implemented with some basic unit and API test done.. I'll start on testing the behavior tomorrow.  The solution does add some complexity, requiring three additional timer driven threads.
1. to monitor and detect a loss of connect, and to start a reconnection cycle.
2. to manage the reconnection cycle.
3. to notify the RTSP state-change-listeners on change of state

The additional thread for the state-change-listener is required to allow the client to update the buffer timeout and reconnection params for 1. and 2. while they are currently running .
Any help in reviewing the code and testing this will be appreciated.   Everything is implemented in `DslSourceBintr.cpp` ... Lines 1369 through 1550

I've also created/implemented the following [issue](https://github.com/canammex-tech/deepstream-services-library/issues/358), checked into the same [branch](https://github.com/canammex-tech/deepstream-services-library/tree/rtsp_stream_mgt).  

Let me know if this seems/sufficient for now.

Thanks again, really appreciate your feedback and help.

jlerasmus, 2020-09-16:
@rjhowell44 thank you for adding this.  I will review and test with live sources and get back to you.

rjhowell44, 2020-09-17:
@jlerasmus just FYI, I did not get a change to work on this further today, so no testing yet... tomorrow for sure.

jlerasmus, 2020-09-17:
@rjhowell44 I did a few quick tests with an existing sample, but probably need to create a separate sample with all the handlers and stuff.  I also need to look throught the code still.

2 quick tests which did not work as expected.  Logs added from the moment the IP camera was rebooted (testing disconnect):
- *1 RTSP Stream, rebooted* - this is probably due to my sample quitting on EOS and not handling this properly
```
0:00:20.704055812  1641      0x9d77b20 ERROR                default gstrtspconnection.c:1004:gst_rtsp_connection_connect_with_response: failed to connect: Could not connect to 192.168.xxx.xxx: Connection refused
0:00:20.704103417  1641      0x9d77b20 ERROR                rtspsrc gstrtspsrc.c:4702:gst_rtsp_conninfo_connect:<src-3> Could not connect to server. (Generic error)
0:00:20.704266024  1641      0xa23a610 ERROR                    DSL src/DslPipelineBintr.cpp:804:HandleErrorMessage:  : Error message 'Could not open resource for reading and writing.' received from 'src-3'
0:00:20.704370817  1641      0xa23a610 ERROR                    DSL src/DslPipelineBintr.cpp:804:HandleErrorMessage:  : Error message 'Internal data stream error.' received from 'src-3'
[2020/09/17 11:07:54] Pipeline EOS event received, quitting application
0:00:20.784005603  1641      0xa23a610 WARN                     DSL src/DslPipelineSourcesBintr.cpp:239:UnlinkAll:  : *********************************unilink-all
0:00:20.784255086  1641      0xa23a610 WARN                     DSL src/DslSourceBintr.cpp:946:UnlinkAll:  : *********************************source
```

- 2 RTSP Streams, rebooting the 2nd
```
0:00:24.857276161 10451     0x1a5e79e0 ERROR                default gstrtspconnection.c:1004:gst_rtsp_connection_connect_with_response: failed to connect: Could not connect to 192.168.xxx.xxx: Connection refused
0:00:24.857335016 10451     0x1a5e79e0 ERROR                rtspsrc gstrtspsrc.c:4702:gst_rtsp_conninfo_connect:<src-3> Could not connect to server. (Generic error)
0:00:24.857594760 10451     0x1aa538c0 ERROR                    DSL src/DslPipelineBintr.cpp:804:HandleErrorMessage:  : Error message 'Could not open resource for reading and writing.' received from 'src-3'
0:00:24.857726794 10451     0x1aa538c0 ERROR                    DSL src/DslPipelineBintr.cpp:804:HandleErrorMessage:  : Error message 'Internal data stream error.' received from 'src-3'
0:01:24.878484633 10451     0x1aa538c0 ERROR                    DSL src/DslSourceBintr.cpp:1480:ReconnectionManager:  : FAILURE occured when trying to sync state for RTSP Source 'src-3'
0:01:24.878550832 10451     0x1aa538c0 ERROR                    DSL src/DslSourceBintr.cpp:1409:StreamManager:  : Failed to start reconnection cycle for source 'src-3'
0:01:24.882124957 10451   0x7ef000cde0 ERROR                default gstrtspconnection.c:1004:gst_rtsp_connection_connect_with_response: failed to connect: Could not connect to 192.168.xxx.xxx: Connection refused
0:01:24.882183708 10451   0x7ef000cde0 ERROR                rtspsrc gstrtspsrc.c:4702:gst_rtsp_conninfo_connect:<src-3> Could not connect to server. (Generic error)
0:01:24.882363972 10451     0x1aa538c0 ERROR                    DSL src/DslPipelineBintr.cpp:804:HandleErrorMessage:  : Error message 'Could not open resource for reading and writing.' received from 'src-3'
0:01:24.978854990 10451     0x1aa538c0 ERROR                    DSL src/DslSourceBintr.cpp:1480:ReconnectionManager:  : FAILURE occured when trying to sync state for RTSP Source 'src-3'
```

rjhowell44, 2020-09-17:
@jlerasmus I'll be working on this again at some point today... but just FYI
Special handling of the EOS event should not be required... To implement this feature (and those to come) I took the existing PadProbetr class, which was used for handling buffers, and derived two new classes of pad probe 
- `PadBufferProbetr`
- `PadEventDownStreamProbetr`

and then two new Pad Probe Event Handler types that can be added to the `PadEventDownStreamProbetr`, 
- `EosConsumerPadProbeEventHandler`
- `EosHandlerPadProbeEventHandler`

I'm just using them internally for now, but they are written such that they can be mapped to the DSL API just as the current ODE, Meter, and Custom  PPHs

I've update the `DslPipelineSourcesBintr.cpp` to add the EOS Consumer to the src pad of the streammux which should prevent EOS messages from getting to the bus-watcher which notifies the client EOS listeners

```C
        // If we're adding an RTSP Source, ensure EOS consumer is added to Streammuxer
        if (pChildSource->IsType(typeid(RtspSourceBintr)) and m_pEosConsumer == nullptr)
        {
            LOG_INFO("Adding EOS Consumer to Streammuxer 'src' pad on first RTSP Source");
            
            // Create the Pad Probe and EOS Consumer to drop the EOS event that occurs on 
            // loss of RTSP stream, allowing the Pipeline to continue to play. Each RTSP source 
            // will then manage their own restart attempts and time management.

            std::string eventHandlerName = GetName() + "-eos-consumer";
            m_pEosConsumer = DSL_PPEH_EOS_CONSUMER_NEW(eventHandlerName.c_str());

            std::string padProbeName = GetName() + "-src-pad-probe";
            m_pSrcPadProbe = DSL_PAD_EVENT_DOWNSTREAM_PROBE_NEW(padProbeName.c_str(), "src", m_pStreamMux);
            m_pSrcPadProbe->AddPadProbeHandler(m_pEosConsumer);
        }
```

rjhowell44, 2020-09-18:
@jlerasmus I fixed one key issue... with the EOS Consumer not working correctly,

jlerasmus, 2020-09-21:
@rjhowell44 some feedback on tests using the latest v0.08.alpha branch

- Having a single source with `timeout=0` drops the EOS event and application does not exit.  The EOS should probably not be added if reconnect is disabled.
  ```
  0:00:36.534600637  2321   0x7ef8004630 INFO                     DSL src/DslPadProbeHandler.cpp:487:HandlePadData:  : EOS Consumer -- dropping EOS Event
  ```
- Having a single source with `timeout=30`, I consistently get `ManageStream callback called before the connection has been establed for source 'src-0'` errors.  Any timeout value greater than 30 produces this randomly.  This is probably related to the time the initial connection is made.
  ```
  0:00:09.781041153  3167     0x1900ef60 INFO                     DSL src/DslPipelineBintr.cpp:680:HandleStateChanged:  : GST_STATE_PAUSED => GST_STATE_PLAYING
  previous state =  3 , new state =  4
  0:00:09.795432406  3167     0x1938e1e0 WARN          v4l2bufferpool gstv4l2bufferpool.c:1057:gst_v4l2_buffer_pool_start:<nvv4l2decoder0:pool:src> Uncertain or not enough buffers, enabling copy threshold
  0:00:09.797235199  3167     0x1900ef60 INFO                     DSL src/DslPipelineBintr.cpp:610:HandleBusWatchMessage:  : Message type:: GST_MESSAGE_STREAM_STATUS
  0:00:09.797344889  3167     0x1900ef60 INFO                     DSL src/DslBintr.h:211:GetState:  : Get state returned 'PLAYING' for Bintr 'src-0'
  0:00:09.797399108  3167     0x1900ef60 INFO                     DSL src/DslSourceBintr.cpp:1516:SetCurrentState:  : Changing state from READY to PLAYING for RtspSourceBintr 'src-0'
  0:00:09.797441088  3167     0x1900ef60 ERROR                    DSL src/DslSourceBintr.cpp:1392:StreamManager:  : ManageStream callback called before the connection has been establed for source 'src-0'
  0:00:09.797704270  3167     0x1900ef60 INFO                     DSL src/DslPipelineBintr.cpp:610:HandleBusWatchMessage:  : Message type:: GST_MESSAGE_STREAM_STATUS
  ```
  and in this case on quitting the following is logged
  ```
  0:00:07.389746829  4759     0x1e41a8f0 INFO                     DSL src/DslPipelineBintr.cpp:747:HandleXWindowEvents:  : Key released = 'q'
  key released =  q
  0:00:07.390052616  4759     0x1e41a8f0 INFO                     DSL src/DslServices.h:664:GetMainLoopHandle:  : Returning Handle to MainLoop
  0:00:07.391121177  4759     0x1e09bf60 INFO                     DSL src/DslBintr.h:223:SetState:  : Changing state to 'NULL' for Bintr 'pipeline'
  0:00:07.391276753  4759     0x1e09bf60 INFO                     DSL src/DslBintr.h:229:SetState:  : State change completed synchronously for Bintr'pipeline'
  0:00:07.391401183  4759     0x1e09bf60 INFO                     DSL src/DslPipelineSourcesBintr.cpp:230:UnlinkAll:  : Unlinking stream_muxer from src-0
  0:00:07.391454152  4759     0x1e09bf60 INFO                     DSL src/DslSourceBintr.cpp:144:UnlinkFromSink:  : Unlinking and releasing request Sink Pad for StreamMux stream_muxer
  0:00:07.392032913  4759     0x1e09bf60 WARN                     DSL src/DslSourceBintr.cpp:946:UnlinkAll:  : *********************************source

  (python3:4759): GLib-CRITICAL **: 13:54:30.856: Source ID 2 was not found when attempting to remove it
  0:00:07.392177291  4759     0x1e09bf60 INFO                     DSL src/DslSourceBintr.cpp:959:UnlinkAll:  : Stream management disabled for RTSP Source 'src-0'
  0:00:07.392315314  4759     0x1e09bf60 INFO                     DSL src/DslTapBintr.cpp:109:UnlinkFromSource:  : Unlinking and releasing requested Source Pad for Decode Source Tee src-0-record-tap
  0:00:07.392406253  4759     0x1e09bf60 ERROR                    DSL src/DslTapBintr.cpp:114:UnlinkFromSource:  : TapBintr 'src-0-record-tap' failed to unlink from Decode Source Tee
  Segmentation fault (core dumped)
  ```
- Having a single source with `timeout=60` where the above error does not occur, the reconnect also fails due to the camera not online yet, but no further attempts is performed.  Herewith the log details when the reconnect is tried after 60 seconds.  
  ```
  0:01:27.213821285  5963     0x163b2f60 INFO                     DSL src/DslBintr.h:211:GetState:  : Get state returned 'PLAYING' for Bintr 'src-0'
  0:01:27.213895922  5963     0x163b2f60 INFO                     DSL src/DslSourceBintr.cpp:1404:StreamManager:  : Buffer timeout of 60 seconds exceeded for source 'src-0'
  0:01:27.213921704  5963     0x163b2f60 INFO                     DSL src/DslSourceBintr.cpp:1438:ReconnectionManager:  : Resetting RTSP Source 'src-0' with reset count = 1
  0:01:27.213952017  5963     0x163b2f60 INFO                     DSL src/DslBintr.h:223:SetState:  : Changing state to 'NULL' for Bintr 'src-0'
  0:01:27.215243812  5963     0x16718320 WARN                 rtspsrc gstrtspsrc.c:5628:gst_rtspsrc_loop:<src-0> we are not connected
  0:01:27.234652197  5963     0x163b2f60 INFO                     DSL src/DslBintr.h:229:SetState:  : State change completed synchronously for Bintr'src-0'
  0:01:27.234718084  5963     0x163b2f60 INFO                     DSL src/DslSourceBintr.cpp:1516:SetCurrentState:  : Changing state from PLAYING to NULL for RtspSourceBintr 'src-0'
  0:01:27.236664996  5963     0x163b2f60 INFO                     DSL src/DslBintr.h:262:SyncStateWithParent:  : State change completed synchronously for Bintr'src-0'
[... cut ...]
  0:01:27.337792051  5963     0x163b2f60 INFO                     DSL src/DslBintr.h:211:GetState:  : Get state returned 'PAUSED' for Bintr 'src-0'
  0:01:27.338257477  5963     0x163b2f60 INFO                     DSL src/DslSourceBintr.cpp:1516:SetCurrentState:  : Changing state from PLAYING to PAUSED for RtspSourceBintr 'src-0'
  0:01:27.338369979  5963     0x163b2f60 ERROR                    DSL src/DslSourceBintr.cpp:1480:ReconnectionManager:  : FAILURE occured when trying to sync state for RTSP Source 'src-0'
  ```

Let me know if there is anything I can test or confirm for you.

rjhowell44, 2020-09-21:
@jlerasmus thanks for the detailed notes on this, especially since I've be struggling with my own setup. I just upgraded my internet service and equipment, and it's taking me a little time to get everything working.  ... bit I can see several obvious needed fixes in the logs above.

rjhowell44, 2020-09-22:
@jlerasmus updates have been pushed into v0.08.alpha branch that covers the obvious issues

jlerasmus, 2020-09-22:
- Single source with `timeout=0` now exists application
  ```
  0:00:37.245756868 28799     0x1675c400 WARN                 rtspsrc gstrtspsrc.c:5280:gst_rtspsrc_loop_udp:<src-1> warning: The server closed the connection.
  0:00:37.260800684 28799     0x1675c400 WARN                 rtspsrc gstrtspsrc.c:5293:gst_rtspsrc_loop_udp:<src-1> warning: Unhandled return value -7.
  0:00:37.260936520 28799     0x1675c400 WARN                 rtspsrc gstrtspsrc.c:5361:gst_rtspsrc_loop_udp:<src-1> error: Could not receive message. (System error)
  0:00:37.261019022 28799     0x1675c400 WARN                 rtspsrc gstrtspsrc.c:5653:gst_rtspsrc_loop:<src-1> error: Internal data stream error.
  0:00:37.261037720 28799     0x1675c400 WARN                 rtspsrc gstrtspsrc.c:5653:gst_rtspsrc_loop:<src-1> error: streaming stopped, reason error (-5)
  0:00:37.261234287 28799     0x16c05980 ERROR                    DSL src/DslPipelineBintr.cpp:885:HandleErrorMessage:  : Error message 'Could not read from resource.' received from 'src-1'
  0:00:37.261361425 28799     0x16c05980 ERROR                    DSL src/DslPipelineBintr.cpp:885:HandleErrorMessage:  : Error message 'Internal data stream error.' received from 'src-1'
  0:00:37.264428575 28799     0x1675bc00 WARN             matroskamux matroska-mux.c:3468:gst_matroska_mux_finish:0x16757520 unable to get final track duration
  0:00:37.345290018 28799   0x7f1c004e30 WARN             nvstreammux gstnvstreammux.c:2096:gst_nvstreammux_src_push_loop:<stream_muxer> warning: No Sources found at the input of muxer. Waiting for sources.
  0:00:37.345389811 28799     0x16c05980 WARN                     DSL src/DslSourceBintr.cpp:946:UnlinkAll:  : *********************************source
  0:00:37.346131598 28799     0x16c05980 ERROR                    DSL src/DslTapBintr.cpp:114:UnlinkFromSource:  : TapBintr 'src-1-record-tap' failed to unlink from Decode Source Tee
  ```

- Single source with `timeout=30`. On the first 2 retries camera was not up, but on third retry is was up but resulting in an application exit.
  ```
  0:00:17.324281763 28964     0x3d248000 WARN                 rtspsrc gstrtspsrc.c:5280:gst_rtspsrc_loop_udp:<src-1> warning: The server closed the connection.
  0:00:17.326119614 28964     0x3d248000 ERROR                default gstrtspconnection.c:1004:gst_rtsp_connection_connect_with_response: failed to connect: Could not connect to 192.168.xxx.xxx: Connection refused
  0:00:17.326167011 28964     0x3d248000 ERROR                rtspsrc gstrtspsrc.c:4702:gst_rtsp_conninfo_connect:<src-1> Could not connect to server. (Generic error)
  0:00:17.326199928 28964     0x3d248000 WARN                 rtspsrc gstrtspsrc.c:5348:gst_rtspsrc_loop_udp:<src-1> error: Could not connect to server. (Generic error)
  0:00:17.326314305 28964     0x3d248000 WARN                 rtspsrc gstrtspsrc.c:5653:gst_rtspsrc_loop:<src-1> error: Internal data stream error.
  0:00:17.326340243 28964     0x3d248000 WARN                 rtspsrc gstrtspsrc.c:5653:gst_rtspsrc_loop:<src-1> error: streaming stopped, reason error (-5)
  0:00:17.326516237 28964     0x3d6f1b20 ERROR                    DSL src/DslPipelineBintr.cpp:885:HandleErrorMessage:  : Error message 'Could not open resource for reading and writing.' received from 'src-1'
  0:00:17.326638948 28964     0x3d6f1b20 ERROR                    DSL src/DslPipelineBintr.cpp:885:HandleErrorMessage:  : Error message 'Internal data stream error.' received from 'src-1'
  0:00:17.328554769 28964     0x3d248050 WARN             matroskamux matroska-mux.c:3468:gst_matroska_mux_finish:0x3d243610 unable to get final track duration
  0:00:44.112456165 28964   0x7f1408a1e0 WARN                 rtspsrc gstrtspsrc.c:3155:on_timeout:<src-1> source 56263b99, stream 56263b99 in session 0 timed out
  0:00:47.331765331 28964     0x3d248000 WARN                 rtspsrc gstrtspsrc.c:5628:gst_rtspsrc_loop:<src-1> we are not connected
  0:00:47.345523794 28964     0x3d6f1b20 ERROR                    DSL src/DslSourceBintr.cpp:1410:StreamManager:  : Failed to start reconnection cycle for source 'src-1'
  0:00:47.447474717 28964   0x7ef4003cf0 ERROR                default gstrtspconnection.c:1004:gst_rtsp_connection_connect_with_response: failed to connect: Operation was cancelled
  0:00:47.447521749 28964   0x7ef4003cf0 ERROR                rtspsrc gstrtspsrc.c:4702:gst_rtsp_conninfo_connect:<src-1> Could not connect to server. (Generic error)
  0:00:47.447543416 28964   0x7ef4003cf0 WARN                 rtspsrc gstrtspsrc.c:7469:gst_rtspsrc_retrieve_sdp:<src-1> error: Failed to connect. (Generic error)
  0:00:47.447614928 28964   0x7ef4003cf0 WARN                 rtspsrc gstrtspsrc.c:7548:gst_rtspsrc_open:<src-1> can't get sdp
  0:00:47.450370454 28964     0x3d6f1b20 ERROR                    DSL src/DslPipelineBintr.cpp:885:HandleErrorMessage:  : Error message 'Could not open resource for reading and writing.' received from 'src-1'
  0:01:05.980375218 28964     0x3d248850 ERROR                default gstrtspconnection.c:1004:gst_rtsp_connection_connect_with_response: failed to connect: Could not connect to 192.168.xxx.xxx: No route to host
  0:01:05.980436053 28964     0x3d248850 ERROR                rtspsrc gstrtspsrc.c:4702:gst_rtsp_conninfo_connect:<src-1> Could not connect to server. (Generic error)
[... cut ...]
  (python3:28445): GStreamer-WARNING **: 07:03:09.978: Name 'src-depay' is not unique in bin 'src-1', not adding
  0:02:31.754637965 28445     0x1eb08050 ERROR                    DSL src/DslNodetr.h:333:AddChild:  : Failed to add src-depay to src-1'
  terminate called without an active exception
  ```

It seems that the retry mechanism is not started.  But I also think there might be an issue related to the `bufferTimeout` as this should probably only kick of the initial retry mechanism, and should not fire each and every time on `bufferTimeout` but I might be mistaken

rjhowell44, 2020-09-25:
@jlerasmus  I'm able to consistently disconnect and reconnect with a single camera. The defaults still need to be reviewed to see if they could be optimized.  

When you get a chance, if you could try this out on our setup, Id would sure appreciate it.

Also, I have not had a change to look at the client data issue with the record components yet.  Should get to it next...

jlerasmus, 2020-09-25:
@rjhowell44 thank for the update. I will test this again in the next day or two and provide feedback.

There is definitely an issue with the client data being passed. I did not have a chance to look deeper, but this is happening randomly (e.g. in my last run 3 of the sources had this on first occurence, the rest were fine... basically causing those 3 sources to stop as the trigger could not be reset).

jlerasmus, 2020-09-28:
@rjhowell44, I can confirm that the RTSP reconnection functionality works as expected.  Really appreciate your work on this.

I have tested the following cases

- Single source `with timeout=0` exists application without error, but with known `nvbuf_utils` message
- Single source with `timeout=10` continues to run and reconnects when camera is rebooted
- 9 sources with `timeout=10` in the following sequence
  - Disconnecting Nano's LAN cable, wait 60 seconds, reconnect LAN cable => all 9 sources reconnects
  - Rebooting each camera in sequence with a 20 seconds interval => all 9 sources reconnects
  - Disconnect Nano's LAN cable, wait 5 minutes, reconnect LAN cable => all 9 sources reconnects

Just some observation as reference for future functionality... (this is probably related to the streammuxer)
- When disconnecting the Nano's LAN cable, the last frame is shown until reconnect
- When rebooting the Nano, the tile goes black until reconnect

I prefer that the tile goes black to show the source is offline.  In the future it would be nice to add functionality to display some text or logo in the tile to show it is offline.

rjhowell44, 2020-09-28:
@jlerasmus re: *I prefer that the tile goes black to show the source is offline. In the future it would be nice to add functionality to display some text or logo in the tile to show it is offline.*

I will need to look into this as I'm not sure how best to handle... I don't know if it's the Tiler or the XWindow that is maintaining the image as there should be not data from the streammux.  I will try it with an overlay sink to see if the behavior is the same.  

Adding text to the Tile could be done by adding META data between the Tiler and OSD..  Just thinking out loud here, but this could be done by the client listening to the RTSP State, and enabling/disabling an "always" trigger to overlay some text.  

I'll add a feature request for this specifically, and we can look into.  I'm gong to close this issue, and anything else that comes up can be addressed separately.

closed 2020-09-28 by rjhowell44

rjhowell44, 2020-10-05:
@jlerasmus just FYI, I refactored and optimized the reconnection logic, which included changes to the API, see [RTSP Stream Connection Management](/docs/overview.md#rtsp-stream-connection-management)

ayanasser, 2022-10-23:
Hello @rjhowell44, 
Thank you for your contributions !
The RTSP stream connection management link is not connected, could you paste the working one ! 
thanks

rjhowell44, 2022-10-23:
Hi @ayanasser please see ... https://github.com/prominenceai/deepstream-services-library/blob/master/docs/overview.md#rtsp-stream-connection-management

ayanasser, 2022-10-23:
Thank you @rjhowell44 could you check this quick question please ? 
https://github.com/prominenceai/deepstream-services-library/issues/864

scknight2, 2023-01-31:
The RTSP stream connection management does not seem to handle the case where the RTSP stream is not yet available on application start. The pipeline fails to play. 

My Pipeline is setup to record from multiple sources on trigger, so if one of them is unavailable at start up, the whole pipeline fails to run.

rjhowell44, 2023-02-06:
@scknight2 my apologies for the slow response, but I wanted to do a little research before I commented. 

First, to answer your question.. ***Anyway to add this functionality?*** ... yes, I believe it can be done.  

There is a little history as to why the RTSP Source is implemented this way.  It has to do with a bug in nvidia's streammux plugin ... https://forums.developer.nvidia.com/t/releasing-nvstreammux-request-pad-results-in-a-deadlock/155271 

The workaround for this issue was to test the RTSP URI before linking the Source to the streammux and fail if the URI was down. 

My understanding is that nvidia's new streammux plugin https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_plugin_gst-nvstreammux2.html#gst-nvstreammux-new was created in part to fix this issue.

I just need time to integrate and update the connection logic. I'll add a new issue for this work. 

However, it may take me a couple of weeks to get to this.

Regards,
Robert.

scknight2, 2023-02-06:
@rjhowell44 Much appreciated!

scknight2, 2023-03-07:
Hi @rjhowell44 , sorry to bug, but any updates on this? I did not see any new issues created for it.

rjhowell44, 2023-03-08:
@scknight2 your timing is good... started working on it this week... will add the issue.

rjhowell44, 2023-03-10:
@scknight2 here's the new issue

-- *Extend RTSP stream-manager to manage first connection - retry on timeout #927*

This work has been completed and merged into the [v0.26.alpha](https://github.com/prominenceai/deepstream-services-library/tree/v0.26.alpha)

Please give it a try if you can and let me know how everything works for you.  There are breaking changes in this branch, so please review the running [release notes](https://github.com/prominenceai/deepstream-services-library/blob/v0.26.alpha/Release%20Notes/v0.26.alpha.md) carefully. 

This is the last issue I plan to cover for this branch which I plan to release next week. Any in-field testing you can do for me will be greatly appreciated.