prominenceai/deepstream-services-library #355: Capture Frame Action - Memory Leak

jlerasmus, 2020-09-08:
@rjhowell44, during my longer running tests I found that the capture-frame-action are causing memory leaks.  It seems that that the NVBufSurface stuff might not be cleaned up correctly.

Herewith screenshots showing `jtop` showing memory usage
- [Before running sample](https://snipboard.io/1gz0lc.jpg)
- [Running sample with capture-frame action DISABLED](https://snipboard.io/QCPb8U.jpg)
- [Running sample with capture-frame action ENABLED](https://snipboard.io/sxWBdD.jpg)

This can be reproduced by the following sample.  Note that the trigger limit is set to 230, which will cause sluggishness when maxing out memory. Setting the limit to 250 starts writing buffer related errors to the console.

```python
#!/usr/bin/env python

import sys
sys.path.insert(0, "../../")
from dsl import *
import os

uri_file = "../../test/streams/sample_1080p_h264.mp4"

# Filespecs for the Primary GIE
primary_infer_config_file = '../../test/configs/config_infer_primary_nano.txt'
primary_model_engine_file = '../../test/models/Primary_Detector_Nano/resnet10.caffemodel_b8_gpu0_fp16.engine'

PGIE_CLASS_ID_VEHICLE = 0
PGIE_CLASS_ID_BICYCLE = 1
PGIE_CLASS_ID_PERSON = 2
PGIE_CLASS_ID_ROADSIGN = 3

def xwindow_key_event_handler(key_string, client_data):
    """Function to be called on XWindow KeyRelease event"""

    print('key released = ', key_string)
    if key_string.upper() == 'P':
        dsl_pipeline_pause('pipeline')
    elif key_string.upper() == 'R':
        dsl_pipeline_play('pipeline')
    elif key_string.upper() == 'Q' or key_string == '' or key_string == '':
        dsl_main_loop_quit()

def xwindow_delete_event_handler(client_data):
    """Function to be called on XWindow Delete event"""

    print('delete window event')
    dsl_main_loop_quit()

def eos_event_listener(client_data):
    """Function to be called on End-of-Stream (EOS) event"""

    print('Pipeline EOS event')
    dsl_main_loop_quit()

def state_change_listener(old_state, new_state, client_data):
    """Function to be called on every change of Pipeline state"""

    print('previous state = ', old_state, ', new state = ', new_state)
    if new_state == DSL_STATE_PLAYING:
        dsl_pipeline_dump_to_dot('pipeline', "state-playing")

def main(args):

    # Since we're not using args, we can Let DSL initialize GST on first call
    while True:

        # New URI File Source using the filespec defined above
        retval = dsl_source_uri_new('uri-source', uri_file, False, 0, 0, 0)
        if retval != DSL_RETURN_SUCCESS:
            break

        # New Primary GIE using the filespecs above with interval = 0
        retval = dsl_gie_primary_new('primary-gie', primary_infer_config_file, primary_model_engine_file, 0)
        if retval != DSL_RETURN_SUCCESS:
            break

        # New KTL Tracker, setting max width and height of input frame
        retval = dsl_tracker_ktl_new('ktl-tracker', 480, 272)
        if retval != DSL_RETURN_SUCCESS:
            break

        # New Tiled Display, setting width and height, use default cols/rows set by source count
        retval = dsl_tiler_new('tiler', 1280, 720)
        if retval != DSL_RETURN_SUCCESS:
            break

         # Object Detection Event (ODE) Pad Probe Handler (PPH) to manage our ODE Triggers with their ODE Actions
        retval = dsl_pph_ode_new('ode-handler')
        if (retval != DSL_RETURN_SUCCESS):
            break

        # Add the ODE Pad Probe Hanlder to the Sink Pad of the Tiler
        retval = dsl_tiler_pph_add('tiler', 'ode-handler', DSL_PAD_SINK)
        if retval != DSL_RETURN_SUCCESS:
            break

        # Create occurrence trigger
        retval = dsl_ode_trigger_occurrence_new(
            'occurrence-trigger', source=DSL_ODE_ANY_SOURCE, class_id=PGIE_CLASS_ID_PERSON, limit=230)
        if (retval != DSL_RETURN_SUCCESS):
            break

        # Create capture-frame directory and action
        imageDir = './captures/'
        if not os.path.exists(imageDir):
            print(f'Creating directory {imageDir}')
            os.makedirs(imageDir)
        retval = dsl_ode_action_capture_frame_new('capture-frame', outdir=imageDir, annotate=True)
        if retval != DSL_RETURN_SUCCESS:
            break

        # Add the Actions to the trigger for this source.
        retval = dsl_ode_trigger_action_add('occurrence-trigger', action='capture-frame')
        if (retval != DSL_RETURN_SUCCESS):
            break

        # Add trigger to ODE handler
        retval = dsl_pph_ode_trigger_add('ode-handler', trigger='occurrence-trigger')
        if (retval != DSL_RETURN_SUCCESS):
            break

        # New OSD with clock enabled... .
        retval = dsl_osd_new('on-screen-display', True)
        if retval != DSL_RETURN_SUCCESS:
            break

        # New Window Sink, 0 x/y offsets and same dimensions as Tiled Display
        retval = dsl_sink_window_new('window-sink', 0, 0, 1280, 720)
        if retval != DSL_RETURN_SUCCESS:
            break

        # Add all the components to our pipeline
        retval = dsl_pipeline_new_component_add_many('pipeline',
                                                     ['uri-source', 'primary-gie', 'ktl-tracker', 'tiler', 'on-screen-display', 'window-sink', None])
        if retval != DSL_RETURN_SUCCESS:
            break

        # Add the XWindow event handler functions defined above
        retval = dsl_pipeline_xwindow_key_event_handler_add("pipeline", xwindow_key_event_handler, None)
        if retval != DSL_RETURN_SUCCESS:
            break
        retval = dsl_pipeline_xwindow_delete_event_handler_add("pipeline", xwindow_delete_event_handler, None)
        if retval != DSL_RETURN_SUCCESS:
            break

        # Add the listener callback functions defined above
        retval = dsl_pipeline_state_change_listener_add('pipeline', state_change_listener, None)
        if retval != DSL_RETURN_SUCCESS:
            break
        retval = dsl_pipeline_eos_listener_add('pipeline', eos_event_listener, None)
        if retval != DSL_RETURN_SUCCESS:
            break

        # Play the pipeline
        retval = dsl_pipeline_play('pipeline')
        if retval != DSL_RETURN_SUCCESS:
            break

        dsl_main_loop_run()
        retval = DSL_RETURN_SUCCESS
        break

    # Print out the final result
    print(dsl_return_value_to_string(retval))

    dsl_pipeline_delete_all()
    dsl_component_delete_all()

if __name__ == '__main__':
    sys.exit(main(sys.argv))
```

rjhowell44, 2020-09-08:
@jlerasmus really appreciate all the help... I'll look into this.

jlerasmus, 2020-09-08:
@rjhowell44, I believe this is fixed now.  Please double check.  Getting more up to speed with DSL :-)

I will start a new long running test, but with the above sample it seems fine.

rjhowell44, 2020-09-08:
@jlerasmus that's awesome!  I will check and pull in shortly... let me know when you're ready for some new issues to work on :) now that you're getting up to speed. 

Thanks again, really appreciate the help.

rjhowell44, 2020-09-10:
@jlerasmus can we close this issue now?

jlerasmus, 2020-09-10:
This seems to be fixed.

But I believe there is still one or more memory leak somewhere (not related to the capture-frame-action as far as I can tell) as I have had 2 crashes due to memory filling up, each within 8 hours of running.

closed 2020-09-10 by jlerasmus

rjhowell44, 2020-09-10:
@jlerasmus good to know... I would be very surprised at this point in development if there aren't at least a few issues to uncover.