prominenceai/deepstream-services-library #357: Map NvDsSRRecordingInfo to python class using CTypes (in dsl.py?)

rjhowell44, 2020-09-08:
Since Nvidia hasn't supported this, as of yet.. need to look into mapping the structure, most likely in dsl.py.  This will require a new python function to do the python object allocation and copy the data, which will need to be called by the python callback

jlerasmus, 2020-09-10:
May this https://forums.developer.nvidia.com/t/deepstream-5-0-python-bindings-for-gst-nvdsanalytics-access-meta-data/147670 and referenced repo might be of interest.

rjhowell44, 2020-09-20:
@jlerasmus there is a first implementation of this in the v0.08.alpha branch, and I've update one script so far. The data can be accessed in the callback as follows:
```Python
## 
# Function to be called on recording complete
## 
def record_complete_listener(session_info_ptr, client_data):
    print(' ***  Recording Complete  *** ')
    
    session_info = session_info_ptr.contents
    print('sessionId:     ', session_info.sessionId)
    print('filename:      ', session_info.filename)
    print('dirpath:       ', session_info.dirpath)
    print('duration:      ', session_info.duration)
    print('containerType: ', session_info.containerType)
    print('width:         ', session_info.width)
    print('height:        ', session_info.height)
    
    retval, is_on = dsl_sink_record_is_on_get('record-sink')
    print('        is_on flag = ', is_on)
    
    retval, reset_done = dsl_sink_record_reset_done_get('record-sink')
    print('        reset_done flag = ', reset_done)
    
    # reset the Trigger so that a new session can be started.
    print(dsl_return_value_to_string(dsl_ode_trigger_reset('bicycle-occurrence-trigger')))
    
    return None
```

I still need to look through the nvida docs to understand how the `gpointer` return value is used. Just returning a NULL pointer for now which seems to work.

jlerasmus, 2020-09-21:
@rjhowell44, thanks for adding this.  A few things I have noticed:

- The `ode_occurrence_url_start_record_sink_action.py` sample works as expected,  but the recorded filename only starts with an underscore. Herewith the listener output (the filename on disk matches the output)
  ```
  ***  Recording Complete  *** 
  sessionId:      0
  filename:       _00001_20200921-075651_6499.mkv
  dirpath:        ./
  duration:       4633
  containerType:  1
  width:          1920
  height:         1080
          is_on flag =  0
          reset_done flag =  1
  ```

- Running the `ode_occurrence_url_start_record_sink_action` (with handlers updated to print info, see below)
  - Creates filenames with gibberish prefixes (seems like an encoding issue)
    - `8õüw_00000_20200921-084058_7739.mkv`
    ```
    ***  Recording Complete  *** 
    sessionId:      0
    filename:       8õüw_00000_20200921-084058_7739.mkv
    dirpath:        ./recordings/
    duration:       44959
    containerType:  1
    width:          1920
    height:         1080
    ```
  - Does not seem to pass the client_data correctly to the `RecordingStarted` or `RecordCompleted` handlers.  I have had different cases, mostly occurring on first trigger or after a few only a few triggers:
    - The `components` is printed as `{}`, causing the reset to fail
    - It segfaults on the `print('components:', components)` 
    - It print a `PyObject is null` (have only had this once)

Herewith the updated handlers

```python
def RecordingStarted(event_id, trigger,
                     buffer, frame_meta, object_meta, client_data):
    print(' ***  Recording Started  *** ')
    print('trigger:', trigger)

    # cast the C void* client_data back to a py_object pointer and deref
    components = cast(client_data, POINTER(py_object)).contents.value
    print('client_data:', client_data)
    print('components:', components)
    print('Components:', str(vars(components)))

def RecordComplete(session_info_ptr, client_data):
    print(' ***  Recording Complete  *** ')

    session_info = session_info_ptr.contents
    print('sessionId:     ', session_info.sessionId)
    print('filename:      ', session_info.filename)
    print('dirpath:       ', session_info.dirpath)
    print('duration:      ', session_info.duration)
    print('containerType: ', session_info.containerType)
    print('width:         ', session_info.width)
    print('height:        ', session_info.height)

    components: ComponentNames = cast(client_data, POINTER(py_object)).contents.value
    print('Components:', str(vars(components)))

    retval, is_on = dsl_sink_record_is_on_get(components.record_tap)
    print('        is_on flag = ', is_on)

    retval, reset_done = dsl_sink_record_reset_done_get(components.record_tap)
    print('        reset_done flag = ', reset_done)

    # reset the Trigger so that a new session can be started.
    print(dsl_return_value_to_string(dsl_ode_trigger_reset(components.occurrence_trigger)))

    return None
```

rjhowell44, 2020-09-21:
@jlerasmus there was issue in the common RecordMgr class that was setting the file "prefix" to a stack pointer instead of a member variable... basically passing a bad pointer to the nvdssrr lib. This accounts for the invalid prefixes in both your examples.

jlerasmus, 2020-09-21:
@rjhowell44 thanks, I will try this tomorrow and provide feedback.

Any idea on this...

> Does not seem to pass the client_data correctly to the RecordingStarted or RecordCompleted handlers...

rjhowell44, 2020-09-21:
@jlerasmus  I will continue to test, but it could have been caused by the nvidia lib accessing the bad prefix pointer and corrupting the client data pointer

jlerasmus, 2020-09-22:
@rjhowell44 I can confirm that the filenames are fixed.

Regarding the `client_data`, it seems that I only get the "empty object" case where `components` is printed as `{}`. Sometimes the object has the attributes set and it works, but most of the time is does not.

I have tried without the `ode_notify` action (`dsl_ode_action_custom_new`) but there is no difference

rjhowell44, 2020-09-27:
@jlerasmus I found the issue with the `client_data`

It is due to the fact that the `components` object is declared as a stack variable in the CreatePerSourceComponents function
```Python
def CreatePerSourceComponents(pipeline, source, rtsp_uri, ode_handler):
   
    # New Component names based on unique source name
    components = ComponentNames(source)
    
``` 
... and that once the object is passed to the DSL lib in dsl.py and CreatePerSourceComponents returns, there is no longer a reference to the object and the python garbage collector eventually frees it. As a simple, interim solution. I'm just appending the client_data reference to an array in dsl.py that maintains the reference for the life of the application.  This was an issue with the callback pointers when first implemented, so using the same approach for both... for now.

```Python
##
## dsl_ode_action_custom_new()
##
_dsl.dsl_ode_action_custom_new.argtypes = [c_wchar_p, DSL_ODE_HANDLE_OCCURRENCE, c_void_p]
_dsl.dsl_ode_action_custom_new.restype = c_uint
def dsl_ode_action_custom_new(name, client_handler, client_data):
    global _dsl
    c_client_handler = DSL_ODE_HANDLE_OCCURRENCE(client_handler)
    callbacks.append(c_client_handler)   # persisting reference so that callback handler is not freed.
    c_client_data=cast(pointer(py_object(client_data)), c_void_p)
    clientdata.append(client_data)     # persisting reference so client_data is not freed
    result = _dsl.dsl_ode_action_custom_new(name, c_client_handler, c_client_data)
    return int(result)
```

Let me know if see any remaining issues.

jlerasmus, 2020-09-28:
@rjhowell44 thanks for the explanation, this makes sense.  I have done a few tests and the `client_data` is passed correctly, except for the following...

Using your `ode_occurrence_4rtsp_start_record_tap_action.py` (with adjusted sources), after some random time / number of events, I get a segfault in the `record_complete_listener` handler. This seems to occur on the lines that casts the `client_data` as the int value (client_data pointer) is printed, but no `components` object details is printed.
```
    print(' ***  Recording Complete  *** ')
    print(client_data)

    # cast the C void* client_data back to a py_object pointer and deref
    components = cast(client_data, POINTER(py_object)).contents.value
    print(components)
```

I have run the sample with debugger attached, showing where the segfault occurred, see: https://snipboard.io/70pFma.jpg Unfortunately I did not do a debug build to get line numbers, will do that next.

jlerasmus, 2020-09-28:
@rjhowell44 I did a debug build and ran the samples with GDB and got the following:

- When using the custom callback handler (`dsl_ode_action_custom_new`)
```
Thread 11 "python3" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7f80bc51f0 (LWP 24506)]
0x00000000005aea50 in PyObject_IsTrue ()
(gdb) bt
#0  0x00000000005aea50 in PyObject_IsTrue ()
#1  0x000000000052fd94 in _PyEval_EvalFrameDefault ()
#2  0x000000000052b108 in  ()
#3  0x000000000052b69c in  ()
#4  0x000000000052b8f4 in  ()
#5  0x00000000005306c0 in _PyEval_EvalFrameDefault ()
#6  0x000000000052b108 in  ()
#7  0x00000000005df500 in  ()
#8  0x0000000000608da8 in PyObject_Call ()
#9  0x0000007fb761d83c in  () at /usr/lib/python3.6/lib-dynload/_ctypes.cpython-36m-aarch64-linux-gnu.so
#10 0x0000007fb75c9a60 in  () at /usr/lib/aarch64-linux-gnu/libffi.so.6
#11 0x0000007fb75c8db4 in ffi_closure_SYSV () at /usr/lib/aarch64-linux-gnu/libffi.so.6
#12 0x0000007fb70bd1a4 in DSL::CustomOdeAction::HandleOccurrence(std::shared_ptr<DSL::Base>, _GstBuffer*, NvDsDisplayMeta*, _NvDsFrameMeta*, _NvDsObjectMeta*) (this=0x108cce0, pBase=
    std::shared_ptr<DSL::Base> (use count 5, weak count 1) = {...}, pBuffer=0x7f000159a0, pDisplayMeta=0x7ef43d92d0, pFrameMeta=0x7ef444b3a0, pObjectMeta=0x7ef4133b10)
    at src/DslOdeAction.cpp:83
#13 0x0000007fb70cd1a0 in DSL::OccurrenceOdeTrigger::CheckForOccurrence(_GstBuffer*, NvDsDisplayMeta*, _NvDsFrameMeta*, _NvDsObjectMeta*) (this=0x109e870, pBuffer=0x7f000159a0, pDisplayMeta=0x7ef43d92d0, pFrameMeta=0x7ef444b3a0, pObjectMeta=0x7ef4133b10) at src/DslOdeTrigger.cpp:494
#14 0x0000007fb70d3560 in DSL::OdePadProbeHandler::HandlePadData(_GstPadProbeInfo*) (this=0xfa0a00, pInfo=0x7f80bc4650) at src/DslPadProbeHandler.cpp:169
#15 0x0000007fb70d6cb8 in DSL::PadBufferProbetr::HandlePadProbe(_GstPad*, _GstPadProbeInfo*) (this=0xb5d800, pPad=0xf71420, pInfo=0x7f80bc4650) at src/DslPadProbeHandler.cpp:647
#16 0x0000007fb70d77e8 in DSL::PadProbeCB(GstPad*, GstPadProbeInfo*, gpointer) (pPad=0xf71420, pInfo=0x7f80bc4650, pPadProbetr=0xb5d800) at src/DslPadProbeHandler.cpp:717
#17 0x0000007fb5e5b4e8 in  () at /usr/lib/aarch64-linux-gnu/libgstreamer-1.0.so.0
#18 0x0000000000030101 in  ()
```

- When using the `record_complete` handler with the custom callback action removed, thus the first place accessing the `client_data`
```

Thread 204 "enc_que0:src" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7e9d9f91f0 (LWP 29759)]
0x00000000005a2f9c in _PyType_Lookup ()
(gdb) bt
#0  0x00000000005a2f9c in _PyType_Lookup ()
#1  0x00000000005a37ac in  ()
#2  0x0000000000608588 in PyObject_Format ()
#3  0x000000000052f518 in _PyEval_EvalFrameDefault ()
#4  0x000000000052b108 in  ()
#5  0x00000000005df500 in  ()
#6  0x0000000000608da8 in PyObject_Call ()
#7  0x0000007fb761d83c in  () at /usr/lib/python3.6/lib-dynload/_ctypes.cpython-36m-aarch64-linux-gnu.so
#8  0x0000007fb75c9a60 in  () at /usr/lib/aarch64-linux-gnu/libffi.so.6
#9  0x0000007fb75c8db4 in ffi_closure_SYSV () at /usr/lib/aarch64-linux-gnu/libffi.so.6
#10 0x0000007fb6ff0870 in DSL::RecordMgr::HandleRecordComplete(NvDsSRRecordingInfo*) (this=0xf961a0, pNvDsInfo=0x7f3c043bb0) at src/DslRecordMgr.cpp:298
#11 0x0000007fb6ff0ae4 in DSL::RecordCompleteCallback(NvDsSRRecordingInfo*, void*) (pNvDsInfo=0x7f3c043bb0, pRecordMgr=0xf961a0) at src/DslRecordMgr.cpp:312
#12 0x0000007fb60204b8 in  () at /opt/nvidia/deepstream/deepstream-5.0/lib/libnvdsgst_smartrecord.so
#13 0x0000007fb5f8794c in  () at /usr/lib/aarch64-linux-gnu/libglib-2.0.so.0
#14 0x0000007f4d5a07d8 in  ()
```

jlerasmus, 2020-09-28:
@rjhowell44 I believe I have found the persisting `client_data` issue with your interim implementation...

`c_client_data` should be persisted instead of `client_data`
```python
c_client_data=cast(pointer(py_object(client_data)), c_void_p)
clientdata.append(c_client_data)     # persisting reference so c_client_data is not freed
```

This fix seem to work fine (have only tested `dsl_ode_action_custom_new` and `dsl_ode_action_tap_record_start_new`)

rjhowell44, 2020-09-28:
@jlerasmus yes, clearly a mistake. Updates have been checked into v0.08.alpha, thanks.

And just FYI,  I'm planning on cutting a release with the current fixes and new functionality, unless you're aware of anything else that needs to be fixed now?

closed 2020-09-28 by rjhowell44