prominenceai/deepstream-services-library #369: Absence Trigger resetting on different source

jlerasmus, 2020-10-02:
It seems that the `AbsenceOdeTrigger` is resetting on a different source, therefore not correctly taking the source filter into account. 

I have not dived into the source too deeply, but this seems to apply to all trigger's PostProcessFrame handlers.

Also, on first run all absence triggers fires on frame 0.  This makes sense but is not really usable as it introduces a false-positive event.

jlerasmus, 2020-10-02:
@rjhowell44 herewith some print action output

Print action on occurrence trigger
```
Trigger Name    : src-8-1-trigger-occurrence
  Unique ODE Id : 284
  NTP Timestamp : 1601628752805959000
  Source Data   : ------------------------
    Inference   : No
    SourceId    : 7
    BatchId     : 3
    PadIndex    : 7
    Frame       : 24501
    Width       : 1920
    Heigh       : 1080
  Object Data   : ------------------------
    Class Id    : 2147483647
    Occurrences : 1
    Obj ClassId : 2
    Tracking Id : 5
    Label       : Person
    Confidence  : -0.1
    Left        : 596
    Top         : 504.265
    Width       : 280
    Height      : 341.471
  Min Criteria  : ------------------------
    Confidence  : 0
    Frame Count : 1 out of 1
    Width       : 0
    Height      : 0
    Inference   : No
```

Print action on absence trigger (absence trigger added after occurrence trigger)
```
Trigger Name    : src-8-2-trigger-absence
  Unique ODE Id : 285
  NTP Timestamp : 1601628752806050000
  Source Data   : ------------------------
    Inference   : No
    SourceId    : 0
    BatchId     : 4
    PadIndex    : 0
    Frame       : 24488
    Width       : 1920
    Heigh       : 1080
  Object Data   : ------------------------
    Class Id    : 2147483647
    Occurrences : 0
  Min Criteria  : ------------------------
    Confidence  : 0
    Frame Count : 1 out of 1
    Width       : 0
    Height      : 0
    Inference   : No
```

rjhowell44, 2020-10-05:
Thanks @jlerasmus,  this has been fixed.

closed 2020-10-05 by rjhowell44