prominenceai/deepstream-services-library #406: Implement new ODE Persistence Trigger

rjhowell44, 2021-03-23:
Implement a new Trigger to detect ODE occurrence on unique object persistence... meaning, when a unique object as identified by Tracking ID persists beyond a specified period of time. 

An example use case would be to detect loitering. People can pass through the frame without triggering and event, as long as they do so within a specified period of time.

```C
/**
 * @brief Persistence trigger that checks for the persistence of Objects tracked for a. 
 * specified source and object class_id.
 * @param[in] name unique name for the ODE Trigger
 * @param[in] source unique source name filter for the ODE Trigger, NULL = ANY_SOURCE
 * @param[in] class_id class id filter for this ODE Trigger
 * @param[in] limit limits the number of ODE occurrences, a value of 0 = NO limit
 * @param[in] max_persistence the maximum amount of time a unique object can remain detected 
 * before triggering and ODE occurrence - in units of seconds.
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_ODE_TRIGGER_RESULT otherwise.
 */
DslReturnType dsl_ode_trigger_occurrence_new(const wchar_t* name, 
    const wchar_t* source, uint class_id, uint limit, uint max_persistence);
```

rjhowell44, 2021-03-26:
Done.

closed 2021-03-26 by rjhowell44

zjd1988, 2022-07-26:
will support multi class id Persistence Trigger?

rjhowell44, 2022-07-26:
@zjd1988 You can use `DSL_ODE_ANY_CLASS` to check for persistence on all classes

zjd1988, 2022-07-26:
@[rjhowell44](https://github.com/rjhowell44)  thank you for your relpy.  I have a model output 7 class ,but I want to use 4 class persistence in the same polygon area

rjhowell44, 2022-07-26:
@zjd1988, currently, you would need to create 4 persistence triggers. You can then add the same polygon area and actions to each of the 4 triggers.   That said, I've been considering extending the ODE Trigger object to support a list of Class id's.  I'm going to bring this up on our [Discord](https://discord.gg/MJvY9jjpAK) site to discuss with others.  Please join us if you're interested.

zjd1988, 2022-07-27:
@rjhowell44 sounds greate！

rjhowell44, 2022-08-01:
@zjd1988 ...  I plan to add the above (Class id list) in a future release, but I can say as to when I can get to it.  Lots of competing priorities.  Hopefully you're ok creating four separate triggers for the time being... and again, the Trigger can share the same polygon area and actions.

zjd1988, 2022-08-02:
@rjhowell44 Ok, I will learn source codes to implement that by myself for the time being.

rjhowell44, 2022-08-02:
@zjd1988 I just want to make sure that you understand that creating 4 separate triggers and adding each of them to the ODE handler will give you the same behavior as what you're looking for?  And that you'll be updating a tremendous amount of DSL code just to save a couple of lines of code in your application.