prominenceai/deepstream-services-library #576: Implement new Triggers - "dsl_ode_trigger_newest_new" and "dsl_ode_trigger_oldest_new"

rjhowell44, 2021-08-07:
Implement two new Triggers two trigger on the "Newest" and "Oldest" objects within a frame.
```C
/**
 * @brief Newest Trigger that checks for the persistence of Objects tracked 
 * and will trigger on the Object with the least time of persistence (newest)
 * if at least one is found.
 * @param[in] name unique name for the ODE Trigger
 * @param[in] source unique source name filter for the ODE Trigger, NULL = ANY_SOURCE.
 * @param[in] class_id class id filter for this ODE Trigger.
 * @param[in] limit limits the number of ODE occurrences, a value of 0 = NO limit.
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_ODE_TRIGGER_RESULT otherwise.
 */
DslReturnType dsl_ode_trigger_newest_new(const wchar_t* name, 
    const wchar_t* source, uint class_id, uint limit);

/**
 * @brief Oldest Trigger that checks for the persistence of Objects tracked 
 * and will trigger on the Object with the greatest time of persistence (oldest) 
 * if at least one is found.
 * @param[in] name unique name for the ODE Trigger.
 * @param[in] source unique source name filter for the ODE Trigger, NULL = ANY_SOURCE.
 * @param[in] class_id class id filter for this ODE Trigger.
 * @param[in] limit limits the number of ODE occurrences, a value of 0 = NO limit
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_ODE_TRIGGER_RESULT otherwise.
 */
DslReturnType dsl_ode_trigger_oldest_new(const wchar_t* name, 
    const wchar_t* source, uint class_id, uint limit);
```

shuli163love, 2021-08-09:
Hi. The demo videos in youtube show the usage of DSL, and the IVA Studio is similar to a engine editor of the game engine, which is a nice tool. So, do you have a plan to open the source of the IVA Studio? Thanks.

rjhowell44, 2021-08-09:
Hi @shuli163love, perhaps in the future once DSL has been officially released, but not at this time.

rjhowell44, 2021-08-10:
Merged into v0.19.alpha

closed 2021-08-10 by rjhowell44