prominenceai/deepstream-services-library #585: C++ use

rbgreenway, 2021-08-12:
Hi guys!  
I've been using Deepstream for several months now, and stumbled across your library.  Wow, this looks good, could really clean up some of the stuff I've been doing. However, I've started trying to get a simple C++ example going and right off am having issues.  Just a simple player from an rtsp camera source.  BTW, I built your library (dsl-lib.so) and used cmake to link to my project...easy enough.

Here's my simple code (an attempted conversion from one of your python examples):

`#include <iostream>

#include "DslApi.h"

struct custom_data{
  int count;
  float value;
};

void key_handler_callback(wchar_t* key, void* user_data){
  std::cout<<"key released = "<< key << std::endl;
    char ch = std::toupper(static_cast<char>(*key));	
    if (ch == 'P'){	
        dsl_player_pause(L"player");
    } else if(ch=='R'){
        dsl_player_play(L"player");
    } else if(ch=='Q'){
      dsl_main_loop_quit();
    }    
}

int main(int argc, char** argv)
{

  custom_data my_data;
  my_data.count = 1;
  my_data.value = 0.25f;
 
  // build pipeline components
  wchar_t component_name[] = L"rtsp_src";
  wchar_t rtsp_uri[] = L"rtsp://admin:Password1!@10.0.0.37:554/cam/realmonitor?channel=1&subtype=1";
  DslReturnType ret = dsl_source_rtsp_new(component_name, rtsp_uri, DSL_RTP_ALL, DSL_CUDADEC_MEMTYPE_DEVICE, true, 0, 200, 0);
  if(ret != DSL_RESULT_SUCCESS) std::cout<<"Failure!"<<std::endl;
  
  ret = dsl_sink_window_new(L"mysink",0,0,640,480); // <-- this function causes error 
  if(ret != DSL_RESULT_SUCCESS) std::cout<<"Failure!"<<std::endl;
  
  ret = dsl_player_new(L"player",L"rtsp_src",L"mysink");

  ret = dsl_player_xwindow_key_event_handler_add(L"player",(dsl_xwindow_key_event_handler_cb)key_handler_callback,(void*)&my_data);

  ret = dsl_player_play(L"player");

  dsl_main_loop_run();

  dsl_delete_all();
  
  std::cout<<"Goodbye!"<<std::endl;

  return 0;
}`

Compiles fine, but when I try to run, I get the following error when I try to execute, I get:
`DSL src/DslElementr.h:64:Elementr:  : Failed to create new Element 'sink-bin-transform'`

Is there an obvious mistake on my part?  

Thanks for sharing this library, particularly for putting so much time into the documentation that you have.  Very impressive.

rbgreenway, 2021-08-12:
@rjhowell44 - Thanks for the fast response.  

After reading your response, I realized that I didn't say, and this is critical ... I'm running on a discrete GPU system (PC) rather than Jetson.  I run my code on both platforms, but do most of my dev on a PC.

Running inside a Deepstream 5.1 Docker container.

Deepstream 5.1
Cuda 11.1

```
# gst-inspect-1.0 nvegltransform
No such element or plugin 'nvegltransform'
```

This is a discrete GPU vs Jetson issue I'm guessing, and that likely means that your DSL, in it's current form, is not going to work for me...unless there are only minor adjustments to be made (as simple as the Makefile changes I made to get it to compile :-) ), but possibly not.  

If there are simple code edits to make that you can point me to, I'd love to try to make it work.  You have so much functionality built in that I'd like to have...sure will be a disappointment if I can't give it a go.

Thanks again.

rbgreenway, 2021-08-12:
@rjhowell44 Absolutely!!
I wish I were in a better position to help. And by that I mean that I wish I better understood your code so that I could help you make the changes.  Regardless, I have the time and desire to investigate/implement code changes if possible, and certainly the time to test any changes that you make on the hardware I have:  a system with a RTX Titan and another with a GeForce 1080Ti.  Just let me know how to proceed!  You can contact me directly at rbgreenway@gmail.com, if that is more appropriate.

rjhowell44, 2021-08-12:
@rbgreenway please join the DSL [discord server](https://discord.gg/MJvY9jjpAK) and I'll create a channel for us to chat on.  I'm in the process of releasing v0.19.alpha which should be out this afternoon.  Once that's done I'll create a new branch to work on. 

Most of the changes will be in `DslSourceBintr.cpp` and `DslSinkBintr.cpp` ... and we'll be using  

/opt/nvidia/deepstream/deepstream-5.1/sources/apps/apps-common/src/deepstream_source_bin.c /opt/nvidia/deepstream/deepstream-5.1/sources/apps/apps-common/src/deepstream_sink_bin.c 

as our guide to the conditional logic.

closed 2021-08-12 by rjhowell44