prominenceai/deepstream-services-library #675: Add callback to dsl_ode_action_capture_frame_new

rbgreenway, 2021-12-20:
It would be nice to have a configurable callback for dsl_ode_action_capture_frame_new in order to have a way to know the path/filename of the newly saved image file.  This is necessary if you want to be able to keep a record of the files saved.

rbgreenway, 2021-12-20:
Wait....just found dsl_ode_action_capture_complete_listener_add.  I'm guessing that this is what this is for!  :-)
...testing...

rbgreenway, 2021-12-21:
dsl_ode_action_capture_complete_listener_add appears to be undocumented, but it works perfectly!

closed 2021-12-21 by rbgreenway

rjhowell44, 2021-12-21:
@rbgreenway please see 
* [dsl_capture_info](/docs/api-ode-action.md#dsl_capture_info)
* [dsl_capture_complete_listener_cb](/docs/api-ode-action.md#dsl_capture_complete_listener_cb)
* [dsl_ode_action_capture_complete_listener_add](/docs/api-ode-action.md#dsl_ode_action_capture_complete_listener_add)

rbgreenway, 2021-12-21:
@rjhowell44 Thanks.  Just totally missed the documentation.