prominenceai/deepstream-services-library #751: python breakpoint not work in osd_sink_pad_buffer_probe handler

52doho, 2022-05-06:
Hi @rjhowell44 thank you for your great project, it makes deepstream api really simple!

I try all the python examples, only the osd_sink_pad_buffer_probe handler is not able to debug in vs code. I have install python and c++ extension for vs code.

I also google and search in [nvidia](https://forums.developer.nvidia.com/search?q=debug%20osd_sink_pad_buffer_probe) without getting any solution. How to debug [osd_sink_pad_buffer_probe](https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/nvidia_osd_sink_pad_buffer_probe.py#L35) handler in python?

Many thanks.

rjhowell44, 2022-05-07:
Hi @52doho , which script are you running when you are trying to debug the [osd_sink_pad_buffer_probe.py](https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/nvidia_osd_sink_pad_buffer_probe.py#L35) 

I'm sure you're aware of this, but the file is a python module that is included/used by other example scripts. 
See [1uri_file_pgie_iou_tiler_osd_custom_pph_window.py](https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/1uri_file_pgie_iou_tiler_osd_custom_pph_window.py) as one example. 

Are you saying that you are executing one of the scripts that includes the module, but when you set a break point in the module (in VS Code), the execution does not break?

If so... my apologies, but I'm not a VS Code user.  I use Codelite on my Jetson devices and I'm developing in C++.  Perhaps it's a debugger setting, or the way I've defined/included the module? I have not tried debugging this NVIDIA module myself. 

I'm going to bring this up with others on our [Discord](https://discord.gg/MJvY9jjpAK) server. There are a couple of VS Code/Python experts that might be able to help. You're more than welcome to come join us. 

Best regards,
Robert.

52doho, 2022-05-12:
Hi @rjhowell44, sorry for late response.

[1uri_file_pgie_iou_tiler_osd_custom_pph_window.py](https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/1uri_file_pgie_iou_tiler_osd_custom_pph_window.py) calls osd_sink_pad_buffer_probe handler in [osd_sink_pad_buffer_probe.py](https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/nvidia_osd_sink_pad_buffer_probe.py#L35) which is never break in debug.

Please refer these steps to reproduce.
1. breakpoint on pipeline, it works
[image]

2. breakpoint on state_change_listener, it works
[image]

3. breakpoint on osd_sink_pad_buffer_probe, it is not work, breakpoint never stop
[image]

By the way, breakpoint on osd_sink_pad_buffer_probe can stop in c++ debug.
[image]

rjhowell44, 2022-05-12:
@52doho I've brought your question up on our Discord Server... I'll let you know if I get a response. Unfortunately, I don't have the VSCode experience to be able to help.

rjhowell44, 2022-05-13:
@52doho In the meantime, you can always copy the callback function into the script you're running.

52doho, 2022-05-16:
Great! Build apps on python DSL api is faster than cpp api. It will help the whole community. Thanks for your great work!

niderhoff, 2022-05-16:
Breakpoints in python probes never work in VS Code in my experience.
What did work was setting a breakpoint using pdb.

52doho, 2022-05-17:
@niderhoff You are right. after add breakpoint() to osd_sink_pad_buffer_probe handler, vscode python debugger can pause now. 

Thank you @niderhoff
[image]

closed 2022-05-17 by 52doho

rjhowell44, 2022-05-17:
@niderhoff thank you so much for helping out here. Very much appreciated.  I'm going to add a note and link to the VSCode docs page.

... and please consider joining us on our [Discord](https://discord.gg/MJvY9jjpAK) server were we're discussing new features and DSL's roadmap. Insights and new ideas are always welcome.