prominenceai/deepstream-services-library #786: When adding after component remove, the tile view becomes awkward

youngjae-avikus, 2022-06-10:
https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/dyn_uri_file_pgie_ktl_tiler_osd_window.py

If press the key ++++ and then -- and + + , dsl do not get the number of sources in osd. Is it awkward?

[image]

youngjae-avikus, 2022-06-10:
There is no error when retval testing is performed

```python
def xwindow_key_event_handler(key_string, client_data):
    global MAX_SOURCE_COUNT, cur_source_count
    print('key released = ', key_string)
    if key_string.upper() == 'P':
        dsl_pipeline_pause('pipeline')
    elif key_string.upper() == 'R':
        dsl_pipeline_play('pipeline')
    elif key_string.upper() == 'Q' or key_string == '' or key_string == '':
        dsl_main_loop_quit()

    # Add a new source
    elif key_string == '+': 
        if cur_source_count < MAX_SOURCE_COUNT:
            cur_source_count += 1
            source_name = 'source-' + str(cur_source_count)
            print('adding source ', source_name)
            retval = dsl_source_file_new(source_name, uri_h265, True)
            if retval != DSL_RETURN_SUCCESS:
                dsl_main_loop_quit()
            retval = dsl_pipeline_component_add('pipeline', source_name)
            if retval != DSL_RETURN_SUCCESS:
                dsl_main_loop_quit()

    # Remove the last source added
    elif key_string == '-': 
        if cur_source_count > 1:
            source_name = 'source-' + str(cur_source_count)
            print('removing source ', source_name)
            retval = dsl_pipeline_component_remove('pipeline', source_name)
            if retval != DSL_RETURN_SUCCESS:
                dsl_main_loop_quit()
            retval = dsl_component_delete(source_name)
            if retval != DSL_RETURN_SUCCESS:
                dsl_main_loop_quit()
            cur_source_count -= 1
```

rjhowell44, 2022-06-13:
@youngjae-avikus , the example is incomplete... it should be calling [dsl_tiler_tiles_set](/docs/api-tiler.md#dsl_tiler_tiles_set) on `-` key.  See #789

closed 2023-08-17 by rjhowell44