prominenceai/deepstream-services-library #795: Example cpp code for storing mot challenge format for mota, motp measurement...

youngjae-avikus, 2022-06-15:
Hello!
In order to tune the appropriate value for tracker.yaml in the future, it will be saved in the format suggested by mot challenge
Through the monitor_action you developed, the following values are stored in order, and temporarily saved in 'pred.txt'
What do you think about these tasks?

```txt
<frame>, <id>, <bb_left>, <bb_top>, <bb_width>, <bb_height>, <conf>, <x>, <y>, <z>
```

```cpp
/*
The MIT License

Copyright (c) 2021, Prominence AI, Inc.

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, an/opt/prominenceai/deepstream-services-library/examples/cpp/1rtsp_yolov5_dcf_osd_window.cppd/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in-
all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
THE SOFTWARE.
*/

/*
https://github.com/JonathonLuiten/TrackEval/blob/master/docs/MOTChallenge-format.txt

File Format

Please submit your results as a single .zip file. The results for each sequence must be stored in a separate .txt file in the archive's root folder. The file name must be exactly like the sequence name (case sensitive).

The file format should be the same as the ground truth file, which is a CSV text-file containing one object instance per line. Each line must contain 10 values:

<frame>, <id>, <bb_left>, <bb_top>, <bb_width>, <bb_height>, <conf>, <x>, <y>, <z>
The conf value contains the detection confidence in the det.txt files. For the ground truth, it acts as a flag whether the entry is to be considered. A value of 0 means that this particular instance is ignored in the evaluation, while any other value can be used to mark it as active. For submitted results, all lines in the .txt file are considered. The world coordinates x,y,z are ignored for the 2D challenge and can be filled with -1. Similarly, the bounding boxes are ignored for the 3D challenge. However, each line is still required to contain 10 values.

All frame numbers, target IDs and bounding boxes are 1-based. Here is an example:

Tracking with bounding boxes
(MOT15, MOT16, MOT17, MOT20)
  1, 3, 794.27, 247.59, 71.245, 174.88, -1, -1, -1, -1
  1, 6, 1648.1, 119.61, 66.504, 163.24, -1, -1, -1, -1
  1, 8, 875.49, 399.98, 95.303, 233.93, -1, -1, -1, -1
  ...
*/

#include <iostream>
#include <glib.h>
#include <vector>
#include <fstream>

#include "DslApi.h"

static const std::wstring primary_infer_config_file(L"/opt/nvidia/deepstream/deepstream/samples/configs/deepstream-app/config_infer_primary_nano.txt");

static const std::wstring primary_model_engine_file(L"/opt/nvidia/deepstream/deepstream/samples/models/Primary_Detector_Nano/resnet10.caffemodel_b8_gpu0_fp16.engine");

static const std::wstring tracker_config_file(L"/opt/nvidia/deepstream/deepstream/samples/configs/deepstream-app/config_tracker_NvDCF_max_perf.yml");

int tile_width = 1280;
[... 323 lines cut ...]
            eos_event_listener, nullptr);
        if (retval != DSL_RESULT_SUCCESS) break;

        // # Play the pipeline
        retval = dsl_pipeline_play(L"pipeline");
        if (retval != DSL_RESULT_SUCCESS) break;

        // # Join with main loop until released - blocking call
        dsl_main_loop_run();

        f.close();
        retval = DSL_RESULT_SUCCESS;
        break;
    }

    // # Print out the final result
    std::wcout << dsl_return_value_to_string(retval) << std::endl;

    dsl_pipeline_delete_all();
    dsl_component_delete_all();

    std::cout<<"Goodbye!"<<std::endl;  
    return 0;
}
```

rjhowell44, 2022-06-16:
@youngjae-avikus if the MOTChallenge-format is this important, I think we should consider adding the support directly to DSL. 

One thought would be to update the [ODE File Action](/docs/api-ode-action.md#dsl_ode_action_file_new)  so that it supports three formats; TEXT, CSV, and MOTC .  Or create a new action specifically for this.

rjhowell44, 2022-06-21:
Will be added to DSL directly with #799