prominenceai/deepstream-services-library #803: Refactor and update Infer API and InferBintr to support new Preprocessor Component

rjhowell44, 2022-06-29:
Need to add services to get/set the `input-tensor-meta` attribute for the GIE InferBintr - Primary only, classification is still to be supported by NVIDIA. Going to combine with the `output-tensor-meta` attribute, i.e both will be get/set together. 
```C
/**
 * @brief Gets the current input and output tensor-meta settings in use by the 
 * named Primary or Secondary GIE.
 * @param[in] name unique name of Primary or Secondary GIE to query.
 * @param[out] input_enabled if true preprocessing input tensors attached as 
 * metadata instead of preprocessing inside the plugin, false otherwise.
 * @param[out] output_enabled if true tensor outputs will be attached as 
 * meta on GstBuffer.
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_INFER_RESULT otherwise.
 */
DslReturnType dsl_infer_gie_tensor_meta_settings_get(const wchar_t* name, 
    boolean* input_enabled, boolean* output_enabled);

/**
 * @brief Sets the current input and output tensor-meta settings for the 
 * named Primary or Secondary GIE to use.
 * @param[in] name unique name of Primary or Secondary GIE to query
 * @param[in] input_enabled set to true preprocess input tensors attached as 
 * metadata instead of preprocessing inside the plugin, false otherwise.
 * @param[in] output_enabled set to true to attach tensor outputs as 
 * meta on GstBuffer.
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_INFER_RESULT otherwise.
 */
DslReturnType dsl_infer_gie_tensor_meta_settings_set(const wchar_t* name, 
    boolean input_enabled, boolean output_enabled);
``` 

Additionally, when using a Preprocessor, the Primary GIE unique_id must be set to 1... currently, the GIE generates a unique id from the name (hash) ... this needs to be refactored.

youngjae-avikus, 2022-07-11:
> Additionally, when using a Preprocessor, the Primary GIE unique_id must be set to 1... currently, the GIE generates a unique id from the name (hash) ... this needs to be refactored.

Can you explain exactly this part?
Should `target-unique-ids` match **config_preprocess.txt** and `gie-unique-id` match **config_inter.txt**?

rjhowell44, 2022-07-11:
@youngjae-avikus my apologies for the confusing comment above. It is inaccurate.  

The `target-unique-ids` allows you to identify which downstream target component(s) are to process the tensor meta.  What is unfortunate is that you can define a GIE's id via GST-property, but that `target-unique-ids` is fixed to a config file parameter; something I will bring up with nvidia. 

First, to make sense of this Issue... a little DSL history.

I made the decision early on to use unique names for identification of all DSL objects vs. integer ids. And that when defining Secondary Inference components the client would provide the unique name of the Primary or Secondary Inference component to infer on. To simplify this, I decided to have DSL manage the unique id's for all inference components overriding what is defined in the config file (using the GST-property). In the initial design, DSL generated it's unique id based on a hash of the GIE's name, but that made it difficult to work with the Preprocessor so I refactored the GIE component to use a 1-based numbering scheme instead.  I recently added additional information to the Infer docs to cover this ... https://github.com/prominenceai/deepstream-services-library/blob/v0.25.alpha/docs/api-infer.md#unique-id 

With DSL's Unique Id generation (with the refactoring defined in this issue), you can be assured that the first GIE you create in your application will always have id=1, and the second will have id=2 and so on.

youngjae-avikus, 2022-07-15:
Can I just use it as `gie-unique-id=1` for now?

rjhowell44, 2022-07-15:
Yes, the changes have been merged into v0.25.alpha..  DSL will set  `gie-unique-id=1` for first GIE created.

closed 2022-07-21 by rjhowell44