prominenceai/deepstream-services-library #808: Can Gst-interpipe be added to Deepstream-services-library ?

bharathraja, 2022-07-16:
[Gst-interpipe](https://github.com/RidgeRun/gst-interpipe) is used to dynamically handle multiple pipeline inter-connections. Can this be added to the DSL. 

Is there any documentation on how to add a custom plugin ?

rjhowell44, 2022-07-16:
@bharathraja not currently, but the interpipe plugins look very interesting! I will take some time to review this weekend and see what it would take to add to DSL.

rjhowell44, 2022-07-22:
This discussion has been continuing on [Discord](https://discord.gg/MJvY9jjpAK) ... support will most likely be included in the v0.26.alpha milestone.

rjhowell44, 2022-08-18:
This issue has be superseded by **Implement new Inter-Pipe Sink and Source components** #828

closed 2022-08-18 by rjhowell44