prominenceai/deepstream-services-library #818: The pgie interval value does not seem to work when pipeline includes preproc plug-in

youngjae-avikus, 2022-07-27:
Hello
There is a problem that interval does not apply in the preproc cpp example. 
Is it supposed to be like that?

https://github.com/prominenceai/deepstream-services-library/blob/v0.26.alpha/examples/cpp/1file_preproc_pgie_iou_osd_window.cpp

```cpp
// New Primary GIE using the filespecs defined above, with interval and Id
retval = dsl_infer_gie_primary_new(L"primary-gie", 
      primary_infer_config_file.c_str(), primary_model_engine_file.c_str(), 5);
if (retval != DSL_RESULT_SUCCESS) break;        
```

So I applied ode_monitor_action and printed it out, and the interval is continuously zeroing

[image]

I tried `dsl_infer_interval_set=5` and it's the same

Thank you

youngjae-avikus, 2022-07-27:
It's the same with gst-launch. 
Sounds like an interval problem with the nvidia deepstream
Will raising the issue be a substitute to @NVIDIA?

```bash
gst-launch-1.0 filesrc location = /opt/nvidia/deepstream/deepstream-6.0/samples/streams/sample_1080p_h264.mp4 ! qtdemux ! h264parse ! nvv4l2decoder ! m.sink_0 nvstreammux name=m batch-size=1 width=1920 height=1080 ! nvvideoconvert ! nvdspreprocess config-file= /opt/nvidia/deepstream/deepstream-6.0/sources/gst-plugins/gst-nvdspreprocess/config_preprocess.txt  ! nvinfer config-file-path= /opt/nvidia/deepstream/deepstream-6.0/samples/configs/deepstream-app/config_infer_primary.txt input-tensor-meta=1 batch-size=7 interval=3 ! nvtracker tracker-width=480 tracker-height=272 ll-lib-file=/opt/nvidia/deepstream/deepstream/lib/libnvds_nvmultiobjecttracker.so ll-config-file=/opt/nvidia/deepstream/deepstream/samples/configs/deepstream-app/config_tracker_IOU.yml ! nvmultistreamtiler width=1920 height=1080 ! nvvideoconvert ! nvdsosd ! nveglglessink
```

youngjae-avikus, 2022-07-27:
https://forums.developer.nvidia.com/t/pgie-interval-setting-not-work-when-use-nvdspreprocessing/200634/9?u=yeongjae8066

youngjae-avikus, 2022-07-27:
My deepstream is 6.0.1, above patch file soloved problem
if your environment are 6.1, can you test it without patch file?

rjhowell44, 2022-07-29:
@youngjae-avikus I'm also on 6.0.1  ... I may not get a change to upgrade for some time.

youngjae-avikus, 2022-07-29:
I should make a habit of creating a patch file from the file root :(

closed 2024-05-29 by rjhowell44