prominenceai/deepstream-services-library #819: Can only one pgie and tracker set when configuring one pipeline?

youngjae-avikus, 2022-07-27:
Hello!
read dsl API documents to implement the flow (compare tracker about 1 uri-source or pgie-custom-pph) I told you before, 
and it says that only a single pgie and tracker can be configured in one pipeline, is that correct?
In addition, is `splitter` and `demux` the part where only stream elements (not pgie, tracker) are used?

Thank you

rjhowell44, 2022-07-27:
@youngjae-avikus the only thing you would add to your Pipeline is the Splitter.  Then create two branches, each with a pgie and tracker and overlay sink.  Then add the branches to the Splitter. Do not add them to the Pipeline.  Once I implement a Muxer, you'll be able to join the two branches and use a single window sink. But until then, you will nee to use two separate overlay sinks, one for each branch.

youngjae-avikus, 2022-07-27:
1. So it doesn't matter if there are multiple pgies and trackers in one pipeline (multiple branch)?
2. `But until then, you will nee to use two separate overlay sinks, one for each branch.` According to this, it is possible to window sink each branch, but it does not work in my code. This seems to be a problem because I am not used to pipeline connections, so could you look at the problematic pipeline configuration code if I post it?

rjhowell44, 2022-08-25:
@youngjae-avikus using multiple Pipelines, each with their own Interpipe Source, PGIE, Tracker, OSD, and Window Sink, both listening to the same Interpipe Sink is a better way to go.  Also, as discussed on Discord, the Window Sinks can be replaced with Interpipe Sinks. An additional Pipeline with two Interpipe Sources, Tiler, and Window Sink can be used to combine the streams in a single Window. 

And lastly, you could use a single source and Pipeline with a tee and two branches, each with PGIE, Tracker, OSD, and Interpipe Sink and combine the two streams with the additional Pipeline described above -- closing this comment.

closed 2022-08-25 by rjhowell44