prominenceai/deepstream-services-library #820: Request to provide a service that can change the text offset at display render

youngjae-avikus, 2022-08-04:
Hello
Text may cover the box when rendering as shown below 
Of course, I think I can reduce the size of the text, but I want to reduce the text offset while maintaining it
There is a service related to this, but I wonder if I can't find it or not
Thank you.

[image]

https://docs.nvidia.com/metropolis/deepstream/sdk-api/struct__NvOSD__TextParams.html

rjhowell44, 2022-08-04:
@youngjae-avikus the service is not currently available, but this will be easy to add.  I'll get to this when I'm back on DSL next week.

rjhowell44, 2022-08-14:
here's the proposed API
```C
/**
 * @brief Creates a uniquely named "Customize Object Label" ODE Action that offsets 
 * an Object's label from the default location. 
 * @param[in] name unique name for the "Customize Object Label ODE Action. 
 * @param[in] offset_x horizontal offset from the default top left bounding box corner. Use a negative value to move left, positive to move right  in units of pixels.
 * @param[in] offset_y vertical offset from the default top left bounding box corner. Use a negative value to move up, positive to move down in units of pixels.
 * @return DSL_RESULT_SUCCESS on success, one of DSL_RESULT_ODE_ACTION_RESULT otherwise.
 */
DslReturnType dsl_ode_action_label_offset_new(const wchar_t* name,  
    int offset_x, int offset_y);
```

rjhowell44, 2022-08-17:
@youngjae-avikus this work has been merged into the `v0.26.alpha`.  See [dsl_ode_action_label_offset_new](https://github.com/prominenceai/deepstream-services-library/blob/v0.26.alpha/docs/api-ode-action.md#dsl_ode_action_label_offset_new)

youngjae-avikus, 2022-09-23:
dsl_ode_action_label_offset_new seems to have a bug

rjhowell44, 2022-09-23:
@youngjae-avikus can you provide more information including what values you are using for offset?

rjhowell44, 2022-09-28:
bug was fixed re: #851

closed 2023-05-03 by rjhowell44