prominenceai/deepstream-services-library #822: Have DSL support YoloV4,YoloV5 tensorrt models??

Ritesh1991, 2022-08-12:
HI have DSl library support Yolov4, YoloV5 versoin of models or not ?

rjhowell44, 2022-08-12:
@Ritesh1991 yes, although I use the nvidia supplied models for all the examples, I think most users are actually using Yolo versions.

@youngjae-avikus I believe you're using YoloV4 ... correct?

youngjae-avikus, 2022-08-13:
Yolov3, v4, v5 work well

Ritesh1991, 2022-08-16:
yes right now with yolov4 and tiny yolov4.Thanks for your reply. @youngjae-avikus  Do you have any example of yolov4 working with DSL? Can you share it? 
I have one more question which is different from this thread discussion. How to access USB Camera with DSL?
If any example you have then it's very good for me. Thanks in advance

rjhowell44, 2022-08-17:
@Ritesh1991 most of the examples use either a file source or URI source.  Any of the examples can easily be updated to use a USB source instead.  See [dsl_source_usb_new](https://github.com/prominenceai/deepstream-services-library/blob/master/docs/api-source.md#dsl_source_usb_new).

closed 2022-08-17 by rjhowell44

rjhowell44, 2022-08-17:
@Ritesh1991 most examples use a primary GIE by defining the paths to the inference config file and model engine file.
```python
# Filespecs for the Primary GIE and IOU Trcaker
primary_infer_config_file = \
    '/opt/nvidia/deepstream/deepstream/samples/configs/deepstream-app/config_infer_primary_nano.txt'
primary_model_engine_file = \
    '/opt/nvidia/deepstream/deepstream/samples/models/Primary_Detector_Nano/resnet10.caffemodel_b8_gpu0_fp16.engine'
```
The PGIE is then defined as follows
```python
    retval = dsl_infer_gie_primary_new('primary-gie', 
            primary_infer_config_file, primary_model_engine_file, 0)
```
If you're already using yolov4 then all you need to do is to update the file paths to use your config and model engine files.  This should be fairly simple and straight forward.

Ritesh1991, 2022-09-15:
hi Thanks for your reply If i want to use more than 1 usb camera than how to define it with  [dsl_source_usb_new](https://github.com/prominenceai/deepstream-services-library/blob/master/docs/api-source.md#dsl_source_usb_new).

rjhowell44, 2022-09-15:
That's a good question @Ritesh1991. It looks like I need to provide services to get/set the device location which defaults to 
`/dev/video0`  ... see documentation on the `v4l2src` plugin https://gstreamer.freedesktop.org/documentation/video4linux2/v4l2src.html#v4l2src:device.

I will add a new issue for this.

Ritesh1991, 2022-09-16:
thanks for your reply. add service also for multiple csi cameras.

Ritesh1991, 2022-09-27:
I dont have csi cameras but I have several USB cameras. I will test it and let you know

Ritesh1991, 2022-09-28:
Hi I tested with multiple cameras but with set property as well as without set property. But It always take input from first camera and ignore 2nd camera.

I tried with 2uri_file_pgie_ktl_demuxer_1osd_1overlay_1window.py in this example i replace two uri source with two cameras but it wouldnt worked it show only one camera frames on display and not 2nd one 
Then after i trie with dyn source example in that I assigned first source as camera, then next source uri but it only shows the first camera source and then after no update when dynamically added of the source. 

This is my observation upto now.
let me know if i am doing anything wrong 
Thanks .

rjhowell44, 2022-09-28:
@Ritesh1991 could I ask you to run a simple test with two USB cameras while capturing a log.
First, could you type in the following to see how they enumerate. 
`$ ls /dev/video*`

Then, set the debug output to a file
`$ export GST_DEBUG_FILE=./output.txt`
then set the logging output level with the below - this sets the DSL logging level to info. 
`$ export GST_DEBUG=1,DSL:4`
... run your example and then send me the file
I should have more cameras in a couple of days, but If you can send this I would very much appreciate it.

Ritesh1991, 2022-09-28:
I modified below attached file and i Got result as in output.log file both the cameras connected i check it with ls /dev/video*
[output.log](https://github.com/prominenceai/deepstream-services-library/files/9667777/output.log)
`################################################################################
# The MIT License
#
# Copyright (c) 2019-2021, Prominence AI, Inc.
#
# Permission is hereby granted, free of charge, to any person obtaining a
# copy of this software and associated documentation files (the "Software"),
# to deal in the Software without restriction, including without limitation
# the rights to use, copy, modify, merge, publish, distribute, sublicense,
# and/or sell copies of the Software, and to permit persons to whom the
# Software is furnished to do so, subject to the following conditions:
#
# The above copyright notice and this permission notice shall be included in
# all copies or substantial portions of the Software.
#
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT.  IN NO EVENT SHALL
# THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
# FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
# DEALINGS IN THE SOFTWARE.
################################################################################

#!/usr/bin/env python

import sys
import time

from dsl import *

uri_file = "/opt/nvidia/deepstream/deepstream/samples/streams/sample_1080p_h265.mp4"
source_width = 1920
source_height = 1080
# Filespecs for the Primary GIE and IOU Trcaker
primary_infer_config_file = \
    '/opt/nvidia/deepstream/deepstream/samples/configs/deepstream-app/config_infer_primary_nano.txt'
primary_model_engine_file = \
    '/opt/nvidia/deepstream/deepstream/samples/models/Primary_Detector_Nano/resnet10.caffemodel_b8_gpu0_fp16.engine'

## 
# Function to be called on XWindow KeyRelease event
## 
def xwindow_key_event_handler(key_string, client_data):
    print('key released = ', key_string)
    if key_string.upper() == 'P':
        dsl_pipeline_pause('pipeline')
    elif key_string.upper() == 'R':
        dsl_pipeline_play('pipeline')
    elif key_string.upper() == 'Q' or key_string == '' or key_string == '':
        dsl_pipeline_stop('pipeline')
        dsl_main_loop_quit()
 
## 
# Function to be called on XWindow Delete event
## 
def xwindow_delete_event_handler(client_data):
    print('delete window event')
    dsl_pipeline_stop('pipeline')
    dsl_main_loop_quit()

# Function to be called on End-of-Stream (EOS) event
def eos_event_listener(client_data):
    print('Pipeline EOS event')
    dsl_pipeline_stop('pipeline')
    dsl_main_loop_quit()

## 
[... 75 lines cut ...]
        retval = dsl_pipeline_state_change_listener_add('pipeline', state_change_listener, None)
        if retval != DSL_RETURN_SUCCESS:
            break
        retval = dsl_pipeline_eos_listener_add('pipeline', eos_event_listener, None)
        if retval != DSL_RETURN_SUCCESS:
            break

        # Play the pipeline
        retval = dsl_pipeline_play('pipeline')
        if retval != DSL_RETURN_SUCCESS:
            break

        # Join with main loop until released - blocking call
        dsl_main_loop_run()
        retval = DSL_RETURN_SUCCESS
        break

    # Print out the final result
    print(dsl_return_value_to_string(retval))

    dsl_pipeline_delete_all()
    dsl_component_delete_all()

if __name__ == '__main__':
    sys.exit(main(sys.argv))`

rjhowell44, 2022-09-28:
@Ritesh1991 Is there a chance that something else is running that might be accessing the camera?

Ritesh1991, 2022-09-29:
No, because I freshly flash jetson nano with a new image, install deep stream library and dependency. No other thing or application is running that accesses camera. As well as I also used the command fuser /dev/video0  and fuser /dev/video1 for checking which application accessing or using camera but I didn't find anything with it.

Have you checked the python file in which I made changes, did I modify it correctly?
Or you can give me some sample code that I can test it

rjhowell44, 2022-09-29:
@Ritesh1991 I've merged a change into the v0.26.alpha branch that should fix the problem with the constructor. You can remove the call to set the device-location.

There was one mistake in the script
```python
        retval = dsl_source_usb_device_location_set('uri-source1', '/dev/video1')
        retval = dsl_source_usb_new('uri-source1', source_width, source_height, 25, 1)
```
You would need to create the source first, then update its device-location. 

The above is failing which you can see in the log
```
DSL src/DslServicesSource.cpp:202:SourceUsbDeviceLocationSet:[00m  : Component name 'uri-source1' was not found
```
But there should be no need to set it with the fix I've merged in.

Ritesh1991, 2022-09-29:
`retval = dsl_source_usb_new('uri-source0', source_width, source_height, 25, 1)
        if retval != DSL_RETURN_SUCCESS:
            break  
        
 retval = dsl_source_usb_new('uri-source1', source_width, source_height, 25, 1)
        if retval != DSL_RETURN_SUCCESS:
            break    `
this will work after download in recompile correct ?

rjhowell44, 2022-09-29:
@Ritesh1991 yes, it should. Normally, I would full test before merging, but don't want to hold you

Ritesh1991, 2022-09-29:
Thanks for this. These how we all thrive by helping each other

rjhowell44, 2022-09-29:
@Ritesh1991 did it work for you?