prominenceai/deepstream-services-library #825: Example with RTSP inputs and individual RTSP output

micuentadecasa, 2022-08-14:
Hi all, 

thanks for this incredible library, I would like to know if you have some sample about creating dynamic pipelines of RTSP streams , having different RSTP streams as input, processing them with a pgie and some sgies, and a different RTSP ouput for each stream.

I'm trying to do it with Deepstream 6.1, but when adding/deleting new streams I got some issues with Deepstream and I have to restart the pipelines.

Regards.

rjhowell44, 2022-08-14:
@micuentadecasa re: *"I'm trying to do it with Deepstream 6.1, but when adding/deleting new streams I got some issues with Deepstream and I have to restart the pipelines."*

issues with DSL/DeepStream?  or issues just trying with just DeepStream?  Also, are you developing in Python or C/C++ ?

micuentadecasa, 2022-08-14:
Hi @rjhowell44 , I'm doing it in Python and Deepstream 6.1, I started from this code, https://github.com/NVIDIA-AI-IOT/deepstream_python_apps/tree/master/apps/runtime_source_add_delete

rjhowell44, 2022-08-14:
@micuentadecasa There are at least a couple of users that hang out on our [Discord site](https://discord.gg/MJvY9jjpAK) using DSL exactly as you mention.  Please join us if you're interested.  

Unfortunately, we don't have the use case you've described covered in a single example. You will need to pull  from the following.
* here's an example of how to create you're RTSP Sources and Recording Taps (not sure if you need recording) https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/ode_occurrence_4rtsp_start_record_tap_action.py
* here's an example of dynamically adding/removing sources... although it uses a Tiler which doesn't really work well for source remove.  https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/dyn_uri_file_pgie_ktl_tiler_osd_window.py
* and here's an example of how to create an RTSP Sink https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/1csi_live_pgie_osd_overlay_rtsp_h254.py
* finally, here's an example of creating a demuxer with multiple branches and sinks - https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/2rtsp_splitter_demuxer_pgie_ktl_tiler_osd_window_2_file.py 

Happy to help with any problems you might run into.

micuentadecasa, 2022-08-14:
thanks @rjhowell44 , I will take a look at the samples and will connect in the Discord site
Regards.

rjhowell44, 2022-08-14:
@micuentadecasa also, one of the import services you'll need will be [dsl_source_rtsp_state_change_listener_add](https://github.com/prominenceai/deepstream-services-library/blob/master/docs/api-source.md#dsl_source_rtsp_state_change_listener_add) which may not be in the above examples... and see the [RTSP Stream Connection Management](https://github.com/prominenceai/deepstream-services-library/blob/master/docs/overview.md#rtsp-stream-connection-management) as well.

micuentadecasa, 2022-08-14:
many thanks @rjhowell44

closed 2024-05-29 by rjhowell44