prominenceai/deepstream-services-library #851: Concept of x_offset, y_offset position in the dsl_ode_action_label_offset_new function

youngjae-avikus, 2022-09-23:
```cpp
retval = dsl_ode_action_label_offset_new(L"offset-label-action", 0, -15);
```
If write the code like above, it comes out like below
[image]

On the contrary, the offset is formed as I want to, and am I confused with the original offset_x, offset_y concept? I've looked at the internal code of the DSL and it seems that there is no major problem

```cpp
retval = dsl_ode_action_label_offset_new(L"offset-label-action", -15, 0);
```

Thank U

rjhowell44, 2022-09-23:
Thanks for the details @youngjae-avikus ... now I understand.  Your first statement looks correct to me.  Seems like something is wrong. My behavior test is setting both the X and Y offset to the -15, so I may have overlooked an issue.  Perhaps I'm swapping the offset_x and offset_y in the code or I'm just as confused as you about what the original offset X and Y mean.  I will look at it today.

rjhowell44, 2022-09-27:
@youngjae-avikus I found and fixed an issue with my offset logic.  I was using the `source-width` and `source-height` from the frame metatdata to try and ensure that large positive offsets would still remain within view.  The one thing that I forget is that the Tiler sets the `source-width` and `source-height` to 0 on output. Adding the ODE PPH on the Tiler Src Pad or OSD Sink Pad would lead to the strange behavior as you've shown above. 

I've removed the extra logic and verified that the OSD can handle values beyond the frame size without issue.  Let me know if this does not fix your issue.

closed 2023-05-03 by rjhowell44