prominenceai/deepstream-services-library #859: OpenCV 4.2 throws error during building DSL for dGOU

rahulthakur319, 2022-10-02:
`Makefile` throws error during the Building of DSL library related to Opencv 4.2

```
Ubuntu 20.04
NVIDIA-SMI Driver Version: 515.65.01    
CUDA Version: 11.7
Gstreamer: 1.16.3
TensorRT: 8.4.1
OpenCV: 4.2.0
 
```

```
Package opencv was not found in the pkg-config search path.
Perhaps you should add the directory containing `opencv.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv' found
```

rjhowell44, 2022-10-04:
@rahulthakur319 , thanks for reporting this and providing your details. I plan to look at this tomorrow.

rahulthakur319, 2022-10-04:
Thanks for quick revert @rjhowell44. 

Standard installation of OpenCV 4+ is not creating `pkg-config` (opencv.pc) file. This can be referred by to open issue at OpenCV: https://github.com/opencv/opencv/issues/13154

A quick workaround is to build OpenCV using `-D OPENCV_GENERATE_PKGCONFIG=YES` during `cmake`

```
cmake -D CMAKE_BUILD_TYPE=Release -D OPENCV_GENERATE_PKGCONFIG=YES -D CMAKE_INSTALL_PREFIX=/usr/local ..
sudo make -j4
sudo make install
```

The presence of `.pc file` now can tested using

```
pkg-config --cflags opencv4
```

rjhowell44, 2022-10-20:
This needs to be documented somewhere

closed 2024-05-29 by rjhowell44