prominenceai/deepstream-services-library #860: Way to speed up image processing in dsl_source_image_multi_new?

youngjae-avikus, 2022-10-07:
https://github.com/prominenceai/deepstream-services-library/blob/fe3d7c13f5a258a35e5c09a8ac29cd0054334317/src/DslSourceBintr.cpp#L1255-L1272

I understand that it is inferring the image over at the speed of fps_n/fps_d
It's pretty slow for me, now, with just one image per second
So the code says that the part that controls the fps is not active at this time  Is there a way?
Or, as an alternative, I could make videos of the current images

Thank U

rjhowell44, 2022-10-11:
@youngjae-avikus reviewing my notes, I realized that I ran into problems developing the image sources and put the issue aside for a bit.   I found the information I needed in the GStreamer plugin source code so I can finish them. https://github.com/GStreamer/gst-plugins-good/blob/master/gst/multifile/gstmultifilesrc.c   ... interestingly, there is a comment there about using the plugin to convert the images into an ogg video file on the command line.  I don't know anything about ogg, but something for you to look into.

youngjae-avikus, 2022-10-13:
@rjhowell44 You mean there's no solution at the moment, right?

closed 2023-05-03 by rjhowell44