prominenceai/deepstream-services-library #875: Reading detections from file for tracker

SiftingSands, 2022-11-07:
I'm interested in using the NvDCF tracker (https://github.com/prominenceai/deepstream-services-library/blob/master/docs/api-tracker.md), but I believe the typical workflow is to have it after a detection pipeline. However, I already have detection results, so I would like to read the bounding boxes from storage and feed that information along with the RGB arrays to the tracker. Is this possible? If so, are there any good places to start?

rjhowell44, 2022-11-08:
@SiftingSands In what form are your inference results?  You can add your own DeepStream Object metadata to the Frame metadata generated by a NVIDIA Stream-muxer using the DeepStream APIs.  How familiar/experienced with DeepStream and GStreamer are you?   Have you worked with any of there C/C++ examples https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_C_Sample_Apps.html ?

SiftingSands, 2022-11-09:
Detection results are stored in a CSV format. I'm pretty new to DeepStream and GStreamer. I have not worked with the C/C++ examples from your link. I'm assuming that this can't be done purely in Python?

rjhowell44, 2022-11-10:
@SiftingSands See https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_Python_Sample_Apps.html for Python versions.  However, this will be a difficult task for someone new to DeepStream/GStreamer. You will need to find an example that has a Primary GST Inference Engine (PGIE) and Tracker and then remove the PGIE. You will then need to add a Pad Probe Handler function to the Sink Pad of the Tracker to add your inference results as metadata. There are NVIDIA functions to get the batch-metadata from the buffer to get at the Frame metadata. There are routines to create and add Object metadata to each frame's metadata.  Again, none of this will be trivial tough, and not that well documented... especially for Python

SiftingSands, 2022-11-10:
Thanks so much for the detailed response! I had a feeling this was going to be complicated, but I appreciate the confirmation. I'll close this issue for now, as I'll have to put testing out NvDCF on the backburner for my use case.

closed 2022-11-10 by SiftingSands