prominenceai/deepstream-services-library #876: Image processing backends - a.k.a. decoupling from OpenCV

Paskin, 2022-11-13:
First of all - thanks for a great work! Your library really helps to eliminate the "boilerplate" and introduces a lot of cool features, especially in the RTSP area.
OpenCV is a great framework/library too - but it is intended for a research, not the production. In some tests it slower almost by order of a magnitude compared to (for example) Pillow - not to speak about super-optimised libraries from CPU/GPU vendors.
Could you consider extraction of OpenCV dependencies into a separate component - so alternative implementations will be much easier to provide/share?
Thanks!

rjhowell44, 2022-11-14:
@Paskin yes, I would be happy to take a look at this in the context of the next (v0.27.alpha) release.  Trying to wrap up the current v0.26.alpha release ASAP.

Paskin, 2023-01-11:
Now, with recently released CVCUDA framework - importance of the change I've mentioned seems to grow even more ;)

rjhowell44, 2023-01-11:
@Paskin in an effort to better understand the scope of this work, I've created a side branch that makes the use of opencv optional... default = build without.    see... 
https://github.com/prominenceai/deepstream-services-library/tree/opt_capture

The Object and Frame Capture actions will be disabled, but everything else should be functional. 

Feel free to give it a try if you're stuck and unable to build.

rjhowell44, 2023-02-19:
@Paskin this issue has been superseded with the issue:

**Remove and replace all "opencv" usage - remove build dependency #908**

This work has been completed and merged into the **v0.26.alpha** dev branch... which is to be release next week.

closed 2023-02-19 by rjhowell44