prominenceai/deepstream-services-library #878: Receive SIGFPE (Floating point exception (core dumped)) when execute dsl_pipeline_streammux_tiler_add

youngjae-avikus, 2022-11-15:
Problem
Problems arise when 3URI or 3RTSP (both tests) + streammux_tiler
The model was also run with yolov3 and yolov5, and both resulted in exceptions.
The timing of the exception is irregular, so it takes about 2-3 minutes, and it may occur after that.
Here is my code...

```cpp
#include <iostream>
#include <glib.h>
#include <X11/Xlib.h>

#include "DslApi.h"

// Set Camera RTSP URI's - these must be set to valid rtsp uri's for camera's on your network
// RTSP Source URI
std::wstring rtsp_uri_1 = L"rtsp://192.168.1.40:554/h264";
std::wstring rtsp_uri_2 = L"rtsp://192.168.1.40:554/h264";
std::wstring rtsp_uri_3 = L"rtsp://192.168.1.40:554/h264";

// File path for the single File Source
std::wstring file_path1(
    L"/opt/nvidia/deepstream/deepstream-6.0/samples/streams/sample_1080p_h265.mp4");
std::wstring file_path2(
    L"/opt/nvidia/deepstream/deepstream-6.0/samples/streams/sample_qHD.mp4");
std::wstring file_path3(
    L"/opt/nvidia/deepstream/deepstream-6.0/samples/streams/sample_ride_bike.mov");

std::wstring primary_infer_config_file(
    L"/opt/dsl/nas_data/YOLOV3/config_infer_primary_yoloV3.txt");
std::wstring primary_model_engine_file(
    L"/opt/dsl/nas_data/YOLOV3/model_b1_gpu0_fp16.engine");
std::wstring tracker_config_file(
    L"/opt/dsl/nas_data/YOLOV3/config_tracker_IOU.yml");

// File name for .dot file output
static const std::wstring dot_file = L"state-playing";

int TILER_WIDTH = DSL_STREAMMUX_1K_HD_WIDTH; 
int TILER_HEIGHT = DSL_STREAMMUX_1K_HD_HEIGHT;

// Window Sink Dimensions - used to create the sink, however, in this
// example the Pipeline XWindow service is called to enabled full-sreen
int WINDOW_WIDTH = DSL_STREAMMUX_1K_HD_WIDTH;
int WINDOW_HEIGHT = DSL_STREAMMUX_1K_HD_HEIGHT;

int SHOW_SOURCE_TIMEOUT = 3;

//
// Function to be called on XWindow KeyRelease event
//
void xwindow_key_event_handler(const wchar_t* in_key, void* client_data)
{   
    std::wstring wkey(in_key); 
    std::string key(wkey.begin(), wkey.end());
    std::cout << "key released = " << key << std::endl;
    key = std::toupper(key[0]);
    if(key == "P"){
        dsl_pipeline_pause(L"pipeline");
    } else if (key == "R"){
        dsl_pipeline_play(L"pipeline");
    } else if (key == "Q" or key == "" or key == ""){
        dsl_pipeline_stop(L"pipeline");
        dsl_main_loop_quit();
    } else if (key >= "0" and key <= "3"){
        const wchar_t* source;
        
        if (dsl_source_name_get(std::stoi(key), &source) == DSL_RESULT_SUCCESS)
            dsl_tiler_source_show_set(L"tiler", 
                source, SHOW_SOURCE_TIMEOUT, true);
    
[... 186 lines cut ...]
   from /opt/nvidia/deepstream/deepstream/lib/libnvds_nvmultiobjecttracker.so
(gdb) bt
#0  0x00007f551e779c07 in NvTrackedObject::updateTrajectoryBuffer() ()
    at /opt/nvidia/deepstream/deepstream/lib/libnvds_nvmultiobjecttracker.so
#1  0x00007f551e77d537 in NvTrackedObjectManager::updateTrajectoryBuffers(unsigned long) ()
    at /opt/nvidia/deepstream/deepstream/lib/libnvds_nvmultiobjecttracker.so
#2  0x00007f551e780804 in NvTrackedObjectManager::updateBatch(std::map<unsigned long, _NvMOTFrame*, std::less<unsigned long>, std::allocator<std::pair<unsigned long const, _NvMOTFrame*> > > const&, std::map<unsigned long, assocDataOut_t, std::less<unsigned long>, std::allocator<std::pair<unsigned long const, assocDataOut_t> > >) () at /opt/nvidia/deepstream/deepstream/lib/libnvds_nvmultiobjecttracker.so
#3  0x00007f551e76fe15 in NvMultiObjectTrackerBase::update(std::map<unsigned long, _NvMOTFrame*, std::less<unsigned long>, std::allocator<std::pair<unsigned long const, _NvMOTFrame*> > > const&, _NvMOTTrackedObjBatch*&) () at /opt/nvidia/deepstream/deepstream/lib/libnvds_nvmultiobjecttracker.so
#4  0x00007f551e784b73 in NvMOTContext::processFrame(_NvMOTProcessParams const*, _NvMOTTrackedObjBatch*) ()
    at /opt/nvidia/deepstream/deepstream/lib/libnvds_nvmultiobjecttracker.so
#5  0x00007f551e7859fe in NvMOT_Process ()
    at /opt/nvidia/deepstream/deepstream/lib/libnvds_nvmultiobjecttracker.so
#6  0x00007f5570860a2e in NvTrackerProc::processBatch() ()
    at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/deepstream/libnvdsgst_tracker.so
#7  0x00007f557086607a in void std::__invoke_impl<void, void (NvTrackerProc::*)(), NvTrackerProc*>(std::__invoke_memfun_deref, void (NvTrackerProc::*&&)(), NvTrackerProc*&&) ()
    at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/deepstream/libnvdsgst_tracker.so
#8  0x00007f557086288e in std::__invoke_result<void (NvTrackerProc::*)(), NvTrackerProc*>::type std::__invoke<void (NvTrackerProc::*)(), NvTrackerProc*>(void (NvTrackerProc::*&&)(), NvTrackerProc*&&) ()
    at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/deepstream/libnvdsgst_tracker.so
#9  0x00007f557087d397 in decltype (__invoke((_S_declval<0ul>)(), (_S_declval<1ul>)())) std::thread::_Invoker<std::tuple<void (NvTrackerProc::*)(), NvTrackerProc*> >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>) () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/deepstream/libnvdsgst_tracker.so
#10 0x00007f557087d2ce in std::thread::_Invoker<std::tuple<void (NvTrackerProc::*)(), NvTrackerProc*> >::operator()() () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/deepstream/libnvdsgst_tracker.so
```

[image]

[image]

youngjae-avikus, 2022-11-15:
It is data_link
https://drive.google.com/file/d/1Z85OwoBrKsNzVilCj3SXldklAEabktgS/view?usp=share_link

rjhowell44, 2022-11-16:
@youngjae-avikus I've tried running your example on my Jetson device(s) with the NVIDIA supplied models and I'm unable to reproduce your failure.  I've tried to run the example with your YOLOv3 model, but I'm unable to get it to work. My console output is...
```
Output yolo blob names :
yolo_83
yolo_95
yolo_107
Total number of yolo layers: 257
Building yolo network complete!
Building the TensorRT Engine...
Killed
```
Can you try with the NIVIDIA models on your dGPU platform to see if it's model or platform dependent?  I've also noticed that the inference results using NVIDIA's models are terrible. Very few objects detected so the Tracker is doing little in my case.  I tried with different Tiler output dimensions as follows, but no imporovement. 
```C
int TILER_WIDTH = DSL_STREAMMUX_1K_HD_WIDTH*3; 
int TILER_HEIGHT = DSL_STREAMMUX_1K_HD_HEIGHT;

// Window Sink Dimensions - used to create the sink, however, in this
int WINDOW_WIDTH = DSL_STREAMMUX_1K_HD_WIDTH;
int WINDOW_HEIGHT = DSL_STREAMMUX_1K_HD_HEIGHT/3;
```

youngjae-avikus, 2022-11-17:
@rjhowell44 
It seems that the problem of yolov3 not running is `libnvdsinfer_custom_impl_Yolo.so`. I ran it with so built in CUDA 11.4(DGPU), so would you like to rebuild it in 10.2(JETSON)? about below folder...
[image]

youngjae-avikus, 2022-11-17:
> Can you try with the NIVIDIA models on your dGPU platform to see if it's model or platform dependent? I've also noticed that the inference results using NVIDIA's models are terrible. Very few [...]

In my opinion, in order to perform well in a tiled image, it is basically trained with a tiled image size. Also, for another reason, yolov3 input image size is 416x416. So eventually, whatever the tiled image size, it resizes to 416x416

When it is non-tile, the original image is 1920x1080, and when it enters Yolov3, it is a conventionally trained method, so if it is well trained, detection is good.

However, if the tiled image in 3 sheets enters one input, 5840x1080 enters and change to 416x416 (YOLOV3). Then, since the squeeze is tripled on a width basis than 1920x1080, the ratio will be resized to a stranage image and the detection will be rather inferior in performance

youngjae-avikus, 2022-11-17:
To better detect the 5840x1080 image, I think it need to do 3 slice predictions with 1920x1080 using the preproc plug-in (additionally, Full reference)

youngjae-avikus, 2022-11-17:
@rjhowell44 

I didn't know exactly what you were referring to. Are you saying that both dgpu and Jetson are different about the model performance provided by nvidia? Or do you mean that the performance is significantly reduced when you apply tiled input it?

rjhowell44, 2022-11-18:
@youngjae-avikus I'm saying that I can NOT reproduce the error on Jetson with the NVIDIA café model. Having the Streammuxer connected to Tiler works fine.  I'm asking if you can test on dGPU with the NVIDIA model, and on Jetson with your YOLOv3 model.  We need to know if the issue is observed on dGPU only? or your model only?

youngjae-avikus, 2022-11-18:
> @youngjae-avikus I'm saying that I can NOT reproduce the error on Jetson with the NVIDIA café model. Having the Streammuxer connected to Tiler works fine. I'm asking if you can test on dGPU with the [...]

OK. I understand that

youngjae-avikus, 2022-11-18:
When operating with the nvidia default caffe model using dGPU, there is no error on three video inputs, but there is still a problem with three RTSPs
I'll go to work next Monday and verify what I'm doing at Jetson. The dGPU is running in the docker, so I don't know if there's something wrong with this... That's weird

rjhowell44, 2022-11-18:
@youngjae-avikus Sorry, I was confused. I thought you were saying that the issues was caused by adding the Tiler to the Stream-muxer.  Are you using the same RTSP URI for all three sources?  I believe this might be the issue? For my cameras, I need to use different channels such as.
```python
rtsp://username:password@192.168.0.14:554/Streaming/Channels/101
rtsp://username:password@192.168.0.14:554/Streaming/Channels/102
rtsp://username:password@192.168.0.14:554/Streaming/Channels/103
```

youngjae-avikus, 2022-11-27:
@rjhowell44 

Up to four RTSP cameras that I use are supported up to Multicast, so I did the following when running the program

rjhowell44, 2022-11-28:
@youngjae-avikus sorry but I do not understand... you stated ***"so I did the following when running the program"***  ... but there is nothing following your comment? What is "the following"?

Also, you initially stated that you wanted tile three different streams before running inference... tile them together as a single panoramic view . Would you not be using three different cameras?  Why would you tile multiple streams from the same camera? I'm trying to understand you use case.

youngjae-avikus, 2022-11-28:
@rjhowell44 

You're right ! it's just that before the three cameras were built in the field, I ran three same streams for the stress test and dsl_pipeline_streammux_tiler_add work. However, even if the RTSP stream is excluded, the combination of YOLOV3 + 3-VIDEO also caused SIGFPE to occur and left an issue :)

rjhowell44, 2022-11-28:
@youngjae-avikus  What error do you get with 3 video (non-RTSP) streams... something other than "nas_rtp_connect" error?  can you share the log here?

youngjae-avikus, 2022-11-28:
> @youngjae-avikus What error do you get with 3 video (non-RTSP) streams... something other than "nas_rtp_connect" error? can you share the log here?

@rjhowell44 
The error is caused by the same SIGFPE as above. I am currently on a business trip, so I will be able to check it again on Wednesday 
Is there anything else you would like to request?

[image]

rjhowell44, 2022-11-28:
@youngjae-avikus I'm confused... this error is caused using three different RTSP sources?

rjhowell44, 2022-11-28:
@youngjae-avikus I see... I was confused. This error seems related to the multi-object tracker and the results produced by your YOLOv3 model.  The nas_rtsp_connect was misleading.

youngjae-avikus, 2022-11-28:
Does "The nas_rtsp_connect was misleading" mean that the code I wrote is problematic? Or do you think it's a build issue for yolov3.so?

rjhowell44, 2022-11-29:
@youngjae-avikus I'm not sure what this is.  If we look at the three cases
1. untiled batched input with YOLOv3 - no issue
2. tiled input with Cafe model - minimal detections - no issue
3. tiled input with YOLOv3 - significant number of detections - floating point exception.

It seems that there is something in the metadata produced by the PGIE that is causing issues for the tracker. 

Can you add an ODE-Handler with a Print Action to the source pad of the PGIE so we can look at the metadata before it gets to the Tracker?  One think I do know is that the Tiler sets the frame width and height to 0 which may or may not be an issue.

youngjae-avikus, 2022-12-20:
> @youngjae-avikus I'm not sure what this is. If we look at the three cases
>
> 1. untiled batched input with YOLOv3 - no issue
> 2. tiled input with Cafe model - minimal detections - no issue
> 3. tiled [...]

I did pending for a long time, but I'll check the issue up again

closed 2024-02-02 by youngjae-avikus