prominenceai/deepstream-services-library #882: WebRtc Sink Component is failing to create on GStreamer gstreamer 1.20

rjhowell44, 2022-11-22:


rjhowell44, 2023-01-11:
@52doho, unfortunately, the WebRtc Sink is broken in the master do to a merger issue. Your change should work.  I'm in the middle of an issue now which I hope to finish tomorrow.  This issue is next.

Please try with the logging enabled and share with me here.  
```
export GST_DEBUG=1,DSL:4
```

52doho, 2023-01-12:
After switch to a LAN machine, [1file_webrtc_connect_post_play.py](https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/1file_webrtc_connect_post_play.py) works now! The websocket port must be available to listen.

This is the dsl log and web playing video.
[image]

52doho, 2023-01-15:
The [1file_webrtc_connect_post_play.py](https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/1file_webrtc_connect_post_play.py) example can work on first html load. After refresh html page, the webrtc player was hang on. I try to debug, but no idea why.

Here is the log file.
[webrtc-hangs-on-after-refresh-html-page.log](https://github.com/prominenceai/deepstream-services-library/files/10420462/webrtc-hangs-on-after-refresh-html-page.log)

Sometime the program hang on DslNodetr.h Line 612.
[image]

rjhowell44, 2023-01-19:
@52doho my apologies for the late response.  There is an existing issue for this.  see https://github.com/prominenceai/deepstream-services-library/issues/666.  The webrtc sink plugin does not seem to support dynamic removal the way other sinks bins do. I've done some extensive searching and I can't find another example that implements a "connect-post-play" scenario.  All example are similar to the [1file_webrtc_connect_pre_play.py](https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/1file_webrtc_connect_pre_play.py) python example.

That said, I may be able to come up with a work around. I'm trying to wrap up some work so I can get back on this.  I'll follow up with an issue on the webrtc github page as well.

regards,
Robert.

rjhowell44, 2023-02-02:
Thanks for the info @52doho, I should be able to get to this next week.

rjhowell44, 2023-05-04:
The Bintr creation issue has been fixed.  A new issue will be opened for the page-refresh issue described above.

closed 2023-05-04 by rjhowell44