prominenceai/deepstream-services-library #884: Questions about whether rtsp to jpeg decode is supported

youngjae-avikus, 2022-12-01:
I'd like to receive it in rtsp://ip/jpeg format now, but is there any other way so that I can't receive it with the existing `rtsp_source_new`?

[image]

youngjae-avikus, 2022-12-01:
[image]

https://docs.nvidia.com/metropolis/deepstream/6.0.1/dev-guide/text/DS_FAQ.html

youngjae-avikus, 2022-12-01:
[image]

This issue arises

```cpp
std::wstring rtsp_uri_1 = L"rtsp://192.168.1.140:554/jpeg";

retval = dsl_source_rtsp_new(L"rtsp-source-1", rtsp_uri_1.c_str(), DSL_RTP_ALL,     
            false, 0, 100, 2);
if (retval != DSL_RESULT_SUCCESS)    
    return retval;
```

youngjae-avikus, 2022-12-01:
```cpp
bool RtspSourceBintr::HandleSelectStream(GstElement *pBin, uint num, GstCaps *caps)
    {
        GstStructure *structure = gst_caps_get_structure(caps, 0);
        std::string media = gst_structure_get_string (structure, "media");
        std::string encoding = gst_structure_get_string (structure, "encoding-name");

        LOG_INFO("Media = '" << media << "' for RtspSourceBitnr '" << GetName() << "'");
        LOG_INFO("Encoding = '" << encoding << "' for RtspSourceBitnr '" << GetName() << "'");
        
        if (!m_pParser)
        {
            if (media.find("video") == std::string::npos)
            {
                LOG_WARN("Unsupported media = '" << media << "' for RtspSourceBitnr '" 
                    << GetName() << "'");
                return false;
            }
            if (encoding.find("H264") != std::string::npos)
            {
                m_pParser = DSL_ELEMENT_NEW("h264parse", GetCStrName());
                m_pDepay = DSL_ELEMENT_NEW("rtph264depay", GetCStrName());
            }
            else if (encoding.find("H265") != std::string::npos)
            {
                m_pParser = DSL_ELEMENT_NEW("h265parse", GetCStrName());
                m_pDepay = DSL_ELEMENT_NEW("rtph265depay", GetCStrName());
            }
            else
            {
                LOG_ERROR("Unsupported encoding = '" << encoding << "' for RtspSourceBitnr '" 
                    << GetName() << "'");
                return false;
            }
            AddChild(m_pDepay);
            AddChild(m_pParser);

            // If we're tapping off of the pre-decode source stream, then link to the pre-decode Tee
            // The Pre-decode Queue will already be linked downstream as the first branch on the Tee
            if (HasTapBintr())
            {
                if (!m_pDepay->LinkToSink(m_pParser) or !m_pParser->LinkToSink(m_pPreDecodeTee))
                {
                    return false;
                }            
            }
            // otherwise, there is no Tee and we link to the Pre-decode Queue directly
            else
            {
                if (!m_pDepay->LinkToSink(m_pParser) or !m_pParser->LinkToSink(m_pPreDecodeQueue))
                {
                    return false;
                }            
            }
            if (!gst_element_sync_state_with_parent(m_pDepay->GetGstElement()) or
                !gst_element_sync_state_with_parent(m_pParser->GetGstElement()))
            {
                LOG_ERROR("Failed to sync Parser/Decoder states with Parent for RtspSourceBitnr '" 
                    << GetName() << "'");
                return false;
            }
        }
        return true;
```
deepstream-services-library/src/DslSourceBintr.cpp
Looking at this code, I think it's possible to control the rtsp source separately (decode nvjpeg), is it possible?

rjhowell44, 2022-12-02:
@youngjae-avikus the callback is registered with the rtsp plugin ... see https://gstreamer.freedesktop.org/documentation/rtsp/rtspsrc.html?gi-language=c#rtspsrc::select-stream.

The plugin calls the callback once the caps have been read from the actual stream contents.  It is the stream itself that specifies whether it is video or audio... and if video what type of encoding is used H264 or H265.  

I did some searching today but could not find any information on how to handle  rtsp://ip/jpeg

youngjae-avikus, 2022-12-02:
@rjhowell44 
[image]

```bash
gst-launch-1.0 uridecodebin uri=rtsp://192.168.1.140:554/jpeg ! nveglglessink
```

It runs with uridecodebin, but there is a delay in the image

rjhowell44, 2022-12-02:
@youngjae-avikus I'm sorry I don't understand your statement ***but is it possible to do it right away because I modified the DSL code?***  ...are you saying that you tried to modify DSL, or are you asking if I can update DSL?

youngjae-avikus, 2022-12-02:
@rjhowell44 
Yes, I think my explanation was poor. To explain it again, When opening an rtsp stream that has already been encoded with h264, it usually seems to open with the pipeline below

```bash
gst-launch-1.0 -vvv rtspsrc location=rtsp://192.168.1.140:554/h264 ! rtph264depay ! h264parse ! nvv4l2decoder ! nvvideoconvert ! nvdsosd ! nveglglessink sync=0
```

And my situation is that I have to do jpeg decoding now, but the pipeline seems to be the difference of some components as below

```bash
gst-launch-1.0 -vvv rtspsrc location=rtsp://192.168.1.140:554/jpeg ! rtpjpegdepay ! jpegparse ! nvjpegdec ! nvdsosd ! nveglglessink sync=0
```

Currently, DSL does not support `rtsp-jpeg`, so can I simply make it possible to add additional components? (including inference and tracking)

If this is easy, I can challenge it, but I don't think it's just modifying the source part, so could you give me an opinion?

youngjae-avikus, 2022-12-02:
> @youngjae-avikus the callback is registered with the rtsp plugin ... see https://gstreamer.freedesktop.org/documentation/rtsp/rtspsrc.html?gi-language=c#rtspsrc::select-stream.
>
> The plugin calls the [...]

Sometimes there seems to be a request for mjpeg decode in the form of rtsp.
https://stackoverflow.com/questions/61104698/how-to-send-mjpeg-frames-on-rtsp

rjhowell44, 2022-12-02:
@youngjae-avikus check the log again, and if you see something like
```
Unsupported encoding = 'mjpeg' for RtspSourceBitnr 
```
try adding the below to the select-stream
```C
            else if (encoding.find("jpeg") != std::string::npos)
            {
                m_pParser = DSL_ELEMENT_NEW("jpegparse", GetCStrName());
                m_pDepay = DSL_ELEMENT_NEW("rtpjpegdepay", GetCStrName());
            }
```

youngjae-avikus, 2022-12-02:
@rjhowell44 
I added the code as below, but it doesn't print even if I cout the HandleSelectStream
```cpp
bool RtspSourceBintr::HandleSelectStream(GstElement *pBin, uint num, GstCaps *caps)
    {
        std::cout << "HandleSelectStream\n";
        GstStructure *structure = gst_caps_get_structure(caps, 0);
        std::string media = gst_structure_get_string (structure, "media");
        std::string encoding = gst_structure_get_string (structure, "encoding-name");

        LOG_INFO("Media = '" << media << "' for RtspSourceBitnr '" << GetName() << "'");
        LOG_INFO("Encoding = '" << encoding << "' for RtspSourceBitnr '" << GetName() << "'");

        if (!m_pParser)
        {
            if (media.find("video") == std::string::npos)
            {
                LOG_WARN("Unsupported media = '" << media << "' for RtspSourceBitnr '" 
                    << GetName() << "'");
                return false;
            }
            if (encoding.find("H264") != std::string::npos)
            {
                m_pParser = DSL_ELEMENT_NEW("h264parse", GetCStrName());
                m_pDepay = DSL_ELEMENT_NEW("rtph264depay", GetCStrName());
            }
            else if (encoding.find("H265") != std::string::npos)
            {
                m_pParser = DSL_ELEMENT_NEW("h265parse", GetCStrName());
                m_pDepay = DSL_ELEMENT_NEW("rtph265depay", GetCStrName());
            }
            else if (encoding.find("jpeg") != std::string::npos)
            {
                m_pParser = DSL_ELEMENT_NEW("jpegparse", GetCStrName());
                m_pDepay = DSL_ELEMENT_NEW("rtpjpegdepay", GetCStrName());
            }
            else
            {
                LOG_ERROR("Unsupported encoding = '" << encoding << "' for RtspSourceBitnr '" 
                    << GetName() << "'");
                return false;
            }
```

There are still missing packets

[image]

I think this callback function below is not connected, should I activate it separately? I think I need to learn this part

```cpp
g_signal_connect (m_pSourceElement->GetGObject(), "select-stream",
            G_CALLBACK(RtspSourceSelectStreamCB), this);
```

The code is first running based on `rtsp_player_to_test_connections.cpp`.  
The address is still given below.
```cpp
std::wstring rtsp_uri_1 = L"rtsp://192.168.1.140:554/jpeg";
```

rjhowell44, 2022-12-05:
@youngjae-avikus can you tell if the select-stream callback is being called?

youngjae-avikus, 2022-12-05:
@rjhowell44 

Judging from the `g_signal_connect` function below, the `RtspSourceSelectStreamCB` function should be called when it is called normally.

The code now controls rtsp only with `ip(std::wstring rtsp_uri_1 = L"rtsp://192.168.1.140:554/jpeg)` and `dsl_source_rtsp_new`

```cpp
g_signal_connect (m_pSourceElement->GetGObject(), "select-stream",
            G_CALLBACK(RtspSourceSelectStreamCB), this);
```

rjhowell44, 2022-12-05:
@youngjae-avikus the `RtspSourceSelectStreamCB` (select-stream) callback should be called by the `rtspsrc` plugin with the caps from the stream. 

I'm sorry, by I don't understand your statement.... ***The code now controls rtsp only with ip(std::wstring rtsp_uri_1 = L"rtsp://192.168.1.140:554/jpeg) and dsl_source_rtsp_new***  

There is clearly missing functionality in `RtspSourceSelectStreamCB` needed to handle streams other than video/H264 and video/H265 only... but it appears that it is not even called when using your URI, is this correct?

youngjae-avikus, 2022-12-05:
> @youngjae-avikus the `RtspSourceSelectStreamCB` (select-stream) callback should be called by the `rtspsrc` plugin with the caps from the stream.
>
> I'm sorry, by I don't understand your statement.... [...]

Yes, that’s right.
**The code now controls rtsp only with ip(std::wstring rtsp_uri_1 = L"rtsp://192.168.1.140:554/jpeg) and dsl_source_rtsp_new**
means ... 
I'm simply running rtsp with the code below, and I was wondering if there was anything else I needed
```cpp
std::wstring rtsp_uri_1 = L"rtsp://192.168.1.140:554/jpeg)

// # For each camera, create a new RTSP Source for the specific RTSP URI    
retval = dsl_source_rtsp_new(L"rtsp-source", rtsp_uri_1.c_str(), DSL_RTP_ALL,     
       false, 0, 100, 2);
if (retval != DSL_RESULT_SUCCESS)    
       return retval;
```

Is `RtspSourceSelectStreamCB` a function that runs unconditionally? I can't enter the function at all, so what's the problem?

rjhowell44, 2022-12-05:
@youngjae-avikus my understanding is that it should be called by the plugin for each stream received... Audio, Video, Image, etc.

youngjae-avikus, 2022-12-05:
@rjhowell44 
Hmm... I need to do more debugging... I have a separate h265 camera, so I'll test it

youngjae-avikus, 2022-12-06:
As an example, if open the camera with `rtsp_uri_2`, we end up with the last log below

```cpp
#include <iostream>
#include <experimental/filesystem>
#include <sstream>
#include <fstream>
#include <dirent.h>
#include <regex>

#include "DslApi.h"

// Set Camera RTSP URI's - these must be set to valid rtsp uri's for camera's on your network
// RTSP Source URI
std::wstring rtsp_uri_1 = L"rtsp://admin:11qqaa..@192.168.1.145:554/trackID=1";

int WINDOW_WIDTH = DSL_STREAMMUX_DEFAULT_WIDTH;
int WINDOW_HEIGHT = DSL_STREAMMUX_DEFAULT_HEIGHT;

// ##     
// # Function to be called on XWindow KeyRelease event    
// ##     
void xwindow_key_event_handler(const wchar_t* in_key, void* client_data)
{   
    std::wstring wkey(in_key); 
    std::string key(wkey.begin(), wkey.end());
    std::cout << "key released = " << key << std::endl;
    key = std::toupper(key[0]);
    if(key == "P"){
        dsl_player_pause(L"player");
    } else if (key == "R"){
        dsl_player_play(L"player");   
    } else if (key == "Q"){
        dsl_player_stop(L"player");
        dsl_main_loop_quit();
    }
}

int main(int argc, char** argv)
{  
    DslReturnType retval;

    // # Since we're not using args, we can Let DSL initialize GST on first call    
    while(true){    

        // # For each camera, create a new RTSP Source for the specific RTSP URI    
        retval = dsl_source_rtsp_new(L"rtsp-source", rtsp_uri_1.c_str(), DSL_RTP_ALL,     
            false, 0, 100, 2);
        if (retval != DSL_RESULT_SUCCESS)    
            return retval;

        // # New Overlay Sink, 0 x/y offsets and same dimensions as Tiled Display    
        retval = dsl_sink_window_new(L"window-sink", 0, 0, WINDOW_WIDTH, WINDOW_HEIGHT);
        if (retval != DSL_RESULT_SUCCESS) break;

        retval = dsl_player_new(L"player", L"rtsp-source", L"window-sink");
        if (retval != DSL_RESULT_SUCCESS) break;
            
        // # Add the XWindow event handler functions defined above    
        retval = dsl_player_xwindow_key_event_handler_add(L"player", xwindow_key_event_handler, nullptr);    
        if (retval != DSL_RESULT_SUCCESS) break;    
        
        // # Play the player    
        retval = dsl_player_play(L"player");
        if (retval != DSL_RESULT_SUCCESS) break;

        dsl_main_loop_run();
        retval = DSL_RESULT_SUCCESS;
        break;
    }
[... 13 lines cut ...]
0:00:00.007799172 10882 0x5622b12a62d0 INFO                     DSL src/DslServicesInfo.cpp:59:InfoInitDebugSettings:  : Pre-start Log file  = 
0:00:00.008249649 10882 0x5622b12a62d0 INFO                     DSL src/DslServices.cpp:154:GetServices:  : DSL Version: v0.26.alpha
0:00:00.008292335 10882 0x5622b12a62d0 INFO                     DSL src/DslServices.cpp:165:GetServices:  : Libcurl Initialized Successfully
0:00:00.008298114 10882 0x5622b12a62d0 INFO                     DSL src/DslServices.cpp:166:GetServices:  : Version: 7.58.0
0:00:00.008301724 10882 0x5622b12a62d0 INFO                     DSL src/DslServices.cpp:167:GetServices:  : Host: x86_64-pc-linux-gnu
0:00:00.008308864 10882 0x5622b12a62d0 INFO                     DSL src/DslServices.cpp:168:GetServices:  : Features: 4179869
0:00:00.008312227 10882 0x5622b12a62d0 INFO                     DSL src/DslServices.cpp:169:GetServices:  : SSL Version: GnuTLS/3.5.18
0:00:00.008315361 10882 0x5622b12a62d0 INFO                     DSL src/DslServices.cpp:170:GetServices:  : Libz Version: 1.2.11
0:00:00.008320529 10882 0x5622b12a62d0 INFO                     DSL src/DslServices.cpp:171:GetServices:  : Protocols: 0x7f6e165c1620
0:00:00.014266052 10882 0x5622b12a62d0 INFO                     DSL src/DslSourceBintr.cpp:519:DecodeSourceBintr:  : URI Path for File Source 'rtsp-source' = rtsp://admin:11qqaa..@192.168.1.145:554/trackID=1
0:00:00.015449021 10882 0x5622b12a62d0 INFO                     DSL src/DslServicesSource.cpp:907:SourceRtspNew:  : New RTSP Source 'rtsp-source' created successfully
0:00:00.017183620 10882 0x5622b12a62d0 INFO                     DSL src/DslServicesSink.cpp:113:SinkWindowNew:  : New Window Sink 'window-sink' created successfully
0:00:00.017615750 10882 0x5622b12a62d0 INFO                     DSL src/DslServicesPlayer.cpp:60:PlayerNew:  : New Player 'player' created successfully
0:00:00.017658862 10882 0x5622b12a62d0 INFO                     DSL src/DslServicesPlayer.cpp:655:PlayerXWindowKeyEventHandlerAdd:  : Player 'player' added X Window Key Event Handler successfully
0:00:00.545678253 10882 0x5622b12a62d0 INFO                     DSL DslNodetr.h:759:SetState:  : Changing state to 'PLAYING' for Bintr 'player'
0:00:00.550844704 10882 0x5622b12a62d0 INFO                     DSL src/DslPipelineXWinMgr.cpp:250:HandleBusSyncMessage:  : Prepare window handle received from source window-sink-nveglglessink
0:00:00.550967594 10882 0x5622b12a62d0 INFO                     DSL src/DslPipelineXWinMgr.cpp:373:CreateXWindow:  : Creating new XWindow: x-offset = 0, y-offset = 0, width = 1920, height = 1080
0:00:00.551479889 10882 0x5622b12a62d0 INFO                     DSL DslNodetr.h:775:SetState:  : State change will complete asynchronously for Bintr 'player'
0:00:00.612876572 10882 0x5622b134a630 INFO                     DSL src/DslSourceBintr.cpp:2164:HandleSelectStream:  : Media = 'video' for RtspSourceBitnr 'rtsp-source'
[... cut ...]
0:00:00.637913875 10882 0x7f6d900034f0 INFO                     DSL src/DslSourceBintr.cpp:2237:HandleSourceElementOnPadAdded:  : Caps structs name application/x-rtp
0:00:00.637953618 10882 0x7f6d900034f0 INFO                     DSL src/DslSourceBintr.cpp:2238:HandleSourceElementOnPadAdded:  : Media = 'video' for RtspSourceBitnr 'rtsp-source'
0:00:00.638062107 10882 0x7f6d900034f0 INFO                     DSL src/DslSourceBintr.cpp:2257:HandleSourceElementOnPadAdded:  : Video decode linked for RTSP source 'rtsp-source'
Segmentation fault (core dumped)
```

rjhowell44, 2022-12-06:
@youngjae-avikus re: ***What I found this time was that the RtspSourceSelectStreamCB, which was mentioned, was not invoked immediately after the source was created, but was executed at the time of pipeline execution.***  ... yes, this is the way it is supposed to work.  As mentioned, the `rtspsrc` calls the select-stream callback once it has read the caps from the actual stream... so yes, the Pipeline needs to be playing for the source to read the stream.  

This looks good.
```
0:00:00.612876572 10882 0x5622b134a630 INFO                     DSL src/DslSourceBintr.cpp:2164:HandleSelectStream:  : Media = 'video' for RtspSourceBitnr 'rtsp-source'
0:00:00.612898460 10882 0x5622b134a630 INFO                     DSL src/DslSourceBintr.cpp:2165:HandleSelectStream:  : Encoding = 'JPEG' for RtspSourceBitnr 'rtsp-source'
```

But we need to change the code you added to search for the upper case `JPEG` 
```
            else if (encoding.find("jpeg") != std::string::npos)
            {
                m_pParser = DSL_ELEMENT_NEW("jpegparse", GetCStrName());
                m_pDepay = DSL_ELEMENT_NEW("rtpjpegdepay", GetCStrName());
            }
```
should be
```
            else if (encoding.find("JPEG") != std::string::npos)
```
try adding a log statement or breakpoint to make sure the code block is getting executed..

youngjae-avikus, 2022-12-06:
> @youngjae-avikus re: _**What I found this time was that the RtspSourceSelectStreamCB, which was mentioned, was not invoked immediately after the source was created, but was executed at the time of [...]

Oh, I didn't mention it, but this part is the result of the code being modified and executed
But after that, a fault occurs to see if the decoder or sink does not fit

I think I can mix it with `imagesourcesrc pipeline` properly...

youngjae-avikus, 2022-12-07:
> @youngjae-avikus I've had a chance to review the code in more detail... I'm pretty sure I know what we need to do.
> 
> The selection of decoder is currently fixed in the constructor.
> 
> ```c
>         m_pDecodeBin = DSL_ELEMENT_NEW("decodebin", name);
> ```
> 
> This and all associated logic will need to move into the `select-stream` callback so that `decodebin` or `nvjpegdec` can selected and linked in based on stream type. The `LinkAll()` method will need to be updated accordingly.
> 
> I can get these changes in tomorrow, but I will need you to test the final solution with your camera.

Yes, I'll wait

rjhowell44, 2022-12-08:
@youngjae-avikus I've started the process of refactoring the RTSP Source here ... https://github.com/prominenceai/deepstream-services-library/tree/rtsp_jpeg

The `select-stream` is now selecting the decoder to use.

```C
            if (encoding.find("H264") != std::string::npos)
            {
                m_pParser = DSL_ELEMENT_NEW("h264parse", GetCStrName());
                m_pDepay = DSL_ELEMENT_NEW("rtph264depay", GetCStrName());
                m_pDecodeBin = DSL_ELEMENT_NEW("decodebin", GetCStrName());
            }
            else if (encoding.find("H265") != std::string::npos)
            {
                m_pParser = DSL_ELEMENT_NEW("h265parse", GetCStrName());
                m_pDepay = DSL_ELEMENT_NEW("rtph265depay", GetCStrName());
                m_pDecodeBin = DSL_ELEMENT_NEW("decodebin", GetCStrName());
            }
            else if (encoding.find("JPEG") != std::string::npos)
            {
                m_pParser = DSL_ELEMENT_NEW("jpegparse", GetCStrName());
                m_pDepay = DSL_ELEMENT_NEW("rtpjpegdepay", GetCStrName());
                m_pDecodeBin = DSL_ELEMENT_NEW("nvjpegdec", GetCStrName());
            }
```
What I'm not sure of, however, is if the `nvjpegdec` decoder supports the `pad-added` and `child-added` and signaling ... you will need to see if the following causes errors, and if so we'll need to just link up the decoder in the `select-stream` callback. 
```C
            // Connect Decode Setup Callbacks
            g_signal_connect(m_pDecodeBin->GetGObject(), "pad-added", 
                G_CALLBACK(RtspDecodeElementOnPadAddedCB), this);
            g_signal_connect(m_pDecodeBin->GetGObject(), "child-added", 
                G_CALLBACK(OnChildAddedCB), this);
```

youngjae-avikus, 2022-12-08:
@rjhowell44 

Code executed after rtsp-jpeg branch merge
What I checked first is that nvjpegdec is not DSL_ELEMENT_NEW and occurs seg_fault

```cpp
else if (encoding.find("JPEG") != std::string::npos)
{
    m_pParser = DSL_ELEMENT_NEW("jpegparse", GetCStrName());
    LOG_INFO("Define jpegparse");
    m_pDepay = DSL_ELEMENT_NEW("rtpjpegdepay", GetCStrName());
    LOG_INFO("Define rtpjpegdepay");
    m_pDecodeBin = DSL_ELEMENT_NEW("nvjpegdec", GetCStrName());
    LOG_INFO("Define nvjpegdec");
}
```

[image]

I've check `nvjpegdec` using gst-inspect

[image]

youngjae-avikus, 2022-12-14:
@rjhowell44 
Excuse me, but is there anything that has been made? Or is there anything else I need to check?
It would be better to input the image source, right?

rjhowell44, 2022-12-15:
@youngjae-avikus can you increase the gstreamer logging?  I need to see more than just the DSL logs. Try with

`export GST_DEBUG=4`

youngjae-avikus, 2022-12-16:
@rjhowell44 
The file is large, put in discord.

rjhowell44, 2022-12-20:
@youngjae-avikus Looking at the log here...
```
0:00:00.828271016  3696 0x56080aff6c00 INFO                     DSL src/DslSourceBintr.cpp:2632:HandleSelectStream:  : Media = 'video' for RtspSourceBitnr 'rtsp-source'
0:00:00.828285478  3696 0x56080aff6c00 INFO                     DSL src/DslSourceBintr.cpp:2633:HandleSelectStream:  : Encoding = 'JPEG' for RtspSourceBitnr 'rtsp-source'
0:00:00.828585566  3696 0x56080aff6c00 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstjpegformat.so" loaded
0:00:00.828596836  3696 0x56080aff6c00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "jpegparse" named "rtsp-source-jpegparse"
0:00:00.828646973  3696 0x56080aff6c00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseParse@0x7f55e801ec40> adding pad 'sink'
0:00:00.828657730  3696 0x56080aff6c00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseParse@0x7f55e801ec40> adding pad 'src'
0:00:00.828686082  3696 0x56080aff6c00 INFO                     DSL src/DslSourceBintr.cpp:2658:HandleSelectStream:  : Define jpegparse
0:00:00.829178220  3696 0x56080aff6c00 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstrtp.so" loaded
0:00:00.829186671  3696 0x56080aff6c00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "rtpjpegdepay" named "rtsp-source-rtpjpegdepay"
0:00:00.829226547  3696 0x56080aff6c00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstRTPBaseDepayload@0x7f55e8029e60> adding pad 'sink'
0:00:00.829239611  3696 0x56080aff6c00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstRTPBaseDepayload@0x7f55e8029e60> adding pad 'src'
0:00:00.829255753  3696 0x56080aff6c00 INFO                     DSL src/DslSourceBintr.cpp:2660:HandleSelectStream:  : Define rtpjpegdepay
0:00:00.830291341  3696 0x56080aff6c00 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/deepstream/libnvdsgst_jpegdec.so" loaded
0:00:00.830302014  3696 0x56080aff6c00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "nvjpegdec" named "rtsp-source-nvjpegdec"
0:00:00.830348805  3696 0x56080aff6c00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstVideoDecoder@0x7f55e80457e0> adding pad 'sink'
0:00:00.830358482  3696 0x56080aff6c00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstVideoDecoder@0x7f55e80457e0> adding pad 'src'
Segmentation fault (core dumped)
```
Everything looks good right up to the fault.  Can you try a simple test for me just to confirm what I think is happening. Could you add a `return true` to the JPEG case as below.  I need to confirm that it still crashes in the same way.

```C
else if (encoding.find("JPEG") != std::string::npos)
{
    m_pParser = DSL_ELEMENT_NEW("jpegparse", GetCStrName());
    LOG_INFO("Define jpegparse");
    m_pDepay = DSL_ELEMENT_NEW("rtpjpegdepay", GetCStrName());
    LOG_INFO("Define rtpjpegdepay");
    m_pDecodeBin = DSL_ELEMENT_NEW("nvjpegdec", GetCStrName());
    LOG_INFO("Define nvjpegdec");
    return true;
}
```

youngjae-avikus, 2022-12-20:
> @youngjae-avikus Looking at the log here...
>
> ```
> 0:00:00.828271016  3696 0x56080aff6c00 INFO                     DSL src/DslSourceBintr.cpp:2632:HandleSelectStream:  : Media = 'video' for [...]

The result is the same segfault.

youngjae-avikus, 2022-12-20:
> First, I will check if pgie and tracker are connected to the command line

@rjhowell44 

As I said before, I've done it with a command line, It's working well. The test was done with two different rtsps, and they all worked fine.

```bash
gst-launch-1.0 -vvv rtspsrc location=rtsp://192.168.1.140:554/jpeg ! rtpjpegdepay ! jpegparse ! nvjpegdec ! m.sink_0 nvstreammux name=m batch-size=1 width=1920 height=1080 ! nvvideoconvert ! nvinfer config-file-path=/opt/dsl/nas_data/YOLOV3/config_infer_primary_yoloV3.txt ! nvmultistreamtiler width=1920 height=1080 ! nvvideoconvert ! nvdsosd ! nveglglessink sync=0
```

rjhowell44, 2022-12-20:
@youngjae-avikus understand that you can do it from the command line as a non-decode source (meaning a source that can select between H264, H265, and JEPG).  But there is clearly a problem loading the nvjpegdec plugin from the context of the rtspsrc signal callback.  The above could be implemented as a new source type that does not support dynamic stream-selection. 

Question ... what advantage do you think JPEG will give you H256?

youngjae-avikus, 2022-12-20:
> @youngjae-avikus understand that you can do it from the command line as a non-decode source (meaning a source that can select between H264, H265, and JEPG). But there is clearly a problem loading [...]

I'm sorry I did not understand the question correctly because I was shallow about background knowledge.

1. First of all, can you tell me why it works on the command line, but it doesn't work on the dsl code by linking it to `non-decode source`, `select-stream`, and `callback` words?

2. **what advantage do you think JPEG will give you H256?** Are you talking about the reason why I use jpeg instead of h265?

Thank you

rjhowell44, 2022-12-20:
re 1. it appears to be an issue of calling context.  It seems that the `nvjpegdec` plugin crashes when loaded in the context of the `rtspsrc` plugin's `select-stream` callback.... but appears to work when loaded from the application's context. This would need to be confirmed with NVIDIA as the `nvjpegdec` is theirs. I'm curious as to why their dynamic RTSP decode examples only support H264 and H265.

re 2. yes, why use jpeg instead of h265? What advantage does it provide?

youngjae-avikus, 2022-12-21:
@rjhowell44 

re 1. I think it would be convenient to continue the conversation at discord
re 2. This is a very personal situation. But there was something in common that most of the cameras I used supported Rtsp/Jpeg

1. Currently, there are two options for channels that I can use in the inference (HSUB, JPEG). However, as shown in the image attachment, HSUB supports a maximum VGA resolution.
[image]

2. Streaming techniques such as H264 and H265 usually operate on an iframe-based basis (which means compress images during time I), so if the bandwidth is low, sometimes the images are crushed and dented. So, from the point of view of the inference, the crushed video can have fatal consequences. However, in the case of jpeg, it uses a lot of bandwidth, but it has the advantage of not causing image crushing

rjhowell44, 2022-12-26:
@youngjae-avikus this issue has been superseded with a new issue 
**"Extent the RTSP Source to support JPEG encoding #891"** 

closing..

closed 2022-12-26 by rjhowell44